FreeCAD/FreeCAD #16565: FreeCAD segfault while using the measure tool

ilmagico, 2024-09-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm running FreeCAD from latest snap edge channel, EndeavourOS linux (arch based), `FreeCAD 1.1.0 Revision: 38740 (Git)`.
I was trying to measure some part from a STEP file with the measuring tool, I had the part design workbench open, and all of a sudden FreeCAD crashed. I'm not sure if I can reproduce  it (I'll try) but before I lose it, this is the stack trace.

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x78ff65e42520]
#1  0x78ff6634bf9c in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_assign(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) from /lib/x86_64-linux-gnu/libstdc++.so.6+0x2c
#2  0x78ff689396c3 in App::DocumentObjectT::operator=(App::DocumentObjectT const&) from /snap/freecad/1136/usr/lib/libFreeCADApp.so+0x23
#3  0x78ff68939769 in App::DocumentObjectT::DocumentObjectT(App::DocumentObjectT const&) from /snap/freecad/1136/usr/lib/libFreeCADApp.so+0x69
#4  0x78ff68939f78 in App::SubObjectT::SubObjectT(App::SubObjectT const&) from /snap/freecad/1136/usr/lib/libFreeCADApp.so+0x28
#5  0x78ff574bad68 in Measure::MeasureRadius::parseSelection(std::vector<App::MeasureSelectionItem, std::allocator<App::MeasureSelectionItem> > const&) from /snap/freecad/1136/usr/lib/Measure.so+0x48
#6  0x78ff42a6cdf6 in Gui::TaskMeasure::update() from /snap/freecad/1136/usr/lib/MeasureGui.so+0x8e6
#7  0x78ff69ad4ee0 in Gui::SelectionObserver::_onSelectionChanged(Gui::SelectionChanges const&) from /snap/freecad/1136/usr/lib/libFreeCADGui.so+0x30
#8  0x78ff69725501 in boost::signals2::detail::signal_impl<void (Gui::SelectionChanges const&), boost::signals2::optional_last_value<void>, int, std::less<int>, boost::function<void (Gui::SelectionChanges const&)>, boost::function<void (boost::signals2::connection const&, Gui::SelectionChanges const&)>, boost::signals2::mutex>::operator()(Gui::SelectionChanges const&) from /snap/freecad/1136/usr/lib/libFreeCADGui.so+0x1f1
#9  0x78ff69ad6a06 in Gui::SelectionSingleton::slotSelectionChanged(Gui::SelectionChanges const&) from /snap/freecad/1136/usr/lib/libFreeCADGui.so+0x216
#10  0x78ff69725501 in boost::signals2::detail::signal_impl<void (Gui::SelectionChanges const&), boost::signals2::optional_last_value<void>, int, std::less<int>, boost::function<void (Gui::SelectionChanges const&)>, boost::function<void (boost::signals2::connection const&, Gui::SelectionChanges const&)>, boost::signals2::mutex>::operator()(Gui::SelectionChanges const&) from /snap/freecad/1136/usr/lib/libFreeCADGui.so+0x1f1
#11  0x78ff69ad6cbc in Gui::SelectionSingleton::notify(Gui::SelectionChanges&&) from /snap/freecad/1136/usr/lib/libFreeCADGui.so+0x1cc
#12  0x78ff69adc48d in Gui::SelectionSingleton::rmvSelection(char const*, char const*, char const*, std::vector<Gui::SelectionSingleton::SelObj, std::allocator<Gui::SelectionSingleton::SelObj> > const*) from /snap/freecad/1136/usr/lib/libFreeCADGui.so+0x86d
#13  0x78ff697d0f75 in Gui::SoFCUnifiedSelection::setSelection(std::vector<Gui::SoFCUnifiedSelection::PickedInfo, std::allocator<Gui::SoFCUnifiedSelection::PickedInfo> > const&, bool) from /snap/freecad/1136/usr/lib/libFreeCADGui.so+0x3a5
#14  0x78ff697d2b41 in Gui::SoFCUnifiedSelection::handleEvent(SoHandleEventAction*) from /snap/freecad/1136/usr/lib/libFreeCADGui.so+0x331
#15  0x78ff65676095 in SoAction::traverse(SoNode*) from /snap/freecad/1136/usr/lib/x86_64-linux-gnu/libCoin.so.80c+0x65
#16  0x78ff65810908 in SoChildList::traverse(SoAction*, int, int) from /snap/freecad/1136/usr/lib/x86_64-linux-gnu/libCoin.so.80c+0x168
#17  0x78ff65857660 in SoGroup::doAction(SoAction*) from /snap/freecad/1136/usr/lib/x86_64-linux-gnu/libCoin.so.80c+0x50
#18  0x78ff6586c66c in SoSeparator::doAction(SoAction*) from /snap/freecad/1136/usr/lib/x86_64-linux-gnu/libCoin.so.80c+0x2c
#19  0x78ff65676095 in SoAction::traverse(SoNode*) from /snap/freecad/1136/usr/lib/x86_64-linux-gnu/libCoin.so.80c+0x65
#20  0x78ff6567fa9b in SoHandleEventAction::beginTraversal(SoNode*) from /snap/freecad/1136/usr/lib/x86_64-linux-gnu/libCoin.so.80c+0xab
#21  0x78ff656766e8 in SoAction::apply(SoNode*) from /snap/freecad/1136/usr/lib/x86_64-linux-gnu/libCoin.so.80c+0xc8
#22  0x78ff65825559 in SoEventManager::actuallyProcessEvent(SoEvent const*) from /snap/freecad/1136/usr/lib/x86_64-linux-gnu/libCoin.so.80c+0x79
#23  0x78ff6989b792 in SIM::Coin3D::Quarter::QuarterWidget::processSoEvent(SoEvent const*) from /snap/freecad/1136/usr/lib/libFreeCADGui.so+0x22
#24  0x78ff698aa159 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::processSoEvent(SoEvent const*) from /snap/freecad/1136/usr/lib/libFreeCADGui.so+0x59
#25  0x78ff698b6661 in Gui::NavigationStyle::processSoEvent(SoEvent const*) from /snap/freecad/1136/usr/lib/libFreeCADGui.so+0x41
#26  0x78ff698b8801 in Gui::CADNavigationStyle::processSoEvent(SoEvent const*) from /snap/freecad/1136/usr/lib/libFreeCADGui.so+0x211
#27  0x78ff698b68e2 in Gui::NavigationStyle::processEvent(SoEvent const*) from /snap/freecad/1136/usr/lib/libFreeCADGui.so+0x62
#28  0x78ff698e7b1e in Gui::View3DInventorViewer::processSoEvent(SoEvent const*) from /snap/freecad/1136/usr/lib/libFreeCADGui.so+0x6e
#29  0x78ff698922b5 in SIM::Coin3D::Quarter::EventFilter::eventFilter(QObject*, QEvent*) from /snap/freecad/1136/usr/lib/libFreeCADGui.so+0x1f5
#30  0x78ff668bcbaa in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /snap/freecad/1136/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0xaa
#31  0x78ff6756c752 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /snap/freecad/1136/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x72
#32  0x78ff675743a4 in QApplication::notify(QObject*, QEvent*) from /snap/freecad/1136/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xa84
#33  0x78ff6946fa28 in Gui::GUIApplication::notify(QObject*, QEvent*) from /snap/freecad/1136/usr/lib/libFreeCADGui.so+0x98
#34  0x78ff668bce4a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /snap/freecad/1136/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x13a
#35  0x78ff67572e87 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /snap/freecad/1136/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1b7
#36  /snap/freecad/1136/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1c91cb) [0x78ff675c91cb]
#37  /snap/freecad/1136/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1cc525) [0x78ff675cc525]
#38  0x78ff6756c763 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /snap/freecad/1136/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x83
#39  0x78ff6946fa28 in Gui::GUIApplication::notify(QObject*, QEvent*) from /snap/freecad/1136/usr/lib/libFreeCADGui.so+0x98
#40  0x78ff668bce4a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /snap/freecad/1136/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x13a
#41  0x78ff66d431e7 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /snap/freecad/1136/kf5/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0x687
#42  0x78ff66d16dbc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /snap/freecad/1136/kf5/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0xac
#43  /snap/freecad/1136/kf5/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/../../../libQt5XcbQpa.so.5(+0x76f5e) [0x78ff60894f5e]
#44  /snap/freecad/1136/kf5/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x26b) [0x78ff64511d3b]
#45  /snap/freecad/1136/kf5/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0xab258) [0x78ff64567258]
#46  /snap/freecad/1136/kf5/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x78ff6450f3e3]
#47  0x78ff66916338 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /snap/freecad/1136/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x68
#48  0x78ff668bb76b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /snap/freecad/1136/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#49  0x78ff668c3d14 in QCoreApplication::exec() from /snap/freecad/1136/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x94
#50  0x78ff693aadc8 in Gui::Application::runApplication() from /snap/freecad/1136/usr/lib/libFreeCADGui.so+0x6c8
[... cut ...]
Version: 1.1.0dev.38740 (Git) Snap 1136
Build type: Release
Branch: main
Hash: b4923d1c584a4dce794d91a4b090a5293569223a
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Assembly4 0.50.15
  * FreeCAD-themes.backup1725504742.3579178 (Disabled)
  * FreeCAD-themes
  * A2plus 0.4.65
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ilmagico, 2024-09-16:
The bug seems pretty easy to reproduce:

1. Open FreeCAD
2. Create a new Part & Body (or open a file)
    * Add a Sketch, create a rectangle with two circles (holes) inside
    * Close sketch & pad: you should have a part with two holes
3. Try to measure the distance between two holes (or, likely, any distance)
4. Click "Close"
5. Repeat step 3 & 4 until it crashes

If it doesn't crash, keep measuring, clicking "Reset", "Close", etc. it'll eventually crash.

https://github.com/user-attachments/assets/8ed79dfb-fbac-4b1d-9984-6a89ccb157bf

maxwxyz, 2024-09-16:
@hlorus FYI

xtemp09, 2024-09-16:
The segmentation fault occurs because _FreeCAD_ tries to do something with empty selection (what can you do with empty selection?).

<details>

<summary>
Backtrace
</summary>

```cpp
Thread 1 "FreeCAD" received signal SIGSEGV, Segmentation fault.
std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_assign (
    this=0x7fffffffa690, __str=<error reading variable: Cannot access memory at address 0x28>)
    at /usr/include/c++/13/bits/basic_string.tcc:282
282               const size_type __rsize = __str.length();
(gdb) bt
#0  std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_assign (
    this=0x7fffffffa690, __str=<error reading variable: Cannot access memory at address 0x28>)
    at /usr/include/c++/13/bits/basic_string.tcc:282
#1  0x00007ffff52d407f in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::assign (__str=<error reading variable: Cannot access memory at address 0x28>, this=0x7fffffffa690)
    at /usr/include/c++/13/bits/basic_string.h:1577
#2  std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::operator= (
    __str=<error reading variable: Cannot access memory at address 0x28>, this=0x7fffffffa690)
    at /usr/include/c++/13/bits/basic_string.h:814
#3  App::DocumentObjectT::operator= (this=this@entry=0x7fffffffa670, obj=...)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/App/DocumentObserver.cpp:137
#4  0x00007ffff52d4110 in App::DocumentObjectT::DocumentObjectT (this=this@entry=0x7fffffffa670, other=...)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/App/DocumentObserver.cpp:98
#5  0x00007ffff52d5086 in App::SubObjectT::SubObjectT (this=this@entry=0x7fffffffa670)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/App/DocumentObserver.cpp:263
#6  0x00007fffd34f2de9 in App::MeasureSelectionItem::MeasureSelectionItem (this=0x7fffffffa670)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/App/MeasureManager.h:56
#7  Measure::MeasureRadius::parseSelection (this=0x55555a61e1b0, selection=...)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Mod/Measure/App/MeasureRadius.cpp:112
#8  0x00007fffcb4e1611 in Gui::TaskMeasure::update (this=0x55555aa3c9e0)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Mod/Measure/Gui/TaskMeasure.cpp:333
#9  0x00007fffcb4e19b9 in Gui::TaskMeasure::onSelectionChanged (this=<optimized out>, msg=...)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Mod/Measure/Gui/TaskMeasure.cpp:460
#10 0x00007ffff7dce76f in Gui::SelectionObserver::_onSelectionChanged (this=<optimized out>, msg=...)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Gui/Selection.cpp:153
#11 0x00007ffff7ddd123 in std::__invoke_impl<void, void (Gui::SelectionObserver::*&)(Gui::SelectionChanges const&), Gui::SelectionObserver*&, Gui::SelectionChanges const&> (__f=<optimized out>, __t=<optimized out>)
    at /usr/include/c++/13/bits/invoke.h:74
#12 0x00007ffff7ddd141 in std::__invoke<void (Gui::SelectionObserver::*&)(Gui::SelectionChanges const&), Gui::SelectionObserver*&, Gui::SelectionChanges const&> (__fn=<optimized out>)
    at /usr/include/c++/13/bits/invoke.h:90
#13 std::_Bind<void (Gui::SelectionObserver::*(Gui::SelectionObserver*, std::_Placeholder<1>))(Gui::SelectionChanges const&)>::__call<void, Gui::SelectionChanges const&, 0ul, 1ul>(std::tuple<Gui::SelectionChanges const&>&&, std::_Index_tuple<0ul, 1ul>) (__args=..., this=<optimized out>)
    at /usr/include/c++/13/functional:506
#14 std::_Bind<void (Gui::SelectionObserver::*(Gui::SelectionObserver*, std::_Placeholder<1>))(Gui::SelectionChanges const&)>::operator()<Gui::SelectionChanges const&, void>(Gui::SelectionChanges const&) (
    this=<optimized out>) at /usr/include/c++/13/functional:591
#15 boost::detail::function::void_function_obj_invoker1<std::_Bind<void (Gui::SelectionObserver::*(Gui::SelectionObserver*, std::_Placeholder<1>))(Gui::SelectionChanges const&)>, void, Gui::SelectionChanges const&>::invoke(boost::detail::function::function_buffer&, Gui::SelectionChanges const&) (function_obj_ptr=..., 
    a0=...) at /usr/include/boost/function/function_template.hpp:158
#16 0x00007ffff7b19b43 in boost::function1<void, Gui::SelectionChanges const&>::operator() (
    this=<optimized out>, a0=...) at /usr/include/boost/function/function_template.hpp:771
#17 0x00007ffff7b19c01 in boost::signals2::detail::call_with_tuple_args<boost::signals2::detail::void_type>::m_invoke<boost::function<void (Gui::SelectionChanges const&)>, 0u, Gui::SelectionChanges const&>(boost::function<void (Gui::SelectionChanges const&)>&, boost::signals2::detail::unsigned_meta_array<0u>, std::tuple<Gui::SelectionChanges const&> const&, boost::enable_if<boost::is_void<boost::function<void (Gui::SelectionChanges const&)>::result_type>, void>::type*) const (args=std::tuple containing = {...}, func=..., 
    this=<optimized out>) at /usr/include/boost/signals2/detail/variadic_slot_invoker.hpp:105
#18 boost::signals2::detail::call_with_tuple_args<boost::signals2::detail::void_type>::operator()<boost::function<void (Gui::SelectionChanges const&)>, Gui::SelectionChanges const&, 1ul>(boost::function<void (Gui::SelectionChanges const&)>&, std::tuple<Gui::SelectionChanges const&> const&, mpl_::size_t<1ul>) const (
    args=std::tuple containing = {...}, func=..., this=<optimized out>)
    at /usr/include/boost/signals2/detail/variadic_slot_invoker.hpp:90
[... cut ...]
#93 0x0000555555571152 in main (argc=<optimized out>, argv=<optimized out>)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Main/MainGui.cpp:301
(gdb)
```

```ruby
OS: Ubuntu 24.04.1 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38740 (Git)
Build type: Debug
Branch: master
Hash: b4923d1c584a4dce794d91a4b090a5293569223a
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk , OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

</details>

PR #16600 (2024-09-23): [MeasureGui] Store measure type as local variable
This solves #16565 which was introduced in #15122 where the local measureType variable was moved to a class member. 

With this PR the measure type is stored locally again and the measurement label is retrieved from the mode dropdown when saving the measurement.

Fixes #16565
commit af9097ce (2024-09-23): MeasureGui: Move _mMeasureType member to local variable

Fixes #16565
commit b2b0af95 (2024-09-23): MeasureGui: Move _mMeasureType member to local variable

Fixes #16565
closed 2024-09-23 by chennes