FreeCAD/FreeCAD #16566: BIM organisation : setup GitHub Project board for BIM development

marcuspollio, 2024-09-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently, the BIM Workbench has a good fifty or so classified issues. Also no broad and up-to-date "overview" of the current status and future plans is easily available. Some info can be found on the forum, blogs or other places, but these are mostly solo developer efforts, lessening the potential for a more collective move.

With the growing interest in the BIM side of FreeCAD over the recent years and the increasing number of contributors, a degree of organization is advised. It could appeal to users and contributors both professional or enthusiast in the built environment field. The [Mechanical Advisory Group](https://github.com/orgs/FreeCAD/projects/28) has its Project board now. We should setup one for BIM too: The BIM Team, or whatever the name ! =D

A few questions :

1) What criteria to define issues included or not ? Workbenches, scope, field, etc. ?
2) How to sort issues (columns) ? Ideally something else than what Labels and Milestone already provide.
3) Should there be a particular focus ? Helping existing contributors organize themself (set priorities), facilitate on-boarding, etc ?  
4) Who takes care and make sure it stays relevant and useful ? Members of a particular GitHub Team ?
5) Others questions to be added...

Thank you !

### Full version info

```shell
Not applicable
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

marcuspollio, 2024-09-16:
@yorikvanhavre @Roy-043 @paullee0 @furgo16 @kaiwas @ semhustej @gianlucafiore

furgo16, 2024-09-16:
Great initiative, always supportive of collaborative efforts. Happy to help wherever I can.

I wonder if codifying Yorik's roadmap into issues and milestones could be a good starting point: https://github.com/yorikvanhavre/FreeCAD-NativeIFC/blob/main/doc/roadmap2024.md. Prioritizing the current list of BIM-labeled issues in a dedicated GitHub project board could be another strategy to get an overview on where to focus efforts.

I'll be traveling in the next 2 weeks. I might not be too responsive, but I'll be watching the thread and provide input if I can.

kaiwas, 2024-09-16:
> Thank you !

I'd be happy to take part.
Unfortunately, I can't help financially yet (they disabled this option for my country) and I don't know how to write code.
But I can generate ideas or criticize bad decisions )))

> 4\. Who takes care and make sure it stays relevant and useful ? Members of a particular GitHub Team ?

If I understood correctly, Yorik is the main driver of this process.

yorikvanhavre, 2024-09-16:
200% agree! Just waiting a bit to see who here is interested in participating to that effort, and I'll set up a project and give interested people the permissions to edit it. I guess that will configure the beginnings of a "BIM advisory group" :)

My [roadmap](https://github.com/yorikvanhavre/FreeCAD-NativeIFC/blob/main/doc/roadmap2024.md) is a bit of a personal project. It's the list of things I want to work on. It's not necessarily what is the most needed, or a good vision of the future of the BIM workbench. It's of course fine to take it as a starting point, but I'd be more interested in doing something more together, really establishing a plan of where BIM should go.

paullee0, 2024-09-16:
In fact, when the discussion of native IFC was discussed in the forum, the dual support of FC BIM and native IFC objects became one of the issue to be resolved.  Otherwise, not sure what FC BIM would be different from BBIM?  :)

hoshengwei, 2024-09-16:
I would like to participate in the group also.

kaiwas, 2024-09-16:
FreeCAD definitely has something that many other CAD systems don't have. I'm talking about "programmable" objects.
Geometry created using expressions is very flexible. Just imagine that you don't need to change the thickness of 15 walls - you can simply change the value in one cell.

Probably this useful feature has not yet been appreciated, but it is really very convenient. Endless edits during project development are very tiring, but it is much more pleasant to correct one number in one field than ten numbers in ten fields).

I tried to understand how the BIM blender works. And it uses direct modeling, like many others. There are already many such programs. But I don't remember any with expressions (we are talking about programs for designing buildings).
Despite my great love for the blender, I do not consider it as a working tool (simply because I already have another electronic drawing board).

It's a pity that the approach with expressions is not actively developing, but I hope that someday this will happen. In combination with direct modeling, this would be a blast!

(Maybe I'm the only one with such opinions, but I consider this one of the competitive advantages that could be developed)

gianlucafiore, 2024-09-16:
Hola muchachos, gracias por la mención, me encantaría ayudar y colaborar con el proyecto. Realmente Freecad me pareció una herramienta excelente desde que empecé a conocerlo. Yo soy hábil desarrollador en Node Js pero no tengo grandes conocimientos en python, por ende lamentablemente no puedo colaborar con código, ni dispongo del tiempo para abocarme al desarrollo activo de freecad. Pero si puedo colaborar con la hoja de ruta, testing, y lógicas de UX. Soy inminente arquitecto, y dueño de una empresa constructora, uso freecad asiduamente y voy reportando los Bugs que se me presentan, y hay varios aspectos a nivel usuario que puedo ir aportando para que mejoremos

gianlucafiore, 2024-09-16:
> I tried to understand how the BIM blender works. And it uses direct modeling, like many others. There are already many such programs. But I don't remember any with expressions (we are talking about programs for designing buildings. Maybe only Revit is suitable as an analogy). Despite my great love for the blender, I do not consider it as a working tool (simply because I already have another electronic drawing board).
> 

Yes, I have also tried Blender BIM and the approach is not bad, but the problem I find beyond the fact that it still has a long way to go, is the impossibility of handling a vector drawing from scratch without having to do strange things like extruding points and not to mention that the plan views are made by means of a ray-traced rendering, very far from being something precise, editable and compatible with the field of architecture or civil engineering. In short, extremely anti-intuitive for a general public.

I am currently eagerly awaiting the official release of FC 1.0 as I want to teach a course on it together with other existing Open Source tools, with Freecad as the central axis (at the Faculty of Architecture of the National University of Rosario) that no one ends up implementing for obvious marketing reasons in proprietary software.

furgo16, 2024-09-17:
> [`@marcuspolio`]: The [Mechanical Advisory Group](https://github.com/orgs/FreeCAD/projects/28) has its Project board now. We should setup one for BIM too: The BIM Team, or whatever the name ! =D

> [`@yorikvanhavre`]: 200% agree! Just waiting a bit to see who here is interested in participating to that effort, and I'll set up a project and give interested people the permissions to edit it. I guess that will configure the beginnings of a "BIM advisory group" :)

Other designations for such groups that are common in open source projects are Special Interest Group (SIG) or Working Group (WG). I think either of those could work well too, but I'll be happy to go with whatever name the majority agrees to.

To address the specific questions:

> 1. What criteria to define issues included or not ? Workbenches, scope, field, etc. ?

I would suggest starting small with a clearly defined scope:

1. Write a short wiki page with the purpose of the group and its scope.
2. I would make the scope about issues labelled as [~"BIM WB"](https://github.com/FreeCAD/FreeCAD/issues?q=is%3Aissue+is%3Aopen+label%3A%22Mod%3A+BIM%22) for starters

That would give us a clear definition of what we're trying to do, give us actionable items (start with adding issues to the project board), allow us to point others to what the group does, potentially encourage others to join, and provide a foundation to evolve.

In terms of evolving,  some of the smaller steps could be to track also BIM-related issues from Draft, TechDraw or other workbenches. The bigger steps to me would be to include the number of users and contributors to BIM in FreeCAD.

> 2. How to sort issues (columns) ? Ideally something else than what Labels and Milestone already provide.

I think that once we have a set of issues on the project board, we'll have a better picture. I personally find the priority lanes in the [Mechanical Advisory Group's project board](https://github.com/orgs/FreeCAD/projects/28/views/1) a good idea.

> 3. Should there be a particular focus ? Helping existing contributors organize themself (set priorities), facilitate on-boarding, etc ?

I think both of these make sense. I'll mention again the need to document the guidelines for managing issues in the board, so that any individual in the group can make educated decisions on what to include, without having to ask for permission.

> 4. Who takes care and make sure it stays relevant and useful ? Members of a particular GitHub Team ?

I'd say those who have expressed interest in this issue, and whoever else with a traceable contribution record in FreeCAD who also does. They could then be added to a GitHub team which manages the board. 

If this idea takes off, we should also think of setting up a communications channel for the group for coordination.

Roy-043, 2024-09-17:
I am for this and will help where I can.

luzpaz, 2024-09-17:
maybe @Moult and @aothms would be interested in participating as well ?

Moult, 2024-09-17:
It's probably worthwhile having a little show and tell from what we've developed over at Bonsai about how (Native) IFC works.

yorikvanhavre, 2024-09-19:
Yesterday at the FPA meeting @sliptonic raised an important aspect: We want FreeCAD as much as possible to work as an unified application. So he actually advised against forming a concurrent group to the CAD advisory group, and rather work with BIM as a CAD standard too. I liked that idea very much. We should actually work together to have FreeCAD support standards, as many standards as we can, a similar way. Encourage "horizontal" sharing between the different specialties and workbenches.

Basically, we should always speak to each other in the FreeCAD universe, and find ways to harmonize ways of working. So we should actually have BIM specialists inside the CAD advisory group, and when there is a question where BIM specialists have different opinions that, for ex. mechanical specialists, that question should be worked on inside the CAD advisory group and an unified answer should be found. I think that kind of workflow would be hugely beneficial to FreeCAD.

So I'm thinking this discussion should be split into two parts:

1. Any of you would like to join the CAD advisory group? The discussions happen mainly (at the moment) on discord. But the idea is basically that the @FreeCAD/cad-advisory-group is pinged on github whenever someone from that group should have a look on questions regarding standards, or "best practices". Ideally, there should be BIM specialists in there that can giver their opinions on BIM-related questions.
2. We should still do the BIM github project. I'll still set that up and look at giving all of you interested people access to it. I would basically like to put the focus on trying to think globally, FreeCAD-wide, and how to integrate things between BIM and the rest of FreeCAD. I think that could become a good starting point there?

Maybe we still need some discussion outside of that too. I liked @Moult 's idea. What about scheduling two jitsi meetings? One to show Bonsai to FreeCAD BIMers, one to show FreeCAD to Bonsaiers? Basically a bit resurrecting those epic first osarch meetings? ;)

Moult, 2024-09-19:
My 2 cents that although CAD is an important aspect of BIM, BIM also contains a lot of other things:

 - Type relationships
 - Classification systems
 - Document associations
 - System topology and connectivity
 - Analytical geometry for structural, energy analysis, light simulation...
 - Work orders
 - Maintenance scheduling
 - Construction sequencing
 - Resource and equipment management
 - Suppliers, warranties, certification, test results, etc
 - Cost items
 - Carbon and environmental valuation
 - Operations and maintenance manuals

It'll be a great idea to organise 1/2 meeting(s)! Anyone else interested?

Moult, 2024-09-19:
> ... the plan views are made by means of a ray-traced rendering ...

@gianlucafiore by the way, the plan views and all geometry is all vector and made using OpenCASCADE, identical to FreeCAD (but tailored to IFC semantics). This might be something interesting to cover during this show and tell meeting.

aothms, 2024-09-20:
> Anyone else interested?

Count me in :)

paullee0, 2024-09-20:
More users who may had used FC BIM and maybe BBIM also would be helpful?

- F_Rosa?
- Theoryshaw?
- Regis?
- Dimitar?
- etc.

aothms, 2024-09-20:
> More users who may had used FC BIM and maybe BBIM also would be helpful?

cc @DutchSailor

yorikvanhavre, 2024-09-23:
> My 2 cents that although CAD is an important aspect of BIM, BIM also contains a lot of other things

Of course, but when you consider CAD not as traditional (AutoCAD-like) but as other disciplines do nowadays (Mech engineering, product design, etc), most of these things are for them also part of CAD. The idea here is not necessarily use the same solution for everybody (that could be ~~crippling~~ limiting), but to *think* these things together.

The idea we have with this CAD advisory group is never to dictate how developers should develop, but to have a little voice that says "In discipline XXX or YYY things are usually done that way". It could help get new ideas. It could also bring BIM ideas to the other fields. It's still the decision of the developer to listen or implement any of that, and the decision of the maintainers to merge or not merge.

It will be again hard to find a common time between all of us for the meeting :) I'd vote for something like 9h00 europe / 16h00 australia, but that's pretty bad for US people... How did we do at OSArch? I don't remember

marcuspollio, 2024-09-23:
> work as an unified application [...] encourage "horizontal" sharing between the different specialties and workbenches, FreeCAD-wide.

Totally agree !

> We should still do the BIM GitHub project.

Yes, it makes it easier to get organized, even for non code contributors testing, sorting out bugs, etc.

> meetings [...]

If these are recorded, it may be ok too for people in certain time zones (e.g. I am currently UTC +12 =D)

aothms, 2024-09-26:
There's something else we should schedule a meeting on. A little bird just told me that there are some preliminary IFC5 examples online https://github.com/buildingSMART/IFC5-development/ And I think we need to get a feel for (a) how proactive we want to be on this (b) strategy for implementation (does it even make sense to fit these into the same software library, is ifcopenshell as a project still needed?) (c) authoring strategy on how to deal with some of the novel features regarding the 'composition' of 'opinions' (this is USD speak) from multiple sources.

yorikvanhavre, 2024-09-30:
> Is ifcopenshell as a project still needed?

More than ever! Specially if we already have IFC5 models coming. IfcOpenShell is now almost a kind of engine in FreeCAD. (What about trying to get IfcOpenShell in Debian/Ubuntu, BTW? ;) )

I think we might have matter for a third encounter here!

gianlucafiore, 2024-10-01:
>work as an unified application [...] encourage "horizontal" sharing between the different specialties and workbenches, FreeCAD-wide.

In regards to this, I think that as an excellent user experience, could we address a general command bar that allows us to type and write suggestions to find them quickly without having to change the workbench constantly? I know that there are many detractors of these functions that resemble Autocad and perhaps generate hatred and contempt even by the most purist of freecad. But I think it would be a good key for productivity, being able to avoid diverting attention from the drawing to the toolbars to look for the required command, and even worse looking for it between the different workbenches. They will tell me that custom menus can be created, it is true, but there are commands from other workbenches that are invoked much less frequently, and working in BIM already entails dozens of buttons and icons, adding more only further reduces the 3D view for me in my case that I work on a laptop.

yorikvanhavre, 2024-10-01:
There is such a search bar addon IIRC

Moult, 2024-10-01:
I've started a thread about three interconnected topics that we can meet and share about:

https://community.osarch.org/discussion/2537/proposal-for-three-osarch-meetups/

gianlucafiore, 2024-10-01:
> Existe un complemento de barra de búsqueda IIRC

Yes, I have tried it, but it raises alerts about possible bugs, and it really has some bugs to fix, I did not mention it because this plugin is not official, but I do not see it as a bad candidate working correctly, perhaps adding some keyboard shortcut that allows you to start typing without having to go and click the search field

[image]

marcuspollio, 2024-10-02:
Hi @yorikvanhavre 

I took the initiative of creating the base [BIM project board](https://github.com/orgs/FreeCAD/projects/31).
I have put it directly into the FreeCAD organization (so not into the FreeCAD/FreeCAD repository), as I feel its scope is broader than just the FreeCAD software source code.
I must be changed to "Public" but only organization owners can change its visibility. Could one of you do that ? Thanks

I have invited only the few people I could (Yorik, Roy, kaiwas/Projectbp), the others have to be part of the FreeCAD organization before maybe ?

The board is empty for now. We should agree on how to organize it.

Once it's done, I'll close this issue as it's getting a bit off-topic =P

marcuspollio, 2024-10-02:
Hoi @berndhahnebach I've added you too. Let us know if it suits you. Thanks =D

yorikvanhavre, 2024-10-03:
Great initiative @marcuspollio . I just made it public. However, I think it would be best under FreeCAD/FreeCAD, though, as all other FreeCAD projects are there, it might be much easier discoverable.

Whoever else wants to participate, ping me and I'll add you guys to the FreeCAD org.

About the CAD advisory group, would any of you @marcuspollio , @furgo16 , @gianlucafiore , @semhustej  , @paullee0 , @hoshengwei or @kaiwas want to be part of it?

Basically, you would be added to that group here on github, and from time to time when there is a pull request where we think we need advice from that group, we ping the group. That means we would like people from that group to look at the PR, and if they feel they have something to say about it, add a comment.

There is no obligation here or anything, the idea here is basically: If you see something that makes you think "In BIM we usually do like this or like that", say it, it might be useful, or it might not, but it will be interesting to hear about how things work in other disciplines and we want other FreeCAD people to hear more about BIM too.

The CAD advisory group also has a discussion on discord, but that's up to you to participate to it or not, no obligation there either.

yorikvanhavre, 2024-10-03:
> You can also just link it in FreeCAD/FreeCAD (I guess that's the mechanism how the other projects appear there)?

Aah great, I didn't know that, thanks. Just did it!

furgo16, 2024-10-03:
> Whoever else wants to participate, ping me and I'll add you guys to the FreeCAD org.

@yorikvanhavre thank you. I'd be happy to join the org and contribute where I can.

> About the CAD advisory group, would any of you

I don't see myself as an expert in CAD or mechanical design, which is the main reservation I have. In any case, if I can be useful to the project, happy to join the group 

Regarding org-level or project-level board, I'd suggest project-level too. Before adding more things at the top level, I feel it would be good to think a bit more about how to organize it (starting with archiving some of the stale repos, for instance).

hoshengwei, 2024-10-03:
> About the CAD advisory group, would any of you @marcuspollio , @furgo16 , @gianlucafiore , @semhustej , @paullee0 , @hoshengwei or @kaiwas want to be part of it?

I am just an amateur enthusiast who wants to improve my coding skills through the FreeCAD project. In fact, FreeCAD is the first project I am contributing to, and I am very happy to join this project. Please make sure to count me in. Thank you!

paullee0, 2024-10-03:
@yorikvanhavre  Thanks, count me in.

semhustej, 2024-10-05:
> 
> About the CAD advisory group, would any of you @marcuspollio , @furgo16 , @gianlucafiore , @semhustej , @paullee0 , @hoshengwei or @kaiwas want to be part of it?
> 
I am not sure I could add a value, but since it is non-binding, you can sign me up. Thank you.

yorikvanhavre, 2024-10-08:
Ok @paullee0 @furgo16 @hoshengwei and @semhustej I added you to the @FreeCAD/cad-advisory-group . You should see an invitation somewhere on your github interface that you need to accept.

If any of you wants participate to the CAD standards discussions, the group has a discussion thread on discord at https://discord.com/channels/870877411049357352/1277336851044106360

luzpaz, 2024-10-09:
This ticket still relevant anymore ?

furgo16, 2024-10-09:
> Ok [...] I added you to the @FreeCAD/cad-advisory-group . You should see an invitation somewhere on your github interface that you need to accept.

Thanks! I got the invitation via e-mail and after accepting it, I could use the BIM development board with write access.

> This ticket still relevant anymore ?

I think rather than closing it, it might be worth moving it to a GitHub discussion. I'm not a big fan of GH discussions myself, but for the sake of keeping the conversation open, having everything development-related in one place, and for transparency, it might be a good compromise. Everyone in this issue already has a GitHub account and is active on the platform, so it also makes things easier.

furgo16, 2024-10-09:
I've started populating the [BIM development board](https://github.com/orgs/FreeCAD/projects/31) that @marcuspollio created, simply for testing how it could be used.

A few practical notes:

1. For now, I've simply populated the board all the BIM-relevant issues/prs I could find to provide an overview and get a feel for how the board could be used. A group discussion on goals and usage might be worth having.
1. In other words, as it stands now, the board is more or less at a playground stage, and not used to its full capabilities. However, it already provides a good, single place to go to to monitor the status of current BIM-related tasks.  
2. To me, there are a couple of low-hanging fruit items that we could start with: first zooming into the **BIM-related "Good first issue" issues**. The goal here would be to get new contributors up to pace with easy tasks, so that they can get started progressively with addressing more complex features or bugs. I feel this is critical to increase the lottery factor in the BIM project. The tasks for us here would be to ensure that for those issues there is a good understanding of the task at hand and high-level steps description of what's required to complete. The icing on the cake would be to find someone to pick a task up.
1. The second low-hanging fruit item would be to **assign priorities for the current BIM issues**. This would require a few guidelines on the criteria for priorities, but I think this could be discussed and documented rather quickly.  
1. It's not obvious from the UI, but issues/prs can be added to the board in bulk using standard GitHub search filters](https://docs.github.com/en/issues/planning-and-tracking-with-projects/managing-items-in-your-project/adding-items-to-your-project#bulk-adding-issues-and-pull-requests). I used `label:"Mod: BIM" is:open` on the search box to start adding issues, along with a simple search on `BIM` and `Arch`.
1. Addition of new elements (issue/pr) to the board can be done manually, as described above, or [automatically](https://docs.github.com/en/issues/planning-and-tracking-with-projects/automating-your-project/adding-items-automatically). I think adding elements automatically (i.e. those which have been labeled `Mod: BIM`) is worth exploring.