FreeCAD/FreeCAD #16567: techdraw darstellungs darstellungsfehler

entkon, 2024-09-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

die Linien werden nicht dargestellt
dieses teil können sie unter diesem link herunterladen
FreeCAD22 KA002 O-Ring Torus
https://entkon.de/freecad22-part-design-einfuehrung-in-klassische-konstruktion/

### Full version info

```shell
38728
2024/09/13
```

### Subproject(s) affected?

TechDraw

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-09-16:
It's a regression.

Open the attached [file](https://github.com/user-attachments/files/17010916/td-torus-test-021.FCStd.zip) (remove the zip extension) in V0.21 and 0.22.
In V0.22 the outline of the torus is missing from the TD view.

Roy-043, 2024-09-16:
@maxwxyz Should this be qualified as a Blocker?

WandererFan, 2024-09-16:
Is this different from https://github.com/FreeCAD/FreeCAD/issues/16136 and
https://github.com/FreeCAD/FreeCAD/pull/16145 ?

What version are you running.  Seems to work here.
[image]
[image]

OS: Linux Mint 21.2 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38740 +1 (Git)
Build type: debug
Branch: winFilespec
Hash: 78c1a110f4c6ee0b69444f5a821c0b83b2fa8a14
Python 3.10.12, Qt 5.15.2, Coin 4.0.0, Vtk , OCC 7.8.0
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fasteners 0.5.21

entkon, 2024-09-16:
Error in FreeCAD Version 22_38728 (13.09.2024):

In the TechDraw environment, lines are not displayed when using the "Torus" function. This may cause issues when creating technical drawings.

To download the part "FreeCAD22 KA002 O-Ring Torus," please use the following link:
https://entkon.de/wp-content/uploads/2024/09/FreeCAD22-KA002.zip

https://entkon.de/freecad22-part-design-einfuehrung-in-klassische-konstruktion/

luzpaz, 2024-09-16:
@entkon we need full [About](https://wiki.freecad.org/About) info

maxwxyz, 2024-09-16:
@entkon could you please try the latest dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Roy-043, 2024-09-16:
The problem occurs in the latest weekly version. I suppose the previous fixes are included in that version.
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38728 (Git)
Build type: Release
Branch: main
Hash: 9de78e27f4148e8032594788a9804486e769388b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

WandererFan, 2024-09-17:
Is it possible that 38768 AppImage and local build have source differences?  Qt, OCC and Python versions are different.  

today's main local build - OK
OS: Linux Mint 21.2 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38768 (Git)
Build type: debug
Branch: main
Hash: 758674d40be45e365f4148fb2a5ea6bd6f93bc74
Python 3.10.12, Qt 5.15.2, Coin 4.0.0, Vtk , OCC 7.8.0
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fasteners 0.5.21

today's RC1 local build - OK
OS: Linux Mint 21.2 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38752 (Git)
Build type: debug
Branch: releases/FreeCAD-1-0
Hash: 814945f72846f965aa53595a202f082a4618d424
Python 3.10.12, Qt 5.15.2, Coin 4.0.0, Vtk , OCC 7.8.0
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fasteners 0.5.21

recent AppImage - error
OS: Linux Mint 21.2 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38768 (Git) AppImage
Build type: Release
Branch: main
Hash: 758674d40be45e365f4148fb2a5ea6bd6f93bc74
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/Canada (en_CA)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fasteners 0.5.21

Syres916, 2024-09-17:
@WandererFan it's definitely an OCC issue with version 7.7.2, two builds based on the exactly the same libraries except OCC, the 7.8.2dev displays as expected and the 7.7.2 build gives the same incorrect view as reported.

Bad:
```
OS: EndeavourOS (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38768 (Git)
Build type: Release
Branch: main
Hash: 758674d40be45e365f4148fb2a5ea6bd6f93bc74
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * FreeCAD-themes 2024.7.24
```

Good:
```
OS: EndeavourOS (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38768 (Git)
Build type: Release
Branch: main
Hash: 758674d40be45e365f4148fb2a5ea6bd6f93bc74
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.2.dev
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * FreeCAD-themes 2024.7.24
```

Syres916, 2024-09-18:
@adrianinsaval I believe the PPA has OCCT 7.8.1 now, are the Conda builds still a ways from also going to the same version?

WandererFan, 2024-09-18:
Compiling OCCT 7.7.2 now.  Thanks for looking into this.

adrianinsaval, 2024-09-18:
It should be doable to go to occt 7.8, but I'm a little hesitant to do it this late in the game (my fault for not doing it before). It seems that 7.8 might be in a better state than 7.7 though so maybe it makes sense after all. I'll see about making that happen at least for the weeklies and if everything goes well I might do it for the rc too

commit 0d6f47d1 (2024-09-23): [TD]fix BrokenView selection (fix #16567)
commit 98cb3ae8 (2024-09-23): [TD]fix BrokenView selection (fix #16567)
closed 2024-09-23 by chennes