FreeCAD/FreeCAD #16569: Link popup window doesn't stay on top on macOS

pgilfernandez, 2024-09-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In macOS, when relinking any element using a "link popup window", the window doesn't stay on top when working again in the main window (for instance, when selecting edges for a pad that is created with a profile).
[image]
I have tested Windows builds and there it works as expected.

### Full version info

```shell
OS: macOS 12.6.7
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38728 (Git)
Build type: Release
Branch: main
Hash: 9de78e27f4148e8032594788a9804486e769388b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * PieMenu 1.8.0
  * ProDarkThemePreferencePack 1.0.0
  * Nodes 0.1.36
  * freecad.gears 1.2.0
  * OpenTheme 2024.9.1
  * Dracula 0.0.5
  * Silk 0.1.5
  * SelectorToolbar
  * fasteners 0.5.28
  * ExtremeProDark 2.9.0
  * Curves 0.6.47
```

### Subproject(s) affected?

PartDesign

### Anything else?

[test.zip](https://github.com/user-attachments/files/17010349/test.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

IsabelDR, 2024-09-16:
I'm experiencing the same problem: https://forum.freecad.org/viewtopic.php?t=90628

OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)

theosib, 2024-09-16:
Is this a duplicate of https://github.com/FreeCAD/FreeCAD/issues/15130 ?

pgilfernandez, 2024-09-16:
My issue is just that the window goes behind the main window, once you realize about it you can always select (relink) the elements you want, then recover the link popup and press OK button to confirm changes...

IsabelDR, 2024-09-16:
> My issue is just that the window goes behind the main window, once you realize about it you can always select (relink) the elements you want, then recover the link popup and press OK button to confirm changes...

We have the same issue, but you were/are smarter than me putting the Link Window out of the main window. To me the Link window just disappeared, now I know it doesn't.
I did what you said: move the Link Window out of the main window, select and hit OK.

As you said it is difficult for new users to find this on their own.

theosib, 2024-09-16:
This is what confused me for a while too. We can just Command-[grave accent] to switch back to that window, but it's definitely a bug.

pgilfernandez, 2024-09-17:
As a margin note, I found a curious behavior: when the link popup just opens and it is still on top of the main window, if you select anything (elements, features, edges,...) using the **cmd button** pressed, the popup doesn't go to the background.

FEA-eng, 2025-10-17:
@pgilfernandez or @maxwxyz Can you check if it's still relevant on macOS?