FreeCAD/FreeCAD #16570: When Wireframe draw style is selected in the sketcher view, draw style cannot be changed outside of sketcher view.

blank-supportgis, 2024-09-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Create a new Project and select `View->Draw Style->Flat Lines`
2. Create a new Sketch, draw a simple rectangle, close the sketch and extrude it with the pad command to make it a solid. This step is not related to the problem, but you need some kind of solid to actually see the effect.
3. Create another sketch on top of the solid and open it.
4. In the sketch view, select `View->Draw Style->Wireframe` and close the sketch view.
5. Select `View->Draw Style->Flat Lines` again. FreeCAD should claim that it is in Flat Lines mode, but the actual view should remain in Wireframe.
6. Switch back to Wireframe and then to Flat Lines again to fix the view.

### Full version info

```shell
OS: Ubuntu 20.04.6 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * btl 0.9.9 (Disabled)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-16:
Is this fixed by https://github.com/FreeCAD/FreeCAD/pull/16378 ?

xtemp09, 2024-09-16:
@maxwxyz, the PR #16378 does not fix this issue, I've just tested.

In this issue, _FreeCAD_ claims that _Flat Lines_ mode is on, while actually _Wireframe_ is, I reproduced it on

```ruby
OS: Ubuntu 24.04.1 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38740 (Git)
Build type: Debug
Branch: master
Hash: b4923d1c584a4dce794d91a4b090a5293569223a
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk , OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

CalligaroV, 2024-09-16:
@maxwxyz @xtemp09 I can confirm.

I made some tests with

```
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38768 +1 (Git)
Build type: Debug
Branch: Sketcher-ViewProviderSketch-unsetEdit-check-3D-view
Hash: b0ee10ffe6ed8ee218c2e661b94287d4681b416f
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.8.1
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * Assembly4 0.50.15
  * freecad.gears 1.2.0
  * Beltrami 1.1.0
  * sheetmetal 0.4.24
  * Curves 0.6.44
  * fasteners 0.5.26
  * kicadStepUpMod 11.2.7
  * OpenTheme 2024.8.10
  * Render 2024.8.4
  * Ondsel-Lens 2024.7.5.02

``` 
And the issue is still unsolved.

I noticed a couple of things during my tests:
 - This behavior isn't caused by the order (you can select `Wireframe` outside the sketch view, then select `Flat Lines` in sketch view and then, out of the sketch view, having troubles setting going back to `Wireframe`)
 - This behavior can be replicated regardless of the 2 draw styles you choose

Another interesting (but luckily not much surprising) thing I noticed is that, at least in my tests, if I uncheck `Disable shading in edit mode` from `Preferences->Sketcher->General` I can't reproduce the issue anymore.
Can someone of you check if this change in preference gives the same result I get?

If it's confirmed then I may come out with a fix in the next days.

TIA!

blank-supportgis, 2024-09-17:
> Another interesting (but luckily not much surprising) thing I noticed is that, at least in my tests, if I uncheck `Disable shading in edit mode` from `Preferences->Sketcher->General` I can't reproduce the issue anymore.

Yes, I can confirm this.

CalligaroV, 2024-09-17:
Thanks @blank-supportgis for confirming the workaround!
@xtemp09 I guess your :+1: in @blank-supportgis 's comment and mine means that you too confirm this, if not please correct me.

Anyway, to prepare a good fix I'd like to ask a question:
After we leave the sketch edit mode, which `Draw Style` do we want? The last selected (for example, in this issue `Wireframe`) or the one used before entering the sketch edit mode (in this issue `Flat Lines`)?

Asking because AFAICS one behavior needs modification in a certain part of the code, while the other one in other part(s)

TIA!

mpictor, 2024-10-13:
Possibly related: in 1.0.0RC2 I have something weird with the draw style and sketcher.
* Draw Style button in toolbar doesn't update on entering Sketcher
* Sketcher starts with `No shading`
* Draw Style is restored when leaving sketcher
  * but not always (?!)

TS;DR (too short; didn't read):
I have a part in Part Design workbench. Draw Style is set to `Flat lines`, but if I create a sketch, the draw style changes. It _shows_ as whatever was selected in part design, but seems to actually be `No shading` - if I click that one, the view is the same, whereas if I click on the already-selected type (same as in Part Design), it changes back to that style.

mpictor, 2024-11-07:
FYI @maxwxyz @luzpaz see my comment above about behavior in 1.0.0RC2. Not sure if it's the same bug or a different one. Let me know if I should create a separate issue for it.

maxwxyz, 2024-11-07:
Your last comment is a setting for sketcher which you can turn off in the preferences

mpictor, 2024-11-12:
My bad, thanks

closed 2025-02-24 by chennes