FreeCAD/FreeCAD #16573: BIM: The "manager IFC quantity"  does not take into account mirror elements.

kaiwas, 2024-09-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Very often, when constructing a building model, there are mirrored elements (for example, a rafter system). We have the ability to construct what we need using an array and its mirroring.
Visually, we have the desired result, but we cannot take it into account in the specification (quantity).

Video demonstration.

https://github.com/user-attachments/assets/c013e5b5-4aee-4e18-abdc-48ccc666f436

Probably, some way to take this into account is needed. Perhaps some property of the mirror is required.

The file for tests is attached (remove the zip extension).
[Mirror_test.FCStd.zip](https://github.com/user-attachments/files/17012293/Mirror_test.FCStd.zip)

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38728 (Git) AppImage
Build type: Release
Branch: main
Hash: 9de78e27f4148e8032594788a9804486e769388b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * PitchedRoof
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-09-16:
Indeed we did a similar fix for Draft arrays some time ago in https://github.com/FreeCAD/FreeCAD/issues/12802 , I think a similar fix is needed here.

furgo16, 2024-09-18:
While addressing this, it might be worth looking at https://github.com/FreeCAD/FreeCAD/pull/12899#issuecomment-2014534360 as well.