FreeCAD/FreeCAD #16574: Draft/BIM: "Expand Array  + Mirror" is an unusual result.

kaiwas, 2024-09-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The array in freecad has a great ability - we can change the position of some elements in any order. (For example, so that the rafter does not fall on the pipe).

But in the case of using a mirror, we get an incorrect result. In this case, the preview shows what we need.

Demonstration on video.

https://github.com/user-attachments/assets/cfcb65d1-4f4b-4edd-9a43-fe20e7cd5d66

The file for tests is attached (remove the zip extension).
[Mirror_test (EXP_Array).FCStd.zip](https://github.com/user-attachments/files/17012705/Mirror_test.EXP_Array.FCStd.zip)

In addition, the number of elements is not specified in the IFC manager quantity (This is an old feature). If necessary, I will create another topic on this issue.

https://github.com/user-attachments/assets/a89624d2-fa44-4fe0-bb6f-010cfd332e9c

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38728 (Git) AppImage
Build type: Release
Branch: main
Hash: 9de78e27f4148e8032594788a9804486e769388b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * PitchedRoof
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2024-09-18:
> In addition, the number of elements is not specified in the IFC manager quantity (This is an old feature). If necessary, I will create another topic on this issue.

If that's ok with you, I think it might be worth creating the separate issue for this. I too use expanded arrays for beams and displaced individual array elements, and some mirrors. I'm starting to use schedules and managers on my model, and I'd also test myself if the number of mirror elements is counted in other places than the IFC manager. Thanks!

marcuspollio, 2024-09-23:
I can confirm the issue with all "link" Array followed by a mirror on :

```
OS: GNU/Linux Ubuntu 22.04 LTS (DE: Gnome/pop)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38829 (Git)
Build type: Release
Branch: main
Hash: 9b409da59c34ba0d302d15cd2046685091e8a06f
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk , OCC 7.6.3
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

I could not find any other second Draft/BIM operations where this error also occurs.

FEA-eng, 2026-02-23:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.2.0dev.20260217 (Git shallow)
Build date: 2026/02/17 21:26:14
Build type: Release
Branch: (HEAD detached at 3bad5dabd)
Hash: 3bad5dabde38f1a901fa1475e30f5caee7f9846a
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```