FreeCAD/FreeCAD #16575: v1.0 RC AppImage uses ephemeral /tmp/ dir for CAM tool library

smartin015, 2024-09-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm trying out the new [1.0 release candidate of freecad](https://blog.freecad.org/2024/09/10/the-first-release-candidate-of-freecad-1-0-is-out/comment-page-1/), which is packaged as an AppImage. 

When I go to `Tools > Edit Parameters` and browse to `BaseApp/Preferences/Mod/CAM/` I see `LastPathToolLibrary` is set by default to something like `/tmp/.mount_FreeCAEG5EZ6/usr/Mod/CAM/Tools/Library`. Best I can tell, this directory is not used the next time I run the AppImage, which means any tools added to the library effectively vanish on the next run. I like it when my tool configs persist across restarts :)

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * btl 0.9.9
```

### Subproject(s) affected?

CAM

### Anything else?

I'm attempting to use https://github.com/knipknap/better-tool-library to manage my tooling, but AFAICT this problem is caused by AppImage default preferences and the addon itself isn't at fault.

I'm happy to contribute a fix, but I would need some guidance as it'd be my first contribution to FreeCAD.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-09-16:
What you are describing is well known but you perhaps didn't find the related [issue](https://github.com/FreeCAD/FreeCAD/issues/6707) as it applies to more than one core workbench:

CAM previously known as Path (Tool Library)
TechDraw (Templates, Hatches, LineGroups, Welding Symbols etc)
FEM (default folder for analysis etc)

plus some Addons such as CfdOF (I may well have missed some)

At present AddonManager stores it's cache in a variable _FreeCAD.getUserAppDataDir()_/webdata/cache so it doesn't matter whether I use an AppImage or self compiled build the location is always available. The current thinking is why not use an exact similar method but obviously under each workbench's name for their corresponding data. It's too late to get anything for 1.0 but it could be done for the following release if their is agreement and enough resources to get all to the same stage.

sliptonic, 2024-09-16:
I don't think there's anything to fix for CAM.   When you first start CAM and access the tool dock or library manager, it presents a dialog asking where you want to put the tool data.  The user can create a new location or select an existing location.  That is what is stored into the LastPathToolLibrary key.   It's only if the user presses 'cancel' that the temp location is written.  The user can still change the tool data location.

So I don't know what else we can do.  We're asking the user at an appropriate time to pick a persistent location.  If the user declines we're stuck.

I'm closing this issue as there's nothing to do but will happily reopen it if someone has a suggestion how to handle this case better.

closed 2024-09-16 by sliptonic

adrianinsaval, 2024-12-15:
> So I don't know what else we can do.  We're asking the user at an appropriate time to pick a persistent location.  If the user declines we're stuck.

IMO if the user declines setting a path the preference should be left empty and be dynamically determined on each run