FreeCAD/FreeCAD #16577: [Feature] Assembly joints editing improvement

pure-machinery, 2024-09-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The proposal is to make the parts whose geometry is not used by the joints hidden or transparent with the same color. If we were to choose the transparency option then the selection of said parts would be disabled. This will make it easier when editing joints. 

A problem arises when a user wants to edit the joint so that not the two initially selected parts are connected. 

For example: 
- Initially **Part 1** and **Part 2** are selected with a joint
- User edits the joint and deletes the reference from **Part 2** 
- User wants to select a geometric entity from **Part 3** 
- Problem

If **Part 3** is unselectable or hidden it will be an issue selecting it. 

Another option is to have the assembly temporary exploded while editing joints. When you edit the joint the camera can zoom out and after editing it will get back to it's previous position.

Another useful thing might be is to tell the user they are in selection mode if they delete one of the references:
[image]

If anyone has any other ideas feel free to discuss it!

### Full version info

```shell
OS: Arch Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: makepkg
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.12.6, Qt 6.7.2, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Installed mods: 
  * BillOfMaterials 0.1.4.2
  * sheetmetal 0.4.25
  * fasteners 0.5.28
  * fcgear 1.0.0
  * OpenTheme 2024.9.1
  * IconThemes
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-16:
cc @PaddleStroke

PaddleStroke, 2024-09-16:
Thanks for the suggestion. What could be done is to set other objects as transparent (or hidden) only when you have 2 markers selected. If you unselect one, then the rest comes back to full visibility (or backup at least in case it's already transparent).

Having a way to explode radially while making joints is also interesting. Needs some thoughts about how to implement it.

PaddleStroke, 2024-09-16:
You can assign me @luzpaz please

pure-machinery, 2024-09-16:
> Thanks for the suggestion. What could be done is to set other objects as transparent (or hidden) only when you have 2 markers selected. If you unselect one, then the rest comes back to full [...]

What do you mean by markers?

PaddleStroke, 2024-11-12:
@wwmayer I'm working on this feature request. The idea is that when a joint is being edited, the unrelated components get transparent and unselectable.

Is there a way to make objects truly unselectable? Currently you can do `vobj.Selectable = False` but that does not make the mouse pass-through. ie you cannot click on what is behind. Also it behaves weirdly if you double click the object is still being selected.

wwmayer, 2024-11-12:
> Is there a way to make any objects truly unselectable but temporarily?

Use an SoPickStyle node and set its style filed to `SoPickStyle::UNPICKABLE`

PaddleStroke, 2024-11-12:
Thanks. However this is rather tedious. I'm not sure even possible since there may not be a place to put the node before and only the required objects's geometry.

Is it ok if instead I fix the Selectable property? Currently it clearly does not work. I don't know what it is supposed to do but it is very weird that a "Selectable" property does NOT mean that an object is not selectable if this property is set to false.

PaddleStroke, 2024-11-12:
So my proposal is to :
Add to `ViewProviderGeometryObject` a SoPickStyle.

Change setSelectable function so that it just change the VP's SoPickStyle

wwmayer, 2024-11-12:
> Currently it clearly does not work.

Example?

If I create a Part box, set its Selectable property to false I cannot select it in the 3d view any more. I haven't checked how it's implemented but it doesn't use an SoPickStyle.

PaddleStroke, 2024-11-12:
I have just tried to replace the current system by a SoPickStyle in ViewProviderGeometricObject, and it appears that it changes nothing : 
- The bug with App::Part is still here
- The object is still no mouse-click pass through.

So that's not much progress

PaddleStroke, 2024-11-12:
There is already an opened issue about this App::Part + selectable issue btw : https://github.com/FreeCAD/FreeCAD/issues/14219

PaddleStroke, 2024-11-12:
I have a PR up, but it is blocked by 3 bugs : 
https://github.com/FreeCAD/FreeCAD/issues/14219
https://github.com/FreeCAD/FreeCAD/issues/17815
https://github.com/FreeCAD/FreeCAD/issues/17820

If you have some inputs on those @wwmayer I'm most interested. Thanks

wwmayer, 2024-11-12:
OK, I have to have a closer look at this. It will take me some time to analyze the problem...

wwmayer, 2024-12-04:
https://github.com/FreeCAD/FreeCAD/pull/18298

closed 2025-10-03 by chennes