FreeCAD/FreeCAD #16579: Possibility to block navigation cube

Roberto8303, 2024-09-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I ask for the possibility of being able to lock the navigation cube, useful when you are in a 2D sketcher environment and you use the right mouse button to exit the functions, inadvertently you can move the front view by tilting the sketch.

### Full version info

```shell
I ask for the possibility of being able to lock the navigation cube, useful when you are in a 2D sketcher environment and you use the right mouse button to exit the functions, inadvertently you can move the front view by tilting the sketch.
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-09-16:
Ok, now #16561 can be closed.

Roberto8303, 2024-09-16:
if you draw a 2d sketch only and dimension it to remove for example the dimension tool with the right mouse button, the work plane tilts and you have to press the front view again to see the 2d sketch better

Syres916, 2024-09-16:
Personally I think the title and description should be along the lines _Add the ability to lock the Viewport/Scene while carrying out an operation in Sketcher Edit Mode_. Some of us don't even use the Navigation Cube but would still find the facility useful especially if there was an easy to use keyboard shortcut to toggle it on/off. In Blender the equivalent is _Lock Camera_

luzpaz, 2024-09-16:
@Roberto8303 can you make a quick screencast of the issue and drag/drop it in to this ticket for the devs to quickly see the issue you're encountering ?

Roberto8303, 2024-09-17:
there may be a view lock toggle lock right next to the navigation cube