FreeCAD/FreeCAD #16581: Draft: Mirror "destructive mode".

kaiwas, 2024-09-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Sometimes you need to mirror an object without preserving the "parent" version of the object.

But it seems that there is no such option in FreeCAD.
The Draft_mirror tool works great, but it always leaves a parent copy. This is similar to the mirror modifier in Blender (the upper version in the video)
But maybe you need a tool that applies the mirror operation once? Let's call it a one-time mirror or a destructive mirror (for example, in the same Blender this is done by setting a negative scale along one of the axes. The lower version in the video)

https://github.com/user-attachments/assets/f6f4e251-8386-4d83-b784-a4f7b1d400e8

I am not against the "construction history", but I think that such a simple action from direct modeling would not be superfluous.

### Full version info

```shell
any versions
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kaiwas, 2024-09-16:
For example, I have put together a quick example with checkboxes in blender (geometry nodes).
Look how convenient this could be (three checkboxes for each axis in the object properties.

https://github.com/user-attachments/assets/d6e7230d-1295-42f5-9c79-7571fea4d49e

kaiwas, 2024-09-16:
> Draft WB care about how the PartDesign WB works?

Perhaps Part Design has conceptual limitations. But there is also Tech Draw. Maybe it would not be superfluous there either.
One of the simplest analogies is the "House". It can have only two options:

1) Original
2) Mirrored

The third option simply does not exist (it cannot be upside down).
And it is not clear what is easier to do - mirror only the drawing or mirror the entire model).
It seems that mirroring a cube along the X axis is not as easy as we would like. It seems that when changing the direction vectors, the object rotates, not the drawing.

https://github.com/user-attachments/assets/5a7b872a-62f8-45c6-8aba-f700a1956d5b

In general - the topic of "mirror" is quite extensive )))

kaiwas, 2024-09-16:
I think I even came up with a name for this tool (in Draft WB) so that there would be no confusion with the original. "Axial reflections". And it's not even hard to come up with an icon. You can take the original, add an axis and shade one of the reflections)

yorikvanhavre, 2024-09-19:
It would be quite easy to add an option to "dumb-shapify" the result of the mirroring and then delete the original object, but that's not what is want here I guess, because that would be easy for the user to do themselves.

The problem is that depending on the object, mirroring and keeping all the parametrics could be a huge difficulty. Imagine an object based on a sketch with constraints... Some of the constraints would need to be displaced, modified, others not... Basically there is no way to implement such a tool in one place, each and every type of object would need its own mirroring code. This seems very complicated to me.

And while this, the current mirroring tool does the job while also keeping the parameters. And you can move the "original" object in a hidden group or something. I honestly think we shouldn't invest time in going the complicated way for a result that won't be as good anyway...

kaiwas, 2024-09-19:
> And while this, the current mirroring tool does the job while also keeping the parameters. And you can move the "original" object in a hidden group or something. I honestly think we shouldn't invest time in going the complicated way for a result that won't be as good anyway...

Maybe I don't understand all the complexity. But it seems to me that if you introduce some "global" parameter for the object, which would be on the top level and simply change the scale to negative, it would not be difficult.

After all, the current tool does the same thing. Only it saves the original object.
What if you do not need to delete the original object, but apply to it the same parameters that are created when creating a mirror copy.

Deleting and creating a new one in its place will clearly cause a problem with names. I think another way is needed.

kaiwas, 2024-09-19:
I may be wrong, but if the scale is done via a macro, then negative values ​​are possible (I am not an expert in programming languages ​​and I see that there are errors, but it is still possible). Demonstration on video.

https://github.com/user-attachments/assets/a6efdf99-3563-478e-8037-341f1e645019

Roy-043, 2024-09-20:
You can use negative scale factors for a Draft Clone (in the Property editor). In the task panel of the Draft Scale command negative scale factors are indeed not possible. This may be unintentional. Obviously zero values are problematic, but it should be possible to enter negative values.

FlachyJoe, 2024-09-20:
@kaiwa is it too easy to hide the original shape?
**RMB, Toggle visibilty** + **RMB, Toggle visibility in the tree view** results in a clean tree and 3D view but keeps the internal
[image]

kaiwas, 2024-09-20:
> is it too easy to hide the original shape?

Have you seen what house projects turn into in a tree view?
And if you need to make a mirror copy of an asymmetric house with all its thousands of subcomponents - mirror all this to create a copy?

The "mirror" operation is certainly direct design. But this operation is BASIC. You are simply looking from the point of view of habit.

Movement, rotation and scale (without additional conditions) are three basic functions EVERYWHERE!
Imagine if a copy were also created when moving (it could be hidden). Would this seem convenient to you?

You can argue for a long time about the need for this function, but I would prefer not to breed unnecessary entities that need to be hidden.

The house may have elements that are mirrored and do not need to be hidden (rafters). Others must be hidden. It is easy to get confused in all this.
And if there was such an opportunity, you could simply mirror and not think about the choice. Just do the operation and that's it.

Or mirror the drawing (not possible now).

kaiwas, 2024-09-20:
Look at this mess on the left) How can I avoid making a mistake with my choice and make this house mirrored? )))
Do you think that everyone adheres to a strict construction in the tree view and does not create unnecessary things? Monitor each component so as not to miss the mark in choosing an object for a mirror operation? Well, that's nonsense
[image]

kaiwas, 2024-09-20:
Move, rotate, scale, mirror in blender. And no unnecessary subobjects are created
[image]

FlachyJoe, 2024-09-20:
> And if you need to make a mirror copy of an asymmetric house with all its thousands of subcomponents - mirror all this to create a copy?

Create a container and mirror + hide the container.

> I just do a simple operation with one button.

What I say is one button can mirror and hide (and create the container if needed).

Quick and dirty POC macro:
```python
doc = App.ActiveDocument
sel = Gui.Selection.getSelection()
src = doc.addObject("Std::Part", "MirrorSource")
src.addObjects(sel)

import Draft
Draft.mirror(src, App.Vector(), App.Vector(1,0,0))
src.Visibility = False
src.ViewObject.ShowInTree=False
doc.recompute()
```

kaiwas, 2024-09-21:
And you are persistent )))
I don't want to create containers and hide something. If the selection of elements is large, this becomes a problem. Because in FreeCAD there are no convenient ways to select only the necessary.

And if I need to change something among the non-mirrored options (for example, add another window to the wall that I mirrored) - I need to unhide these objects again (with some inconvenient cut) Insert a new window. Mirror it already.

https://github.com/user-attachments/assets/a8414c70-ad00-4698-af9f-a71830be58da

This is nonsense.

Besides, this is ALWAYS an unnecessary action. I would like to just be able to do mirroring and not hide anything. As it happens in most programs.

Stop looking at construction design as creating a metal part with an ordered construction tree.

The test for inconvenience is quite simple.
Make many different objects in different groups.
Then mirror.
Then try to make a change (it is best to try to add another object).

kaiwas, 2024-09-21:
You can stop asking me why I need this. I just want to be able to quickly change both versions of the house (original and mirrored) simply because I need both.
People order projects in both the original and mirrored versions.

I don't need ONLY the mirrored version. I need quick access to each of their versions.

[image]
[image]

FlachyJoe, 2024-09-21:
> And you are persistent

I'm sorry for the mistake.

I still don't understand well your process:

> The house may **have elements that are mirrored and do not need to be hidden** (rafters). **Others must be hidden**. It is easy to get confused in all this.
And if there was such an opportunity, you could **simply mirror and not think about the choice**. Just do the operation and that's it.

How could FreeCAD know about which need copies and other not?

> I just want to be able to quickly change both versions of the house (original and mirrored) simply because I need both.

In this case you need the mirror AND the original as the current feature give.

> You can stop asking me why I need this.

You can request what you want but if nobody works on it your issue can remains open for years.

kaiwas, 2024-09-21:
with a destructive mirror, the freecade does not need to think about what I need to mirror.
I just select everything and do it. there and back whenever I want without thinking about a difficult choice.

there is another point in the current version. if I mirror a beam or a wall, it turns into a "mirror" object and ceases to be a beam or a wall. at least into a tree view.
in order for me to correct the object, I am forced to go to another level (open the mirror).

I understand that perhaps the tool I need will not appear soon, or maybe never. but I just want to say that the current version is not the only one, and in many situations it is also inconvenient.

later I will give more examples of why this is so.

kaiwas, 2024-09-23:
Perhaps things with mirror objects in workbenches are not so bad, part of the design. But in Draft/Bim it resembles walking through a minefield.

1. During the work process, you can do some cataloging of objects by type (try to bring order to the scene).
Site-Building-Levels and so on.
When using the mirror tool, all cataloging is broken. Objects fall out.
Video demonstration.

https://github.com/user-attachments/assets/ff4ce30a-6fff-49bc-be3c-42c3229c103c

2. When we create mirror objects, the appearance of mirror elements in the tree view is always the same (all icons are the same).
That is, if we had a wall or a beam, the icon does not inform about it. It is good if we assigned the right name - this is at least some way to see what kind of object was mirrored.

[image]

3. You need to constantly think about the correct choice of objects.
Depending on the choice, we get a different result.
Video demonstration.

https://github.com/user-attachments/assets/2db3c46e-a742-4d97-847e-8517d6fc188f

4. After the mirror tool, you may need to hide the original object. And depending on the selection (via tree view or in a 3D window, we get different results.
If you select all objects with a frame and hide them, the windows do not disappear because a double operation occurs. The wall is hidden along with the windows, and when you press the space bar, the windows are displayed again. If the house already has mirrored elements (rafters above the house), then there is also a problem with them.
In general, there is an obvious difficulty with hiding unnecessary objects.

https://github.com/user-attachments/assets/e73f2672-e721-4147-bc93-300e0323e6d8
5. If after mirroring the house I need to make changes (for example, add a window), I need to do a lot of unnecessary actions with unhiding the object, adding a window and mirroring it.
Some errors appear because I accidentally selected a building.
In general, all this resembles some kind of suffering.

Video demonstration.

https://github.com/user-attachments/assets/81936eb7-5a81-40e5-96ab-f5fe354078cf

An example from Blender where I can globally reflect an object and work with it on any itap, it did not occupy some position at the moment.
Video demonstration where I add a window constantly mirrors the whole house.

https://github.com/user-attachments/assets/7bdb4a5e-16d7-4599-a1df-123fa1cb6409

kaiwas, 2024-09-23:
Perhaps the issue would be easier to resolve if each object in the scene had a new BASE property. 
The property, like vectors, should be able to take negative values.
Sentence in the picture.

[image]