FreeCAD/FreeCAD #16587: Zoom in/out buttons for the light source cover the axes

FEA-eng, 2024-09-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The newly added Zoom in/out buttons cover the axes in Preferences --> Display --> Light Sources. You can only see the Y letter here:

[image]

P.S. Those buttons are named Push In and Push Out. Wouldn't Zoom In and Zoom Out be more intuitive?

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38728 (Git)
Build type: Release
Branch: main
Hash: 9de78e27f4148e8032594788a9804486e769388b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-09-17:
The buttons move camera, these movements are called _push in_ and _pull out_, I checked. I placed them to cover these axes, because I did not know how to get rid of them (at least, without returning the _NaviCube_ back). You might have different font settings or theme/style, so you can see a letter. =)

I don't mind if somebody improves the dialog, I just don't see room for improvement, besides hiding those axes.

FEA-eng, 2024-09-17:
> The buttons move camera, these movements are called _push in_ and _pull out_, 

Yeah, but they seem to do the same with the camera as e.g. https://wiki.freecad.org/Std_ViewZoomIn so I suggested more intuitive names.

> I checked. I placed them to cover these axes, because I did not know how to get rid of them (at least, without returning the _NaviCube_ back). You might have different font settings or theme/style, [...]

Couldn't the buttons just go to the other (left) side so that both them and the axis triad can be visible?

xtemp09, 2024-09-17:
> 
> Yeah, but they seem to do the same with the camera as e.g. https://wiki.freecad.org/Std_ViewZoomIn so I suggested more intuitive names.

Sorry, I'm not a native English speaker, so I prefer to use literature for precise terms. According to the terminology, when the camera moves closer to an object, it is referred to as "pushing in," and when it moves farther away, it is called "pulling out."[^1]

[^1]: https://www.studiobinder.com/blog/different-types-of-camera-movements-in-film/

PR #16610 (2024-09-20): [GUI] Move the Push In and Pull Out buttons to the left side
Closes #16587.

Perfection is a state in which any change can make only worse. I thought that _Light Sources dialog_ finally reached it, but @FEA-eng has found a way to improve even further. This commit moves the push in and pull out buttons to the left side of the viewport, allowing user to see the axes I deliberately covered before.

The screenshot below shows how the dialog looks now.

<details><summary>Screenshot</summary>

[image]

</details>
commit c9452e2c (2024-09-20): [GUI] Move the Push In and Pull Out buttons to the left side

Closes #16587.
commit 28bcea66 (2024-09-20): [GUI] Move the Push In and Pull Out buttons to the left side

Closes #16587.
closed 2024-09-20 by chennes