FreeCAD/FreeCAD #16589: Measurement: Segfault when using Python console with Measure tool active

echd, 2024-09-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Open empty file from start screen
2. Import STEP object
3. Click on measure icon in toolbar
4. Show the python console (View > Panels > Python console)
5. In the python console, press return at prompt

FreeCAD crashes and emits this stack trace.

> Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3d1d0) [0x700f54e4c1d0]
#1  0x700f1dab8ffc in Gui::TaskMeasure::saveObject() from /usr/lib/freecad/lib/MeasureGui.so+0x9c
#2  0x700f1dabb2a6 in Gui::TaskMeasure::apply() from /usr/lib/freecad/lib/MeasureGui.so+0x16
#3  0x700f1dabb3fe in Gui::TaskMeasure::eventFilter(QObject*, QEvent*) from /usr/lib/freecad/lib/MeasureGui.so+0x9e
#4  0x700f54944dc8 in QCoreApplicationPrivate::sendThroughApplicationEventFilters(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0xe8
#5  0x700f55afc90a in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0xda
#6  0x700f57ca1c89 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x99
#7  0x700f54945aa8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x158
#8  0x700f55179924 in QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) from /usr/lib/libQt6Gui.so.6+0x264
#9  0x700f551e7724 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Gui.so.6+0x304
#10  /usr/lib/qt6/plugins/platforms/../../../libQt6XcbQpa.so.6(+0x51b87) [0x700f4f2a7b87]
#11  /usr/lib/libglib-2.0.so.0(+0x5d299) [0x700f533cd299]
#12  /usr/lib/libglib-2.0.so.0(+0xbfec7) [0x700f5342fec7]
#13  /usr/lib/libglib-2.0.so.0(g_main_context_iteration+0x35) [0x700f533cc795]
#14  0x700f54ba82bd in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Core.so.6+0x8d
#15  0x700f5494ff66 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Core.so.6+0x1a6
#16  0x700f5494a11d in QCoreApplication::exec() from /usr/lib/libQt6Core.so.6+0x8d
#17  0x700f57baa2a6 in Gui::Application::runApplication() from /usr/lib/freecad/lib/libFreeCADGui.so+0xb46
#18  /usr/bin/FreeCAD(+0x6d0c) [0x59039e219d0c]
#19  /usr/lib/libc.so.6(+0x25e08) [0x700f54e34e08]
#20  /usr/lib/libc.so.6(__libc_start_main+0x8c) [0x700f54e34ecc]
#21  /usr/bin/FreeCAD(+0x7105) [0x59039e21a105]

### Full version info

```shell
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: makepkg
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.12.6, Qt 6.7.2, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-17:
@hlorus FYI

byteofsoren, 2024-09-17:
Is there a chance that this is related to #16564? There seams to be a null pointer in that case.

luzpaz, 2024-09-17:
@hlorus please add [About](https://wiki.freecad.org/About) info

echd, 2024-09-17:
@hlorus STEP file is not necessary. I was able to duplicate with

1. Create empty file from start screen
2. Create body from Tasks window > Start Part (the Part Design workbench comes up by default in my installation)
3. Click on measure icon in toolbar
4. Open python console (View > Panels > Python console)
5. Press enter at the python prompt

No measurement created, just the "empty" measure dialog. I do not interact in any way with the measurement tool after clicking its toolbar icon.

echd, 2024-09-17:
I found another way to get the same crash and call stack.

1. Create an empty project from the start screen
2. Switch to Part workbench
3. Add a cube
4. Select an edge on the cube
5. Click measure icon on toolbar
6. Go to python console
7. Press return at python prompt
8. Press return again at python prompt

After pressing the first return in the python console, this is printed in the console:
```
# Gui.Selection.clearSelection()
```
In all the ways that I've triggered this crash, the measurement dialog is open when I'm interacting with the python console.

echd, 2024-09-17:
After seeing a comment (https://github.com/FreeCAD/FreeCAD/issues/16591#issuecomment-2356068805) that Arch Linux users were running outdated builds, I tried the latest weekly and I was no longer able to duplicate the crash through any of the ways I previously posted.

However, while you can type in the python console, it no longer receives the return key when the measurement dialog is open which makes it hard to execute commands from it.

Version info

>OS: Fedora Linux 40.20240913.0 (Kinoite) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38768 (Git) AppImage
Build type: Release
Branch: main
Hash: 758674d40be45e365f4148fb2a5ea6bd6f93bc74
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

echd, 2024-09-17:
This is duplicate of #16401. I apologize to whoever spent time on this. I didn't remove the `is:open` filter from my issue search.

closed 2024-09-17 by echd