FreeCAD/FreeCAD #16591: Segfault after array creation

Sanaki, 2024-09-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Following [this tutorial](https://www.instructables.com/Advanced-Hexagonal-Mesh-in-Freecad/) (with some adjustments due to changed behaviors), FreeCAD segfaults regularly on attempting to reposition/zoom the view. I've created a saved file that reliably reproduces this behavior.

<details>
<summary>Crash log</summary>

```
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3d1d0) [0x7ea8c2c4c1d0]
#1  0x7ea8c604dee4 in Gui::LinkView::linkGetElementPicked(SoPickedPoint const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) const from /usr/lib/freecad/lib/libFreeCADGui.so+0x8d4
#2  0x7ea8c604f15b in Gui::ViewProviderLink::getElementPicked(SoPickedPoint const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) const from /usr/lib/freecad/lib/libFreeCADGui.so+0x12b
#3  0x7ea8c5e6234f in Gui::SoFCUnifiedSelection::getPickedList(SoHandleEventAction*, bool) const from /usr/lib/freecad/lib/libFreeCADGui.so+0x59f
#4  0x7ea8c5e66740 in Gui::SoFCUnifiedSelection::handleEvent(SoHandleEventAction*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x1f0
#5  0x7ea8c1ab537c in SoAction::traverse(SoNode*) from /usr/lib/libCoin.so.80+0x6c
#6  0x7ea8c1c8843b in SoChildList::traverse(SoAction*, int, int) from /usr/lib/libCoin.so.80+0xfb
#7  0x7ea8c1cd7486 in SoGroup::doAction(SoAction*) from /usr/lib/libCoin.so.80+0x56
#8  0x7ea8c1cee26f in SoSeparator::doAction(SoAction*) from /usr/lib/libCoin.so.80+0x2f
#9  0x7ea8c1ab537c in SoAction::traverse(SoNode*) from /usr/lib/libCoin.so.80+0x6c
#10  0x7ea8c1abc564 in SoHandleEventAction::beginTraversal(SoNode*) from /usr/lib/libCoin.so.80+0xc4
#11  0x7ea8c1aac551 in SoAction::apply(SoNode*) from /usr/lib/libCoin.so.80+0xd1
#12  0x7ea8c1ca1359 in SoEventManager::actuallyProcessEvent(SoEvent const*) from /usr/lib/libCoin.so.80+0x79
#13  0x7ea8c5f43352 in SIM::Coin3D::Quarter::QuarterWidget::processSoEvent(SoEvent const*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x22
#14  0x7ea8c5f52d4e in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::processSoEvent(SoEvent const*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x5e
#15  0x7ea8c5f6121d in Gui::NavigationStyle::processSoEvent(SoEvent const*) from /usr/lib/freecad/lib/libFreeCADGui.so+0xdd
#16  0x7ea8c5f62f2b in Gui::CADNavigationStyle::processSoEvent(SoEvent const*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x22b
#17  0x7ea8c5f6143c in Gui::NavigationStyle::processEvent(SoEvent const*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x6c
#18  0x7ea8c5f9627f in Gui::View3DInventorViewer::processSoEvent(SoEvent const*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x6f
#19  0x7ea8c5f3c985 in SIM::Coin3D::Quarter::EventFilter::eventFilter(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x165
#20  0x7ea8c2744ef8 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x118
#21  0x7ea8c38fc8bc in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x8c
#22  0x7ea8c3901c92 in QApplication::notify(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0xf32
#23  0x7ea8c5aa1c89 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x99
#24  0x7ea8c2745aa8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x158
#25  0x7ea8c38f3f9d in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /usr/lib/libQt6Widgets.so.6+0x21d
#26  /usr/lib/libQt6Widgets.so.6(+0x1635c2) [0x7ea8c39635c2]
#27  /usr/lib/libQt6Widgets.so.6(+0x164c20) [0x7ea8c3964c20]
#28  0x7ea8c38fc8cc in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x9c
#29  0x7ea8c5aa1c89 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x99
#30  0x7ea8c2745aa8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x158
#31  0x7ea8c2f78140 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /usr/lib/libQt6Gui.so.6+0x8d0
#32  0x7ea8c2fe7724 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Gui.so.6+0x304
#33  /usr/lib/libQt6Gui.so.6(+0x6041e4) [0x7ea8c34041e4]
#34  /usr/lib/libglib-2.0.so.0(+0x5d299) [0x7ea8c11cd299]
#35  /usr/lib/libglib-2.0.so.0(+0xbfec7) [0x7ea8c122fec7]
#36  /usr/lib/libglib-2.0.so.0(g_main_context_iteration+0x35) [0x7ea8c11cc795]
#37  0x7ea8c29a82bd in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Core.so.6+0x8d
#38  0x7ea8c274ff66 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Core.so.6+0x1a6
#39  0x7ea8c274a11d in QCoreApplication::exec() from /usr/lib/libQt6Core.so.6+0x8d
#40  0x7ea8c59aa2a6 in Gui::Application::runApplication() from /usr/lib/freecad/lib/libFreeCADGui.so+0xb46
#41  freecad(+0x6d0c) [0x637b190b0d0c]
#42  /usr/lib/libc.so.6(+0x25e08) [0x7ea8c2c34e08]
#43  /usr/lib/libc.so.6(__libc_start_main+0x8c) [0x7ea8c2c34ecc]
#44  freecad(+0x7105) [0x637b190b1105]
```

</details>

### Full version info

```shell
OS: Garuda Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: makepkg
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.12.6, Qt 6.7.2, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Anything else?

Unfortunately github won't allow me to upload files for some reason, so I've uploaded the file in question to mega:
[FreeCAD file to reproduce issue](https://mega.nz/file/aUADxLIA#aekGFq_539afyR67W1noRJ71bwtGLDFHI8lRIH5uDME)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-16:
Thanks for the report. I was able to attach zip to github ticket. 

> Python 3.12.6, Qt 6.7.2, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1

Just a note that some of these dependencies are some of the most up to date. Very much like Arch linux users.

luzpaz, 2024-09-17:
@Syres916 Indeed. I agree since we aren't releasing RCs fast enough and it creates a gap.

luzpaz, 2024-10-14:
@Sanaki can you test the weekly build and try to reproduce ? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Sanaki, 2024-10-26:
Apologies for the long delay. Yeah, it appears to be fixed in the latest weekly.

closed 2024-10-26 by Sanaki