FreeCAD/FreeCAD #16592: Sketcher: cannot use reference constraints in other constraints

alfrix, 2024-09-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

in versions 0.22+ it's no longer possible to use a reference constraint as the value of another constraint in the same sketch, 
this seems to be a regression to me unless it's expected but it should at least be added into the known limitations

[image]

[image]

[image]

i just realized this while doing a frame for a monitor, and i wanted the height to be calculated from the width, but the width was a reference.
i can workaround this limitation without issues, but i wanted to raise the attention to at least document this change somewhere if it's known

### Full version info

```shell
OS: Debian GNU/Linux trixie/sid (KDE/plasmawayland)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38738 (Git)
Build type: Unknown
Branch: main
Hash: 5fb2361d63a76920687d19418006c4bd18a809af
Python 3.12.6, Qt 5.15.13, Coin 4.0.2, Vtk , OCC 7.8.1
Locale: Spanish/Argentina (es_AR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Dark/Fusion
Installed mods: 
  * CurvedShapes 1.0.9
```

### Subproject(s) affected?

Sketcher

### Anything else?

for reference this already happened before:
https://forum.freecad.org/viewtopic.php?t=24574

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kaiwas, 2024-09-17:
I confirm. This is a clear regression.
Sometimes you need to make a measurement using an auxiliary constraint and use this number to calculate some element. This is an important opportunity.

FlachyJoe, 2024-09-17:
Duplicate #12655
It's seems a claimed feature added by #11716

FlachyJoe, 2024-09-17:
If really needed, references can still be hidden for circular detector with the `href()` function.

kaiwas, 2024-09-17:
I don't know why exactly the cyclic dependency is not solvable in Freakada. But for example, "web engines" have the ability to receive information in two directions.
A parent can receive data from a child, and a child can receive data from a parent.

Perhaps this is really difficult to implement, but if it were done, it would greatly expand the possibilities.
For example, it would be possible to specify "Take the height from the parent" in the child (without specifying its specific parent name). And the parameter would be received.

The same is true for this example - the reference as data exists, and it would be useful to be able to use this data to calculate something, but it seems that now there is no such possibility.

If we proceed from logical reflections on this limitation, then perhaps it would be possible to create some "data layer" that would process information (a kind of buffer not subject to the problem of cyclic dependency) and would give this information upon request.

FlachyJoe, 2024-09-17:
@kaiwas the problem is with logic:
_Length = Height + 10
Height = Length + 20_
Can you tell me _Height_ and _Length_ values?

By the way, in FreeCAD you can do this:
[image]

The expression value is calculated only at recompute.

maxwxyz, 2024-09-17:
I'm closing this in favor of https://github.com/FreeCAD/FreeCAD/issues/12655

closed 2024-09-17 by maxwxyz

kaiwas, 2024-09-17:
hahaha.
this is a great solution to the problem - just take away the ability to do something (even though it was possible before) without an alternative.
you'll forgive me, of course, but this is ridiculous.

kaiwas, 2024-09-17:
> the problem is with logic: _Length = Height + 10 Height = Length + 20_ Can you tell me _Height_ and _Length_ values?
> 
> By the way, in FreeCAD you can do this:

In your example there are two unknowns. And it is very easy to identify such a case.

But if instead of length (or height) there is a specific number entered by the user - the example turns into a regular ratio of sides (this is used everywhere in geometry).

Pythagoras would turn over in his grave from this discussion.

frank-trampe, 2024-09-17:
@Syres916, breaking the feature was a very clumsy way to solve the circular dependency problem. There are a lot of other ways to do it.

frank-trampe, 2024-09-17:
@bgbsww, Wikipedia uses the following definition of a regression.

[A software regression](https://en.wikipedia.org/wiki/Software_regression) is a type of [software bug](https://en.wikipedia.org/wiki/Software_bug) where a feature that has worked before stops working.

This feature (intra-sketch reference constraints) worked before and now does not work. That is a regression. It is no solace to people who built their models on that feature that there is a piece of documentation somewhere that suggests that the feature ought not to have existed in the first place. I am now running my own patched fork of FreeCAD to support a project that used the feature, and I cannot benefit from new features in FreeCAD without patching each new set of binaries.

I have proposed a simple fix in #12655 that safely allows intra-sketch references and even stable circular references by checking for stable convergence of the sketch solution. Look at it and tell me what you think.

bgbsww, 2024-09-17:
I am done with this issue, please stop pinging me.  If you want to submit a fix, great.  I won't be working on it.

kaiwas, 2024-09-18:
> By the way, in FreeCAD you can do this:

I have a question unrelated to this topic.
And was the "length" inscription visible in the picture made in an image editor? Or did Frecad add the ability to display names in a sketch? )