FreeCAD/FreeCAD #16595: QBackingStore::endPaint() called with active painter; did you forget to destroy it or call QPainter::end() on it?

faustocarias, 2024-09-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have had this issue when closing a sketch after changing view between task and the spreadsheet view.Profile is the sketch I was modifying. I was using the measure tool.
[image]
. 

### Full version info

```shell
QBackingStore::endPaint() called with active painter; did you forget to destroy it or call QPainter::end() on it? 

This is the error I get in the error view. Looks like is trying to recover from this error because it shows it several times.

You do a great job guys. Thank you.
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-09-17:
How many messages like this have you received? Can you reproduce it all the time? Can you apply Help → About → Copy to clipboard info?

faustocarias, 2024-09-17:
I can't remember what I was doing differently. I managed to step out of
this problem but I think it may be a memory leak problem because I had 3
part files open and after the second crash I just opened two.
I am sure that I was filling the spreadsheet with measurements while
editing the open sketch switching views without closing the sketch.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Libre
de virus.www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

maxwxyz, 2024-09-17:
@faustocarias please add full about info. Is it also happening in the stable version?

faustocarias, 2024-09-17:
I am using FreeCAD 1.0RC. I would have to try on 0.21.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Libre
de virus.www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

faustocarias, 2024-09-17:
One other thing. This is the first time I use 3D Surface tool.In this
version it affects the performance of the Adaptive operation. Is
this normal ?.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Libre
de virus.www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

luzpaz, 2024-09-17:
@faustocarias please:

1. reply directly on github (Click here to respond: https://github.com/FreeCAD/FreeCAD/issues/16595)
2. Please update your OP with your full [About](https://wiki.freecad.org/About) info.

TIA

faustocarias, 2024-09-18:
Ok.

faustocarias, 2024-09-18:
OS: Debian GNU/Linux 12 (bookworm) (XFCE/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)

faustocarias, 2024-09-18:
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)

closed 2025-08-07 by github-actions[bot]