FreeCAD/FreeCAD #16616: Unexpected pop or Start page and crash when Start page is closed

pinkavaj, 2024-09-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The Freecad exhibits unexpected behaviour by switching from current document to Start page and crashes when the Start page is closed.

Reproduction:
* start Freecad
* create new document and select it in tree view
* go to Property View dialog
* Edit any value and finish the editing by <enter>
* pres <tab> 2x -> the Start page tab becomes visible (WIERD)
* close the Start page -> crash

Note: when the original document is selected (becomes visible) and then the tab with Start page is closed, there is no crash.

### Full version info

```shell
OS: Arch Linux (KDE)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38779 (Git)
Build type: Release
Branch: makepkg
Hash: 4f83896290d8ffbab9b563872dfb281fbfcbc5af
Python 3.12.6, Qt 6.7.2, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pinkavaj, 2024-09-17:
Backtrack:

````
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3d1d0) [0x97a61d0]
#1  /usr/lib/libQt6Widgets.so.6(+0x2b3f9a) [0x81f1f9a]
#2  /usr/lib/libQt6Widgets.so.6(+0x2b4eab) [0x81f2eab]
#3  0x81f5b78 in QMdiArea::resizeEvent(QResizeEvent*) from /usr/lib/libQt6Widgets.so.6+0x298
#4  0x808b0b7 in QWidget::event(QEvent*) from /usr/lib/libQt6Widgets.so.6+0x977
#5  0x8122a78 in QFrame::event(QEvent*) from /usr/lib/libQt6Widgets.so.6+0x38
#6  0x8ff1ef8 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x118
#7  0x803a8bc in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x8c
#8  0x4f12139 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x99
#9  0x8ff2aa8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x158
#10  0x808072f in QWidgetPrivate::setGeometry_sys(int, int, int, int, bool) from /usr/lib/libQt6Widgets.so.6+0x25f
#11  0x8081a92 in QWidget::setGeometry(QRect const&) from /usr/lib/libQt6Widgets.so.6+0x1b2
#12  /usr/lib/libQt6Widgets.so.6(+0x1e371b) [0x812171b]
#13  /usr/lib/libQt6Widgets.so.6(+0x1e3d7c) [0x8121d7c]
#14  /usr/lib/libQt6Widgets.so.6(+0x2b6944) [0x81f4944]
#15  /usr/lib/libQt6Widgets.so.6(+0x2b6cb8) [0x81f4cb8]
#16  0x81f4fdb in QMdiArea::viewportEvent(QEvent*) from /usr/lib/libQt6Widgets.so.6+0x21b
#17  0x8ff1ef8 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x118
#18  0x803a8bc in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x8c
#19  0x4f12139 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x99
#20  0x8ff2aa8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x158
#21  0x820427e in QMdiSubWindow::closeEvent(QCloseEvent*) from /usr/lib/libQt6Widgets.so.6+0xde
#22  0x808abbc in QWidget::event(QEvent*) from /usr/lib/libQt6Widgets.so.6+0x47c
#23  0x803a8cc in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x9c
#24  0x4f12139 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x99
#25  0x8ff2aa8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x158
#26  0x80858e3 in QWidgetPrivate::handleClose(QWidgetPrivate::CloseMode) from /usr/lib/libQt6Widgets.so.6+0x143
#27  /usr/lib/libQt6Core.so.6(+0x1a35ea) [0x90505ea]
#28  0x824fc44 in QTabBarPrivate::closeTab() from /usr/lib/libQt6Widgets.so.6+0xd4
#29  /usr/lib/libQt6Core.so.6(+0x1a3397) [0x9050397]
#30  /usr/lib/libQt6Widgets.so.6(+0x21064b) [0x814e64b]
#31  /usr/lib/libQt6Widgets.so.6(+0x210b48) [0x814eb48]
#32  0x815b58e in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /usr/lib/libQt6Widgets.so.6+0xde
#33  0x808b376 in QWidget::event(QEvent*) from /usr/lib/libQt6Widgets.so.6+0xc36
#34  0x803a8cc in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x9c
#35  0x803fe59 in QApplication::notify(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x10f9
#36  0x4f12139 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x99
#37  0x8ff2aa8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x158
#38  0x8031f9d in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /usr/lib/libQt6Widgets.so.6+0x21d
#39  /usr/lib/libQt6Widgets.so.6(+0x1635c2) [0x80a15c2]
#40  /usr/lib/libQt6Widgets.so.6(+0x164c20) [0x80a2c20]
#41  0x803a8cc in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x9c
#42  0x4f12139 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x99
#43  0x8ff2aa8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x158
#44  0x87c5140 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /usr/lib/libQt6Gui.so.6+0x8d0
#45  0x8834724 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Gui.so.6+0x304
#46  /usr/lib/libQt6Gui.so.6(+0x6041e4) [0x8c511e4]
#47  /usr/lib/libglib-2.0.so.0(+0x5d299) [0xaeb9299]
#48  /usr/lib/libglib-2.0.so.0(+0xbfec7) [0xaf1bec7]
#49  /usr/lib/libglib-2.0.so.0(g_main_context_iteration+0x35) [0xaeb8795]
#50  0x92552bd in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Core.so.6+0x8d
#51  0x8ffcf66 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Core.so.6+0x1a6
#52  0x8ff711d in QCoreApplication::exec() from /usr/lib/libQt6Core.so.6+0x8d
#53  0x4e1feb0 in Gui::Application::runApplication() from /usr/lib/freecad/lib/libFreeCADGui.so+0xc40
#54  freecad(+0x7d7a) [0x10fd7a]
#55  /usr/lib/libc.so.6(+0x25e08) [0x978ee08]
#56  /usr/lib/libc.so.6(__libc_start_main+0x8c) [0x978eecc]
#57  freecad(+0x81e5) [0x1101e5]
```

pinkavaj, 2024-09-17:
`valgrind` output

```
==1878382== Conditional jump or move depends on uninitialised value(s)
==1878382==    at 0x81F1F98: UnknownInlinedFun (qsharedpointer_impl.h:752)
==1878382==    by 0x81F1F98: UnknownInlinedFun (qpointer.h:74)
==1878382==    by 0x81F1F98: UnknownInlinedFun (qpointer.h:82)
==1878382==    by 0x81F1F98: QMdiAreaPrivate::subWindowList(QMdiArea::WindowOrder, bool) const (qmdiarea.cpp:1372)
==1878382==    by 0x81F2EAA: QMdiAreaPrivate::rearrange(QMdi::Rearranger*) (qmdiarea.cpp:893)
==1878382==    by 0x81F5B77: UnknownInlinedFun (qmdiarea.cpp:935)
==1878382==    by 0x81F5B77: QMdiArea::resizeEvent(QResizeEvent*) (qmdiarea.cpp:2320)
==1878382==    by 0x808B0B6: QWidget::event(QEvent*) (qwidget.cpp:9195)
==1878382==    by 0x8122A77: QFrame::event(QEvent*) (qframe.cpp:521)
==1878382==    by 0x8FF1EF7: QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) (qcoreapplication.cpp:1281)
==1878382==    by 0x803A8BB: QApplicationPrivate::notify_helper(QObject*, QEvent*) (qapplication.cpp:3281)
==1878382==    by 0x4F12138: Gui::GUIApplication::notify(QObject*, QEvent*) (GuiApplication.cpp:83)
==1878382==    by 0x8FF2AA7: QCoreApplication::notifyInternal2(QObject*, QEvent*) (qcoreapplication.cpp:1142)
==1878382==    by 0x808072E: QWidgetPrivate::setGeometry_sys(int, int, int, int, bool) (qwidget.cpp:7414)
==1878382==    by 0x8081A91: QWidget::setGeometry(QRect const&) (qwidget.cpp:7302)
==1878382==    by 0x812171A: QAbstractScrollAreaPrivate::layoutChildren_helper(bool*, bool*) (qabstractscrollarea.cpp:416)
==1878382==  Uninitialised value was created by a heap allocation
==1878382==    at 0x48447A8: malloc (vg_replace_malloc.c:446)
==1878382==    by 0x9101BEF: UnknownInlinedFun (qarraydata.cpp:139)
==1878382==    by 0x9101BEF: UnknownInlinedFun (qarraydata.cpp:181)
==1878382==    by 0x9101BEF: UnknownInlinedFun (qarraydata.cpp:157)
==1878382==    by 0x9101BEF: QArrayData::allocate(QArrayData**, long long, long long, long long, QArrayData::AllocationOption) (qarraydata.cpp:201)
==1878382==    by 0x848D2C2: UnknownInlinedFun (qarraydata.h:140)
==1878382==    by 0x848D2C2: UnknownInlinedFun (qarraydatapointer.h:479)
==1878382==    by 0x848D2C2: QArrayDataPointer<QPointer<QMdiSubWindow> >::reallocateAndGrow(QArrayData::GrowthPosition, long long, QArrayDataPointer<QPointer<QMdiSubWindow> >*) [clone .constprop.0] (qarraydatapointer.h:228)
==1878382==    by 0x81F8AAE: UnknownInlinedFun (qarraydatapointer.h:209)
==1878382==    by 0x81F8AAE: void QtPrivate::QMovableArrayOps<QPointer<QMdiSubWindow> >::emplace<QPointer<QMdiSubWindow> >(long long, QPointer<QMdiSubWindow>&&) (qarraydataops.h:831)
==1878382==    by 0x81F96C3: UnknownInlinedFun (qlist.h:866)
==1878382==    by 0x81F96C3: UnknownInlinedFun (qlist.h:451)
==1878382==    by 0x81F96C3: QMdiAreaPrivate::appendChild(QMdiSubWindow*) (qmdiarea.cpp:790)
==1878382==    by 0x81F9AF0: QMdiArea::addSubWindow(QWidget*, QFlags<Qt::WindowType>) (qmdiarea.cpp:1963)
==1878382==    by 0x54C1599: Gui::MainWindow::addWindow(Gui::MDIView*) (MainWindow.cpp:1220)
==1878382==    by 0x3E1974F5: CmdStart::activated(int) (Manipulator.cpp:62)
==1878382==    by 0x4F7818B: Gui::Command::_invoke(int, bool) (Command.cpp:444)
==1878382==    by 0x4F787E6: Gui::Command::invoke(int, Gui::Command::TriggerSource) (Command.cpp:408)
==1878382==    by 0x3E182EDF: Launch (AppStartGui.cpp:86)
==1878382==    by 0x3E182EDF: operator() (AppStartGui.cpp:76)
==1878382==    by 0x3E182EDF: call (qobjectdefs_impl.h:137)
==1878382==    by 0x3E182EDF: call<QtPrivate::List<>, void> (qobjectdefs_impl.h:345)
==1878382==    by 0x3E182EDF: QtPrivate::QCallableObject<StartGui::StartLauncher::StartLauncher()::{lambda()#1}, QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) (qobjectdefs_impl.h:555)
==1878382==    by 0x9050396: UnknownInlinedFun (qobjectdefs_impl.h:469)
==1878382==    by 0x9050396: void doActivate<false>(QObject*, int, void**) (qobject.cpp:4086)
==1878382==
==1878382== Use of uninitialised value of size 8
==1878382==    at 0x81F1F9A: UnknownInlinedFun (atomic_base.h:501)
==1878382==    by 0x81F1F9A: UnknownInlinedFun (qatomic_cxx11.h:202)
==1878382==    by 0x81F1F9A: UnknownInlinedFun (qbasicatomic.h:36)
==1878382==    by 0x81F1F9A: UnknownInlinedFun (qsharedpointer_impl.h:752)
==1878382==    by 0x81F1F9A: UnknownInlinedFun (qpointer.h:74)
==1878382==    by 0x81F1F9A: UnknownInlinedFun (qpointer.h:82)
==1878382==    by 0x81F1F9A: QMdiAreaPrivate::subWindowList(QMdiArea::WindowOrder, bool) const (qmdiarea.cpp:1372)
==1878382==    by 0x81F2EAA: QMdiAreaPrivate::rearrange(QMdi::Rearranger*) (qmdiarea.cpp:893)
==1878382==    by 0x81F5B77: UnknownInlinedFun (qmdiarea.cpp:935)
==1878382==    by 0x81F5B77: QMdiArea::resizeEvent(QResizeEvent*) (qmdiarea.cpp:2320)
==1878382==    by 0x808B0B6: QWidget::event(QEvent*) (qwidget.cpp:9195)
==1878382==    by 0x8122A77: QFrame::event(QEvent*) (qframe.cpp:521)
==1878382==    by 0x8FF1EF7: QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) (qcoreapplication.cpp:1281)
==1878382==    by 0x803A8BB: QApplicationPrivate::notify_helper(QObject*, QEvent*) (qapplication.cpp:3281)
==1878382==    by 0x4F12138: Gui::GUIApplication::notify(QObject*, QEvent*) (GuiApplication.cpp:83)
==1878382==    by 0x8FF2AA7: QCoreApplication::notifyInternal2(QObject*, QEvent*) (qcoreapplication.cpp:1142)
[... cut ...]
==1878382==    by 0x3E182EDF: call (qobjectdefs_impl.h:137)
==1878382==    by 0x3E182EDF: call<QtPrivate::List<>, void> (qobjectdefs_impl.h:345)
==1878382==    by 0x3E182EDF: QtPrivate::QCallableObject<StartGui::StartLauncher::StartLauncher()::{lambda()#1}, QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) (qobjectdefs_impl.h:555)
==1878382==    by 0x9050396: UnknownInlinedFun (qobjectdefs_impl.h:469)
==1878382==    by 0x9050396: void doActivate<false>(QObject*, int, void**) (qobject.cpp:4086)
```

pinkavaj, 2024-09-17:
[valgrind-log.txt](https://github.com/user-attachments/files/17035247/valgrind-log.txt)

luzpaz, 2024-09-17:
Thanks for the super thorough bug report :+1:

pinkavaj, 2024-09-18:
> > go to Property View dialog
> 
> _Data_ tab?
> 
> > Edit any value and finish the editing by
> > pres 2x -> the Start page tab becomes visible (WIERD)
> 
> What is that? My first thought was "_press Enter twice_," but it does not give this result. "_press the second edit box(?)_", "_type "2x"_?"

Ahhh, sorry fixed. I put the key in `<..>`, but that resulted in (invalid) HTLM ...

xtemp09, 2024-09-18:
I cannot reproduce.

```ruby
OS: Ubuntu 24.04.1 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38740 (Git)
Build type: Debug
Branch: master
Hash: b4923d1c584a4dce794d91a4b090a5293569223a
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk , OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

pinkavaj, 2024-09-19:
The same code with Qt5 `5.15.15` does not crash for mee, seems Qt6 related. Can someone pls verify that?

pinkavaj, 2024-09-19:
@xtemp09 Can you pls test again with the following (updated) instructions:

* Do equivalent of `rm -rf ~/.config/FreeCAD ~/.cache/FreeCAD ~/.local/share/FreeCAD`
* Start FreeCad
* Select Dark theme in the initial screen -> click done
* Go to Edit -> Preferences -> Tree view and Propery view mode -> select Independent -> Ok
* Restart dialog pops-up -> click OK, restart the FreeCad
* start FreeCad
* Create new document
* Select the document in TreeView
* Go to Property View and start editing Label, finished the editing with Enter
* Press Tab Tab (2x tab key) -> Start page should pop-up (unexpected)
* Close the Start page -> crash

xtemp09, 2024-09-20:
@pinkavaj, still no crash.

> Start page should pop-up (unexpected)

Why on _Earth_ activation of the _Start_ page is unexpected? The tab receives keyboard focus, therefore activates, it happens eveywhere all the time.

```ruby
OS: Ubuntu 24.04.1 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38740 (Git)
Build type: Debug
Branch: master
Hash: b4923d1c584a4dce794d91a4b090a5293569223a
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk , OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

```ruby
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38728 (Git)
Build type: Release
Branch: main
Hash: 9de78e27f4148e8032594788a9804486e769388b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

xtemp09, 2024-09-20:
When Qt 6.7.3 is released, I'll try to reproduce the crash with it.

pinkavaj, 2024-09-20:
> @pinkavaj, still no crash.
> 
> > Start page should pop-up (unexpected)
> 
> Why on _Earth_ activation of the _Start_ page is unexpected? The tab receives keyboard focus, therefore activates, it happens eveywhere all the time.

Thx for verification, really seems Qt6 related.

Hmm, the activation of Start page does not hapend to me when I build with Qt5. The tab just jumps over the navigation elements, I need to press enter or click to activate them. IMHO just pressing tab key (on keyboard) should not change the active tab on window, especially not on the one who does not have focus.

Mey-be better to add screencast, gimme few minutes ...

pinkavaj, 2024-09-21:
@xtemp09 Added screencast. I did more testing and with Qt5, the Start page after pressing tab key button pop-ups and sometime not, but it never crashes and valgrind is not complaining. The bahevior depends on the panels layout

https://github.com/user-attachments/assets/a47dd906-b6f1-4eb9-927a-0584d5ffc3ad

xtemp09, 2024-09-22:
@pinkavaj, your video shows normal crashless workflow. The _Start_ page does not always show up because changing mode from _Independent_ to _Combined_ (or back) also changes MainWindow's tab order.

pinkavaj, 2024-09-22:
@xtemp09 I have update the screencast to make it clear the FreeCad crashes

* Key strokes are printed on screen
* Terminal with `freecad` command output added
* Use 1.0.0RC1 from distribution (Arch Linux) (instead of my build of `main`)

The similar behavior can be reproduced with latest `main`. There is one important difference, with my build of `main` is needed to press the Tab key 2x, with the "official" 1.0.0RC1 just pressing Enter key pop-ups the Start page. The backtrace after the try of closing the Start page is pretty the same.

https://github.com/user-attachments/assets/00238dab-d836-4c01-a353-e486465b9c5a

xtemp09, 2024-09-24:
I've been waiting for Qt 6.7.3, but its release has been postponed. So, I've decided to install 6.7.2 and, finally, fix this crash, however, I've failed to reproduce it.

```ruby
OS: Ubuntu 24.04.1 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38830 (Git)
Build type: Debug
Branch: master
Hash: b38c42262f58cf0587b627e7a866d3756da34dc4
Python 3.12.3, Qt 6.7.2, Coin 4.0.2, Vtk , OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
```

pinkavaj, 2024-09-26:
I can reproduce it for Light theme too (but not for classic).

pinkavaj, 2024-09-28:
The code bellow is a workaround for  the issue. But of course it does not make sense to wait 10 seconds to show the start page ... so just hint for debugging.

```
diff --git a/src/Mod/Start/Gui/AppStartGui.cpp b/src/Mod/Start/Gui/AppStartGui.cpp
index 561485d147..1fb5d4c57d 100644
--- a/src/Mod/Start/Gui/AppStartGui.cpp
+++ b/src/Mod/Start/Gui/AppStartGui.cpp
@@ -72,7 +72,7 @@ public:
     StartLauncher()
     {
         // QTimers don't fire until the event loop starts, which is our signal that the GUI is up
-        QTimer::singleShot(100, [this] {
+        QTimer::singleShot(10000, [this] {^M
             Launch();
         });
     }
```

pinkavaj, 2024-09-30:
For me the patch bellow "fixes" the issue. From what I have seen, I'm pretty sure it is a Qt6 bug. Proper fix/workaround MR is WIP. Confirmations/dementations wellcomed.

```
--- a/src/Gui/MainWindow.cpp
+++ b/src/Gui/MainWindow.cpp
@@ -450,7 +450,7 @@ MainWindow::MainWindow(QWidget * parent, Qt::WindowFlags f)
     d->mdiArea->setVerticalScrollBarPolicy(Qt::ScrollBarAsNeeded);
     d->mdiArea->setHorizontalScrollBarPolicy(Qt::ScrollBarAsNeeded);
     d->mdiArea->setOption(QMdiArea::DontMaximizeSubWindowOnActivation, false);
-    d->mdiArea->setActivationOrder(QMdiArea::ActivationHistoryOrder);
     d->mdiArea->setBackground(QBrush(QColor(160,160,160)));
     setCentralWidget(d->mdiArea);
```

pinkavaj, 2024-10-02:
https://bugreports.qt.io/browse/QTBUG-129596

pinkavaj, 2024-10-02:
Btw I was able to make even easier reproductioins:

* set style to Dark or Light
* close all windows
* create 2 new windows
* try close the inactive one

PR #16967 (2024-10-14): Gui: Workaround for crash on close of MDI window
Close #16616 

The Qt6 (up to Qt 6.7.3 now) contains bug, which can lead, under specific circumstances, to crash of the application, when the MDI window is closed. The circumstances are:

* at least 2 MDI windows needs to be open
* stylesheet must set different size (border, margin) for activated and un-activated tabBar tab
* the closed window must be inactive, but created before the window now active and open
* race condition must occur betwee the closing and resize event handlers for the tabBar (see qt bug for details)

So this bug only occure [... cut]
closed 2024-10-14 by chennes