FreeCAD/FreeCAD #16617: Assembly:  FC crashs when transform an activated assembly

rali717, 2024-09-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

A not activated assembly can transform in space without a problem.

But FC crashs if an **activated assembly** get transformed.

1. Activate the assembly via  mouse context menu on the assembly entry in the tree-view
2. Select the Transform function from the  mouse context menu as well
3. FC crashes when using the Std TransformManip that appeares

If the selection of  the transform tool happend via standard menu, the assembly get deactivated, and the transform can perform. But the assembly become deactiveted, like described in issue  [16238 ](https://github.com/FreeCAD/FreeCAD/issues/16238    )

In the forum the issue was confirmed [Forum entry](https://forum.freecad.org/viewtopic.php?t=90295) and other users provide additional ErrorLogFiles.

### Full version info

```shell
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38768 (Git) AppImage
Build type: Release
Branch: main
Hash: 758674d40be45e365f4148fb2a5ea6bd6f93bc74
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
```

### Subproject(s) affected?

None

### Anything else?

[Forum-Thread with Error-Log-Files ](https://forum.freecad.org/viewtopic.php?t=90295)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-10-08:
Thanks for the report, I'm checking this

PaddleStroke, 2024-10-08:
Ok so to clarify things : 
- That the assembly gets deactivated when transform tool is used, is on purpose. Transform tool requires the edit mode. So assembly cannot hold the edit mode. Which is a problem that can be discussed in https://github.com/FreeCAD/FreeCAD/issues/16238

So for this ticket, the problem that I see, is that if you activate the assembly with the right click context menu entry (instead of double clicking the assembly) then starting the transform tool does not deactivate.

Actually the bug is worse than that. If you activate with the menu entry, it is then impossible to deactivate by double click. It can be deactivated only by the menu entry.

It does not happen with App::Part.

PaddleStroke, 2024-10-08:
It is also to be noted that activating with the menu entry does not set the assembly in edit. ie parts cannot be dragged.

closed 2024-10-14 by chennes