FreeCAD/FreeCAD #16620: Stylesheet: Addon Manager link on first setup hard to read in dark theme

Nashenas88, 2024-09-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When selecting the Dark theme on the first start of the app, the Addon Manager link is very hard to read.
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.60e
  * Assembly3 0.11.4
  * Assembly4 0.12.4
  * fcgear 1.0.0
  * kicadStepUpMod 10.16.5
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-17:
Would you mind testing bleeding edge build to see if you can reproduce ?  
(https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

Syres916, 2024-09-17:
An attempt has been made to [fix](https://github.com/FreeCAD/FreeCAD/pull/15729) it but I'm struggling with some 'special' behaviour of the Start module, I'll have a fresh look later in the week but please note it won't be merged until Post 1.0 due to the translations.

luzpaz, 2025-01-27:
Looks like the PR fell through. 

@MisterMakerNL, in Syres's PR you wrote:

> > > It will be confusing if we have 2 link colors.
> > 
> > So you want it to be 1907817471 (#71b6fb) ?
> 
> I think the best way to do this is too do it through the stylesheet where we can link the 2.

What does that look like ?

MisterMakerNL, 2025-01-27:
@luzpaz Best is too make it follow: QLabel[haslink="true"], but I don't mind if we have to make separate object in the stylesheets. Although for the sake of keeping the code simple it would be nice to use available options.

FEA-eng, 2025-10-17:
Still relevant

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/10/14 15:14:51
Build type: Release
Branch: main
Hash: 41ce37233d64c81c2f105de085635d4dcd669943
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
Installed mods: 
  * A2plus 0.4.68
  * freecad.gears 1.3.0
```

FEA-eng, 2026-04-27:
Initially, it's bad:

[image]

But after changing the default language, it improves for some reason:

[image]

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.2.0dev.20260421 (Git shallow)
Build date: 2026/04/21 23:21:45
Build type: Release
Branch: (HEAD detached at 449725202)
Hash: 449725202d0ab581516aacad59d62cdce80db82a
Python 3.11.15, Qt 6.8.3, Coin 4.0.8, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

luzpaz, 2026-09-22:
@maxwxyz does https://github.com/FreeCAD/FreeCAD/pull/32827 impact this ticket ?

closed 2026-09-22 by kadet1090