FreeCAD/FreeCAD #16623: [Feature] Multi-pass profile operations

davidgilkaufman, 2024-09-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I want to profile a contour, but there is too much material left to remove in all in one pass, so I need to do multiple profile passes. Other software makes this easy; FreeCAD should also have an easy way to generate these successive offset profile passes. I propose adding "number of passes" (integer) and "pass step over" (formula) parameters to the profile operation UI to control generation of multiple successive profile passes. By default, "number of passes" will be 1, and "pass step over" will be disabled (auto enables for larger "number of passes"; default value OpToolDiameter)

### Full version info

```shell
Version 0.21.2-24
Revision number 33771 (git)
Release date 2023/11/14 02:33:39
Operating system Arch Linux
Word size 64-bit
Branch makepkg
Hash b9bfa5c
```

### Subproject(s) affected?

CAM

### Anything else?

I intend to implement this feature myself.

I started a [forum thread](https://forum.freecad.org/viewtopic.php?t=90247) with a detailed description of the problem and a screenshot of the desired paths (and the workaround I used to generate them).

@spanner888 found an [older thread](https://forum.freecad.org/viewtopic.php?t=54497) about a prior attempt to implement a version of this feature, and disagreement over when and how it should be done. As best I can tell, this ultimately resulted in a macro, but nothing integrated into the profile operation.

The "when" part of this disagreement I believe is now deprecated, since much time has passed, but I'd like to address some comments from [a post](https://forum.freecad.org/viewtopic.php?p=472412#p472412) by @mlampert in that thread. @mlampert objected to the feature saying that a profile operation is a line-following operation, and that the proposed feature changed it to an area-clearing operation, and that this is the domain of a pocket operation and not a profile operation.

I'm not intimately familiar with the prior attempt to implement this feature, but in my opinion, the version I am proposing here should not be subject to this objection. The intended use case of this feature is to generate a small number of profile passes -- each one a line-following operation -- with the goal of profiling a surface without cutting too much material in a single pass. This use case is important and not well served by FreeCAD's current features, and imo even with multiple passes it is still correct to think of it as a line-following profile operation. I would agree that a user could pick a very large number of passes to use this feature to cut an area, but I would call this an abuse/misuse of the feature, not an overreach of the profile operation into the land of pocketing.

For additional context, I want to refer to [Fusion 360's Contour 2D operation](https://help.autodesk.com/view/fusion360/ENU/?guid=GUID75B6821B-DE26-4E3B-AF10-4A54131CD9E4), which has a "passes" tab, with controls for the number of passes and the step over. I believe it is standard to include these features in a contour-following operation.

I will post the contents of this issue at the end of my [forum thread](https://forum.freecad.org/viewtopic.php?t=90247) for further discussion -- please discuss it there.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

GeneGH, 2024-09-18:
This would be a very useful expansion of the profile operation. I have the same concern about chewing too much metal at one time or creating "slotting" action in stock corners, for example.

The model reference remains a single edge, similar to profile, not an area which is similar to pocket.

The distinction between the names "profile" and "area-clearing" is only semantics. Call it whatever gets the most votes.

luvtofish, 2024-12-01:
I concur with the need for a multi pass profiling step over capability. Many times the stock in use is larger than the finished part which forces an undesirable slot milling operation during the profile op.

luzpaz, 2024-12-03:
@luvtofish any chance you can test #17326 ?

luvtofish, 2024-12-03:
I’m downloading his PR now. Will compile and test the multi pass functionality

luvtofish, 2024-12-04:
I successfully compiled and tested the multi-pass profile. It worked for me as expected but I noticed a couple things I wish to share. The option to “use start point” didn’t seem to effect the op starting point? I will test this again today to make sure I wasn’t missing something. The second thing is more of an ask. I didn’t see an option for “keep tool down” which changes the behavior of the profiling by not raising the tool at the end of every circuit around the part.  Other than these observations, this functionality is sorely needed in CAM. I’d be fine with merging this now with further refinements added down the road as needed.

luzpaz, 2024-12-04:
@davidgilkaufman :point_up_2:

davidgilkaufman, 2024-12-04:
Thanks Dan! I want to merge it as is, to be refined later. Some quick notes for that refinement, while it's fresh in my head:

I don't think I've written any code that should affect "use start point". My best guess is that the functionality of "use start point" is somehow dependent on path sorting options, and I disabled path sorting on multipass profile operations to ensure that passes are done in the correct order. Tbh I've never gotten start points to work right for me as a user and I don't know how they're implemented either, so I want to leave this issue for later.

"Keep tool down" sounds like a good feature. I imagine that implementing it will involve manually generating g-code to merge consecutive profile passes, which incidentally will allow for turning path sorting back on, and perhaps make "use start point" work as expected again.

luvtofish, 2024-12-05:
Use start point seems to force all tool ops to begin near the part origin. It has the benefit of minimizing unnecessary travel. The options “use start point” and “keep tool down” aren’t strictly necessary but due optimize and reduce axis travel.

dresco, 2024-12-06:
I've just built & tested as well. It does work as expected for me also, with that one caveat of ignoring the start point.

I had a look at the changes, and it's [disabling the path sorting](https://github.com/davidgilkaufman/FreeCAD/blob/13a6b44029cc9bd00f1314bf2b21d7f8600b181b/src/Mod/CAM/Path/Op/Profile.py#L417-L419) that causes it..

Attaching an example of where this is a problem. It's an inside profile with a LeadIn/Out dressup, if this starts in a corner (as it does by default) then the dressup moves cut into the stock. The attached file was saved with the path sorting left in place, but refreshing the op (with the code from the PR) will remove the start point.

[multipass_test.zip](https://github.com/user-attachments/files/18041459/multipass_test.zip)

davidgilkaufman, 2024-12-09:
I spent my morning reading the implementation of "use start point" and my conclusion is that I definitely do not want to extend its implementation to allow for changing the start point of individual closed wires without sorting all wires in this PR. As a brief summary, there is [a clear point](https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/CAM/App/Area.cpp#L3538) where `sortWires` checks for `SortModeNone`, does a small amount of processing on those wires anyway, and exits. All handling of the start point (`_pstart` in that code) occurs after that point, interwoven with sorting functionality. The [separate function](https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/CAM/App/Area.cpp#L3269) (with the same name, `sortWires`) that is actually invoked to change a wire's start point depends on and edits internal state that is clearly intended to be used in reordering wires. I think it is likely possible to reuse this functionality to change the start point without also sorting all wires, but it will require great care and understanding to tease apart these functions and remain confident that the separation does not produce any unexpected behavior. I do not currently have the requisite level of understanding. This is not a small, safe, or easy change. (Frankly I find it terrifying that the start points of generated paths are modified at all after generation, rather than generating them starting from the desired location using the original generation code that actually understands the meaning of the path it's working with, and I'd be very nervous about touching this code.)

My recommendation for enabling the use of start points with multipass profiles is to implement the "keep tool down" feature by generating connecting paths (potentially keeping the tool down) between successive passes, so the full set of passes can be exported in order as a single wire. I believe that this will prevent the sort function from mis-ordering consecutive passes, and thus allow for re-enabling sorting on mulitpass profile operations, "fixing" the "use start point" feature. As I mentioned in my last message, I do not want to do this task as part of this PR either.

Jon, I revalidated my PR, and in its current state, if you open a profile operation created with an older version of freecad and recompute that operation, it generates the same result -- sorting is still enabled (and thus start points are as well). Sorting is only disabled if you have configured the profile operation to generate multiple passes, which must be done in my branch. I suspect you produced your example by:

1. In my branch, create a multipass profile operation
2. Revert to an older version of freecad; recompute the profile operation and add a lead in/out dressup to it. Notice that there are no longer multiple passes on that operation, but ignore that problem
3. Go forward again to my branch. Now, as you say, if you recompute you will get multiple passes, starting from a different start point, with the dressup applied, and that the new start point for the dressup could be an unacceptable location.

This problem actually has nothing to do with the "use start point" feature not working with multiple passes, and everything to do with the fact that you used a new feature and then made further edits to the file with an old version of freecad that doesn't understand that feature. While we should try out best to ensure that upgrading freecad does not change paths for the worse, making edits to a file with an old version of freecad after enabling new features in that file is always an unsupported action and cannot be protected against.

For example, imagine that the "use start point" feature worked the same on both branches, or was otherwise not in effect. It is still expected behavior that the start point of a multipass profile operation changes when it is recomputed on an older version of freecad that doesn't include the multipass feature! Specifically, the start point will be closer to the shape being profiled on older version of freecad, and farther offset due to the additional passes in newer versions of freecad. I can create example files where this change in start point causes a lead in/out dressup to begin in an unacceptable location while changing freecad versions in either direction -- from old to new (on a new file with multipass already in use), as you did, or from simply new to old. This problem is intrinsic in changing the specification on where a path should start (which is a good and necessary thing for doing profiling with multiple passes!) and cannot be avoided. Implementing new "use start point" functionality would not fix this problem.

In short, I think Jon's example cannot be constructed by a user merely upgrading freecad and continuing their work, and requires unusual, unsupported user actions that cannot be protected against.

I would still like to merge as is.

dresco, 2024-12-09:
> I spent my morning reading the implementation of "use start point" and my conclusion is that I definitely do not want to extend its implementation.

Understood.

> I suspect you produced your example by:

No even simpler. I was trying to identify why "use start point" was failing for multi-pass operations, and so re-enabled sorting which brought it back.. Apologies if that wasn't clear from previous comment.

> I would still like to merge as is.

No disagreement here. The lack of start point with unsorted paths is an existing issue. As I looked into it some more, I noticed this behaviour can also be observed in pocket operations - as certain patterns (offset, zigzag offset) disable sorting..

I was going to propose I raise a new feature request to support start points with unsorted paths, but your suggestion above sounds cleaner.

Thanks for your efforts on this!

maxwxyz, 2024-12-16:
Is this issue fixed by https://github.com/FreeCAD/FreeCAD/pull/17326 ?

davidgilkaufman, 2024-12-19:
Yes! This is done now.

closed 2024-12-19 by davidgilkaufman