FreeCAD/FreeCAD #16625: Crash during Check Geometry using IsMicroEdge()

schmidtw, 2024-09-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

During Part Design -> Check Geometry when I have `Too small edge` enabled for the entire tree selected with `ctl-a` the program crashes.  It happens reliably & does not happen when I disable the `Too small edge`.

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x40d00) [0x7fd0d924fd00]
#1  0x7fd09eb55206 in BRep_Tool::Parameter(TopoDS_Vertex const&, TopoDS_Edge const&, double&) from /tmp/.mount_FreeCAPK1dIk/usr/lib/./libTKBRep.so.7.7+0x266
#2  0x7fd09eb5597f in BRep_Tool::Parameter(TopoDS_Vertex const&, TopoDS_Edge const&) from /tmp/.mount_FreeCAPK1dIk/usr/lib/./libTKBRep.so.7.7+0x1f
#3  0x7fd06f83411e in BOPTools_AlgoTools::IsMicroEdge(TopoDS_Edge const&, opencascade::handle<IntTools_Context> const&, bool) from /tmp/.mount_FreeCAPK1dIk/usr/lib/././libTKBO.so.7.7+0x12e
#4  0x7fd06f71c6c1 in BOPAlgo_ArgumentAnalyzer::TestSmallEdge() from /tmp/.mount_FreeCAPK1dIk/usr/lib/././libTKBO.so.7.7+0x311
#5  0x7fd06f721b25 in BOPAlgo_ArgumentAnalyzer::Perform(Message_ProgressRange const&) from /tmp/.mount_FreeCAPK1dIk/usr/lib/././libTKBO.so.7.7+0x265
#6  0x7fd04866c04c in PartGui::TaskCheckGeometryResults::goBOPSingleCheck(TopoDS_Shape const&, PartGui::ResultEntry*, QString const&, Message_ProgressScope const&) from /tmp/.mount_FreeCAPK1dIk/usr/lib/./PartGui.so+0x60c
#7  0x7fd04866dc15 in PartGui::TaskCheckGeometryResults::goCheck() from /tmp/.mount_FreeCAPK1dIk/usr/lib/./PartGui.so+0xe65
```

### Full version info

```shell
OS: Fedora Linux 40 (Workstation Edition) (GNOME/gnome-xorg)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38768 (Git) AppImage
Build type: Release
Branch: main
Hash: 758674d40be45e365f4148fb2a5ea6bd6f93bc74
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * sheetmetal 0.4.25
  * lattice2 1.0.0
```

### Subproject(s) affected?

None

### Anything else?

Attached is the file that causes the crash.

[crash.tar.gz](https://github.com/user-attachments/files/17039690/crash.tar.gz)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-09-18:
With all checkboxes, except the 1st, in the Settings checked: no crash and no errors here. I have checked the Body.
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38728 (Git)
Build type: Release
Branch: main
Hash: 9de78e27f4148e8032594788a9804486e769388b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

MarcusMWilliams, 2024-09-18:
It's Real retirement?

mwganson, 2024-09-28:
I cannot reproduce a crash.  Note that you should not be attempting to check the spreadsheet or the origin, but this should still not cause a crash because checkgeometry skips null shapes.

mwganson, 2024-09-28:
After more investigation I was able to reproduce a crash when expanding the Origin object and attempting to check the X-axis object.  I will submit a PR to skip checking these objects today or tomorrow.

closed 2024-09-30 by yorikvanhavre