FreeCAD/FreeCAD #16628: Document corruption: All View provider Information lost when saving a document in console mode

AIRCAP, 2024-09-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When loading a document without GUI (for example when running freecadcmd or freecad -c) the App objects are loaded, but the Gui objects are not (obviously) - this means the objects have Proxies but not ViewObject's and also no ViewProviders.

If the document is then saved again (for example with App.Activedocument.save()  after making changes from a macro, the document gets saved with no view provider information in it.

When opening it with GUI subsequently, the ability to edit/modify the document is severely hampered, as the proxy in all ViewObject's is now "1".

### Full version info

```shell
[code]
OS: Ubuntu 20.04.6 LTS (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * CurvedShapes 1.0.9
  * CfdOF 1.27.6
  * Plot 2022.4.17
  * freecad.gears 1.2.0
  * AirPlaneDesign 0.4.1
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

Core

### Anything else?

Reproduction:

1. Take any existing or new Freecad document with objects in it
2. Open the document with "freecad -c" in console mode
3. save the document by entering "App.ActiveDocument.save()"
4. Reopen the document in freecad

Expected Result: 
No changes to the document

Actual result:
Document loses all ViewProvider information and can no longer be edited

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-20:
Have you tested in latest 1.0RC1 ? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/1.0rc1

AIRCAP, 2024-09-22:
Cannot reproduce in dev.
I just tested rc1 - in rc1 the problem does not occur, the parts in the part tree remain the type of parts they are supposed to be. So I think this can be marked closed/fixed

closed 2024-09-22 by luzpaz