FreeCAD/FreeCAD #16633: Background of toolbar area has a black stripe

kpemartin, 2024-09-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The background of the toolbar area just below the main menu has a black stripe, regardless of which toolbar(s) occupy that location.

Note that in the attached screenshot the fact that the length of the black stripe seems to match the width of the File toolbar area is mereley coincidence. Placing other toolbars there does not change the size of the stripe.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38768 (Git)
Build type: Release
Branch: master
Hash: 758674d40be45e365f4148fb2a5ea6bd6f93bc74
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.0
Locale: English/Canada (en_CA)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/windowsvista
```

### Subproject(s) affected?

None

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-18:
> Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/windowsvista

Nice. Looks like adding this data to the About info is paying off. Could you change the Qt theme to 'Fusion' and test ? I think the method is in #15868

kpemartin, 2024-09-19:
I looked at #15868 and promptly got lost.

I'm not sure if it is a valid experiment, but if I go to Preferences->Display->UI and in "Style sheet (advanced)" I select either "FreeCAD Light" or "FreeCAD Dark" instead of "No style sheet" and Apply it the black bar goes away but then because of the choice of background and border colours I have a very hard time making out the edges of the various panes on the screen (the latter observation not being relevant to this Issue).

Syres916, 2024-09-28:
@kpemartin the black stripe/rectangle fix is in https://github.com/FreeCAD/FreeCAD/commit/6e2cd4e73303f817e4bf5e908abe39030b7c5fb3

luzpaz, 2024-09-28:
Just for reference, here is the ticket related to the commit: 
#16863

kpemartin, 2025-03-31:
A lot of water under the bridge since then, I'll see if I can reproduce it. I'll have to figure out what my pre-workaround situation was, I think I had no style sheet selected.

kpemartin, 2025-03-31:
It appears to be fixed, at least I (think I have) reset to my preferences at the time and the problem no longer occurs in the 1.0 release version