FreeCAD/FreeCAD #16634: segmentation fault after accepting toponaming recompute dialog in file with multiple links

adrianinsaval, 2024-09-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Files are open source available at https://github.com/GliaX/tourniquet download and open `injection_molding/Glia Tourniquet Inj-Mold Design (FreeCAD)/Assembly.FCStd` in freecad, click yes when prompted to recompute after a few seconds recomputing freecad crashes:
```
FreeCAD 1.1.0, Libs: 1.1.0devR38768 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Sheet Metal workbench loaded
QFont::fromString: Invalid description 'Inter,10,-1,5,400,0,0,0,0,0,0,0,0,0,0,1'
QFont::fromString: Invalid description 'Inter,10,-1,5,400,0,0,0,0,0,0,0,0,0,0,1'
QFont::fromString: Invalid description 'Inter,10,-1,5,400,0,0,0,0,0,0,0,0,0,0,1'
During initialization the error "No module named 'requests'" occurred in /home/adrian/.local/share/FreeCAD/Mod/Ondsel-Lens/./InitGui.py
Please look into the log file for further information
Transformed: Result has multiple solids. Only keeping the first.
PartDesign::Body: Link(s) to object(s) 'Body001' go out of the allowed scope 'Body002'. Instead, the linked object(s) reside within 'N/A'.
PartDesign::Body: Link(s) to object(s) 'Binder' go out of the allowed scope 'Body002'. Instead, the linked object(s) reside within 'Part Part Link'.
App::DocumentObjectGroup: Link(s) to object(s) 'Body002' go out of the allowed scope 'Group'. Instead, the linked object(s) reside within 'Part Part Link Part Part Link'.
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45320) [0x74d7bec45320]
#1  /usr/lib/freecad-daily-python3/lib/Sketcher.so(+0x1cd28a) [0x74d7761cd28a]
#2  0x74d7760dba46 in Sketcher::SketchObject::rebuildExternalGeometry(bool, bool) from /usr/lib/freecad-daily-python3/lib/Sketcher.so+0xf76
#3  0x74d7760ae784 in Sketcher::SketchObject::execute() from /usr/lib/freecad-daily-python3/lib/Sketcher.so+0x74
#4  0x74d7c19f62a4 in App::DocumentObject::recompute() from /usr/lib/freecad-daily-python3/lib/libFreeCADApp.so+0x44
#5  0x74d78846f170 in Part::Feature::recompute() from /usr/lib/freecad-daily-python3/lib/Part.so+0x10
#6  0x74d7c19b531c in App::Document::_recomputeFeature(App::DocumentObject*) from /usr/lib/freecad-daily-python3/lib/libFreeCADApp.so+0x26c
#7  0x74d7c19b193e in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /usr/lib/freecad-daily-python3/lib/libFreeCADApp.so+0x5fe
#8  0x74d7c2260739 in Gui::Application::slotFinishRestoreDocument(App::Document const&) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x389
#9  /usr/lib/freecad-daily-python3/lib/libFreeCADApp.so(+0x1c1cd8) [0x74d7c19c1cd8]
#10  0x74d7c19b1291 in App::Document::afterRestore(bool) from /usr/lib/freecad-daily-python3/lib/libFreeCADApp.so+0x331
#11  0x74d7c1b49d75 in App::Application::openDocuments(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >*, bool) from /usr/lib/freecad-daily-python3/lib/libFreeCADApp.so+0x1e55
#12  0x74d7c1b4a9b0 in App::Application::openDocument(char const*, bool) from /usr/lib/freecad-daily-python3/lib/libFreeCADApp.so+0x130
#13  0x74d7c1b6872b in App::Application::sOpenDocument(_object*, _object*, _object*) from /usr/lib/freecad-daily-python3/lib/libFreeCADApp.so+0xdb
#14  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(+0x1dbe88) [0x74d7c0bdbe88]
#15  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(_PyObject_MakeTpCall+0x8f) [0x74d7c0b827af]
#16  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(_PyEval_EvalFrameDefault+0x40ee) [0x74d7c0b1d5ee]
#17  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(PyEval_EvalCode+0x20f) [0x74d7c0c9d1ff]
#18  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(+0x2f7af8) [0x74d7c0cf7af8]
#19  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(+0x2f7c1c) [0x74d7c0cf7c1c]
#20  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(PyRun_StringFlags+0x74) [0x74d7c0cfb234]
#21  0x74d7c14f69fe in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /usr/lib/freecad-daily-python3/lib/libFreeCADBase.so+0x6e
#22  0x74d7c232590a in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x9a
#23  0x74d7c2325aad in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0xed
#24  0x74d7c225a896 in Gui::Application::open(char const*, char const*) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x846
#25  0x74d7c2330ec2 in StdCmdOpen::activated(int) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x672
#26  0x74d7c2325433 in Gui::Command::_invoke(int, bool) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x243
#27  0x74d7c23257f6 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x136
#28  /lib/x86_64-linux-gnu/libQt5Core.so.5(+0x312e16) [0x74d7bf712e16]
#29  0x74d7c0364f94 in QAction::triggered(bool) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x44
#30  0x74d7c0367eab in QAction::activate(QAction::ActionEvent) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xbb
#31  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x26967a) [0x74d7c046967a]
#32  0x74d7c04697b8 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xf8
#33  0x74d7c056d446 in QToolButton::mouseReleaseEvent(QMouseEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x16
#34  0x74d7c03b0df8 in QWidget::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x278
#35  0x74d7c036bd45 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x85
#36  0x74d7c03746b0 in QApplication::notify(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1290
#37  0x74d7c22f14d0 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x90
#38  0x74d7bf6d8118 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x128
#39  0x74d7c0372874 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1d4
#40  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1caa39) [0x74d7c03caa39]
#41  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1cdfbf) [0x74d7c03cdfbf]
#42  0x74d7c036bd45 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x85
#43  0x74d7c22f14d0 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x90
#44  0x74d7bf6d8118 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x128
#45  0x74d7bfb45a3b in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x80b
#46  0x74d7bfb17bfc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0xac
[... 22 lines cut ...]
Build type: Release
Branch: main
Hash: 758674d40be45e365f4148fb2a5ea6bd6f93bc74
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.1
Locale: C/Default (C) [ OS: English/United Kingdom (en_GB) ]
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * fasteners 0.5.27
  * Curves 0.6.45
  * Defeaturing 1.2.2
  * CurvedShapes 1.0.9
  * AirPlaneDesign
  * sheetmetal 0.4.24
  * OpenTheme 2024.8.17
  * Ondsel-Lens 2024.7.5.02
  * CfdOF 1.27.5
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-18:
cc @bgbsww

bgbsww, 2024-09-18:
We are getting a Sigabrt from this assertion:

```
void SketchObject::rebuildExternalGeometry(bool defining, bool addIntersection)
{
    Base::StateLocker lock(managedoperation, true); // no need to check input data validity as this is an sketchobject managed operation.

    // get the actual lists of the externals
    auto Objects     = ExternalGeometry.getValues();
    auto SubElements = ExternalGeometry.getSubValues();
    assert(externalGeoRef.size() == Objects.size());
```
with 0 externalGeoRef and 1 Object  during the recompute of  the "/home/brad/x5/tourniquet-master/injection_molding/Glia Tourniquet Inj-Mold Design (FreeCAD)/sliderAssy.FCStd" Document.

Opening just that document and attempting an on load recompute also crashes.  However, opening it without the recompute and then recomputing succeeds.

Opening the clip and parameters documents  first and the opening the sliderAssy still crashes.

The quick facile way to solve this would likely to be to modify this code:

https://github.com/bgbsww/FreeCAD/blob/914a7616f7490973434bb086b76d3f730b6a905f/src/Gui/Application.cpp#L982-L990

To only act on the single document loaded, instead of trying to do all the others.  That may have been a mistake to bring in.

bgbsww, 2024-09-18:
> Also, like I said in RC1 the model has errors on a recompute (I haven't looked deeper into it or tried to fix them), should that be considered a regression? if so, is that also a blocker or an acceptable side effect of toponaming?

Without investigation, it's tough to tell.  When I look, I see 2 RuledSurface errors, each of which involves two binders. both inside the A and B plates document.  Those are execute() giving up, and it isn't immediately clear that that is toponaming, although it may be.

adrianinsaval, 2024-09-18:
> To only act on the single document loaded, instead of trying to do all the others. That may have been a mistake to bring in.

if you do this, what should be the workflow for an user that wants to port his files to 1.0? Asking as user, this is way above my head so I won't pretend o know what's the best course of action

bgbsww, 2024-09-18:
Yeah, that's exactly why I'm hesitating, investigating and hoping for maybe input from others.  I'll try to answer that question in a bit, I'm still looking at pinning down the RuledSurface thing so we know where to send it.

bgbsww, 2024-09-18:
RuledSurface recompute fail is https://github.com/bgbsww/FreeCAD/blob/914a7616f7490973434bb086b76d3f730b6a905f/src/Mod/Part/App/TopoShapeExpansion.cpp#L2164-L2167  and that restriction did not exist in the original non-element map code;  maybe https://github.com/FreeCAD/FreeCAD/pull/12431 is the moment we went south? 

So I think that recompute issue is a separate, regression bug specific to ruled surfaces ( there may be others ).  That should get filed.

I'm on to the overall recompute on load question now ...

bgbsww, 2024-09-18:
... And the answer is that you will get the recompute dialog multiple times.  Specifically, once for every sub document that is loaded, which could be a lot and very annoying to the user.  A better solution may be required here.

adrianinsaval, 2024-09-18:
> ... And the answer is that you will get the recompute dialog multiple times. Specifically, once for every sub document that is loaded, which could be a lot and very annoying to the user. A better solution may be required here.

yes to all button?

bgbsww, 2024-09-18:
Implementation of same is turning out to be non trivial.  You can't do it the scope of just the signal from the top level document load, you need to wait until all the documents are in and then do it, or else you risk cross document references breaking, like in this case.  I'm working on it though.

I was just going to assume that if you answer Yes once to the already plural description, we should do all subdocuments.  That's what LS3 does, and if you only want to recompute the top level file, you can 'No' and then do that yourself.

bgbsww, 2024-09-18:
Ruled Surface issue is found.  An issue with Slices has surfaced that I'm looking for, and then Pads.

Good test document set this is.

bgbsww, 2024-09-19:
Okay, the Pad issue seems to have gone away after fixing the Ruled Surface one.

There is a remaining error showing up in Slice_child3 and Slice_child4 - Compound Filters of a Slice that I can't understand enough to try to address.  I'm not sure if there's a modeling problem here or a a code one.

@adrianinsaval any clues you can give?

closed 2024-09-19 by chennes

adrianinsaval, 2024-09-28:
@bgbsww sorry I'm just now getting back into this, from an initial look, it seems the slices aren't the actual problem, the problem is that `Fusion` AKA `shot + slide` in file `A and B Plates` isn't actually fusing it's two components. The sliced compound filter then is expected to fail since some slices will be missing
this is `shot + slide` in 0.21.2:
[image]
and in 1.1.0.38834:
[image]

all of the solid representing the plastic shot is missing

bgbsww, 2024-09-28:
Thanks!  It all starts to make more sense.  Turning off Refine on Shot appears to make the fusion work.  Still an issue with the slice of the buckle that remains though.

We have stra

bgbsww, 2024-09-29:
A test case with crossed AdditivePipes with Cones on the ends of them works perfectly.
[image]

bgbsww, 2024-09-29:
Oh!  It is recompute order dependent.  If you don't accept the overall recompute on opening shot, and then ~recompute Runner system first, everything is fine.~  If you recompute Binder or Shot first, there are problems.  

And the naming fooled me - that's a PD Cone and not a Part Cone.

No that happened once.  So not repeatable.   But that was using revert to disk.  And the sub files were still open.

Fresh session.   Turn off visibility of Shot and Runner System.  Make just "sprue and runners" visible.  turn off visibility,  Recompute "sprue and runners",  turn on visibility of Runner System and off the sprue and runners.   Recompute each of the MasterGates.  then "Runner System".  Now everything is good.  But open Sketch and close it and everything is broken again.

Recompute of Runner system continues to mess up the length and end of sprue going to Windlass.  That seems to be in Sketch,  where the construction geo 9-Line is controlled by Constraint27 which is an expression driven by <<parameters>>.WindlassPosition.  But Master Gate001 has a position that is not controlled by expression.  Deactivating Constraint27 solves this issue - the length of the runner is then correct, so this expression is wrong.

But ultimately this comes down to the calculation of the Cone into sprue & runners - Body011.

If you change the attachment of the Cone from Deactivated to XY on plane using XY_Plane012 to allow an attachment offset, and then change the attachment offset z to 0.1 then everything starts working, so this is really about the computation of the AdditiveShape AdditiveShape001 and Cone coming together.

Let me see if I can successfully add this to the test...  Well, the recompute of the cone fails if the radius if it interferes with the ends of the pipes, so that's a start.

bgbsww, 2024-09-29:
But fundamentally no, that test won't break.  However, on thinking, the Sprue and runners do recalculate just fine, it's the Runner System where it starts to go wrong, and the runner that goes wrong uses Master Gate, which is the only gate that is attached to the end of the additive pipes.  So does the combination of the two cause the rendering failure on the segment of additive pipe?

bgbsww, 2024-09-29:
Okay, I duplicated it.  Not yet sure how to fix it.  The key here is that both pipes must have their "seam" from the closure point of the circle on the "side" of the pipe and then the cone must stop exactly at that seam.  And you must be refining the cone.  Disable the refine, or Move the cone a um, move the rotation of the pipe a fraction of a degree, and everything fixes.  That's the replication part.  Anyone ideas on the fix part?
[image]

adrianinsaval, 2024-09-29:
is the pre toponaming code still around and buildable in main? it could be worth checking if this isn't a regression in occt rather than freecad

PR #16647 (2024-09-19): Core: Move the recompute dialog to after Gui opens not on per doc signal
Fix #16634