FreeCAD/FreeCAD #16635: windows:1.0 rc1 using visial studio 2019 as the generator, then using CMake to build with an error report

yan695432, 2024-09-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

CMake Error at D:/WorkArea/FreeCad/libPack/LibPack-1.0.0-v3.0.0-Release/lib/cmake/Boost-1.85.0/BoostConfig.cmake:141 (find_package):
  Found package configuration file:

    D:/WorkArea/FreeCad/libPack/LibPack-1.0.0-v3.0.0-Release/lib/cmake/boost_filesystem-1.85.0/boost_filesystem-config.cmake

  but it set boost_filesystem_FOUND to FALSE so package "boost_filesystem" is
  considered to be NOT FOUND.  Reason given by package:

  No suitable build variant has been found.

  The following variants have been tried and rejected:

  * libboost_filesystem-vc143-mt-s-x64-1_85.lib (vc143, detected vc142, set
  Boost_COMPILER to override)

  * boost_filesystem-vc143-mt-x64-1_85.lib (vc143, detected vc142, set
  Boost_COMPILER to override)

  * libboost_filesystem-vc143-mt-x64-1_85.lib (vc143, detected vc142, set
  Boost_COMPILER to override)

### Full version info

```shell
os:Windows 10
FreeCAD:FreeCAD-1.0rc1
FreeCAD-LibPack:LibPack-1.0.0 Version 3.0.0 RC1
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2024-09-19:
LibPack 3 requires the latest version of MSVC -- can you upgrade your compiler toolchain?

yan695432, 2024-09-20:
I use Visial Studio 2019 and it's already fine

chennes, 2024-09-20:
You will need to either use Conda to set up your development environment, use the older v2.11 version of the LibPack, or self-compile the Libpack (which will likely require adjustments, Microsoft had made significant advances to the state of the MSVC compiler in recent years).

closed 2024-09-20 by luzpaz