FreeCAD/FreeCAD #16636: Material: What will happen to several "Material" properties of Links?

Roy-043, 2024-09-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Links have an OverrideMaterial property and a ShapeMaterial property. The ShapeMaterial is in fact similar to the new ShapeAppearance property. This is already confusing.

If you set OverrideMaterial to true, the Diffuse color and the Transparency of the Link are determined by its ShapeMaterial property, but for the Ambient, Specular and Emissive colors, as well as for the Shininess, the ShapeAppearance of the Link's original object is used. Even though the Link's ShapeMaterial also contains these values.

Two other "Material" properties that may require attention:
MaterialList
OverrideMaterialList

Example file (remove the zip extension):
[link-override-material.FCStd.zip](https://github.com/user-attachments/files/17045667/link-override-material.FCStd.zip)

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38728 (Git)
Build type: Release
Branch: main
Hash: 9de78e27f4148e8032594788a9804486e769388b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

leoheck, 2024-11-24:
How do I set colors for links?

To change colors, I used to `Ctrl+d` with an Object or a Link selected, then I selected the color, and the original Object was used to change the color.

Now, when I do that the link does not change the color, nor does the original Object (this second part is good). But nothing happens.

When I `Ctrl+d` with a Link selected, I would like to change the color of the Link, not the original Object. So the `Display Properties` could do anything necessary to achieve that, but it could also have a checkbox to paint the Original Object as well.

Also, resetting the color of a Link should reset to the original Object color, not to the Freecad Default color.
But resetting the original Object color should return to the Freecad defaults.

leoheck, 2024-11-25:
Also, when I apply the random color, it does not change the color of the Link immediately. 
The object has to be deselected to make the object update the color.

leoheck, 2025-03-18:
Hey guys, do we have any improvement here? This issue makes working with links hard since we cannot color things differtly.

leoheck, 2025-03-18:
It looks like the ussage of links are not quite mature yet.