FreeCAD/FreeCAD #16640: Qt6: ampersands (&) displayed in dock widget titles

adrianinsaval, 2024-09-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]

seems to only happen in qt6 builds

### Full version info

```shell
OS: KDE Flatpak runtime (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 60a251354e)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 6.7.2, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: C/Default (C) [ OS: English/United Kingdom (en_GB) ]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2024-09-18:
@chennes do you see something like this on windows with libpack3?

chennes, 2024-09-19:
> @chennes do you see something like this on windows with libpack3?

No, those ampersands are not present in my local Qt6/Win11 builds.

obelisk79, 2024-09-19:
I also experience this. 

KDE Plasma/Wayland Session.
Qt6 build.

eszlari, 2024-09-20:
https://bugs.kde.org/show_bug.cgi?id=482657

chennes, 2024-09-20:
Here's the underlying Qt bug: https://bugreports.qt.io/browse/QTBUG-86407

chennes, 2024-09-20:
(@maxwxyz how would you like to mark Upstream bugs? This isn't something we can or should fix locally.)

funny0facer, 2024-09-20:
> Here's the underlying Qt bug: https://bugreports.qt.io/browse/QTBUG-86407
Are you sure? If I am not mistaken, the referenced bug was fixed in 6.7 in march ( https://codereview.qt-project.org/c/qt/qtbase/+/546540 ). The OP is using 6.7.2 from June

chennes, 2024-09-20:
> Are you sure?

No 😁 -- you are right in your read that it was listed as fixed prior to the release of Qt 6.7, but the description and screenshots of that bug look just like this one (to my reading, anyway). So it would seem that even the amended March version of the fix is missing something. If you can find an alternate bug report please link it here, otherwise we should investigate further and then re-open that closed Qt bug (or file a new one).

adrianinsaval, 2024-09-20:
I submited a possible fix to kde but am yet to actually test it, comments or testing are welcome: https://invent.kde.org/plasma/breeze/-/merge_requests/491

Andrei-Pozolotin, 2024-12-06:
a couple of points:

1) the problem is still present as of:
```
OS: Arch Linux (KDE/plasmax11/xcb)
Version: 1.1.0dev.39468 (Git)
Hash: 870e260bc926e50651507c905eaceb5576a2da37
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/breeze
```

2) some people merely override KDE, for example:
https://github.com/eteran/nedit-ng/commit/388e33977b235be0111c13100b70a394b07f571f

VictorElHajj, 2024-12-19:
Trying FreeCAD for the first time and this was really confusing, still an issue on Arch Linux.

Syres916, 2025-04-19:
It's fixed for me using Qt 6.8.3 and above:

```
OS: EndeavourOS (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.41308 (Git)
Build date: 2025/04/18 04:41:18
Build type: Release
Branch: main
Hash: 6be09c29f6217c6338cf09dcc9a658454287bd83
Python 3.13.2, Qt 6.8.3, Coin 4.0.3, Vtk 9.4.1, boost 1_87, Eigen3 3.4.0, PySide 
shiboken , SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: User_dark.qss/FreeCAD Dark/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/96.075/1
Installed mods: 
  * FreeCAD-themes 2025.1.7
  * sheetmetal 0.7.22
  * Reinforcement
  * fasteners 0.5.38
```

```
OS: EndeavourOS (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.41308 (Git)
Build date: 2025/04/18 04:41:18
Build type: Release
Branch: main
Hash: 6be09c29f6217c6338cf09dcc9a658454287bd83
Python 3.13.2, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, boost 1_87, Eigen3 3.4.0, PySide 
shiboken , SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: User_dark.qss/FreeCAD Dark/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/96.075/1
Installed mods: 
  * FreeCAD-themes 2025.1.7
  * sheetmetal 0.7.22
  * Reinforcement
  * fasteners 0.5.38
```

maxwxyz, 2025-04-19:
@adrianinsaval @oursland which versions are used for release/weekly?

oursland, 2025-04-19:
6.8.3

maxwxyz, 2025-04-20:
Thanks, closing then.

closed 2025-04-20 by maxwxyz