FreeCAD/FreeCAD #16644: Part: Crash when changing edge to be too small for fillet

theosib, 2024-09-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have attached a crash report and a document. I get a crash when I try to make the edge for a fillet too small.

Steps to reproduce:
- Navigate to Cut > Fusion001 > Extrude003 > Sketch003
- Change radius of circle to 1mm

[fillet crash.txt.zip](https://github.com/user-attachments/files/17048737/fillet.crash.txt.zip)
[board stencil holder.FCStd.zip](https://github.com/user-attachments/files/17048738/board.stencil.holder.FCStd.zip)

Note: This crash is not reliable reproducible.

### Full version info

```shell
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38768 (Git)
Build type: Release
Branch: main
Hash: 758674d40be45e365f4148fb2a5ea6bd6f93bc74
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US) [ OS: C/Default (C) ]
Stylesheet/Theme/QtStyle: unset/Classic/Qt default
Installed mods: 
  * Curves 0.6.44
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-18:
Hi @johnocfii, do you mind testing on macOS ? TIA!

Roy-043, 2024-09-18:
I can reproduce on Windows. Except there is no crash but a warning:
> Fillet: A fatal error occurred when making fillets

To avoid the problem: change the Refine property of the Cut to true.
```

OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38728 (Git)
Build type: Release
Branch: main
Hash: 9de78e27f4148e8032594788a9804486e769388b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

JohnOCFII, 2024-09-18:
> Hi @JohnOCFII, do you mind testing on macOS ? TIA!

Oh yeah.  Crashes Hard.  

```
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38768 (Git)
Build type: Release
Branch: main
Hash: 758674d40be45e365f4148fb2a5ea6bd6f93bc74
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * Curves 0.6.47
```

Crash Report:

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [26793]
Path:                  /Applications/Other Apps/FreeCAD-38768.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             X86-64 (Translated)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-09-18 16:05:40.2006 -0500
OS Version:            macOS 14.6.1 (23G93)
Report Version:        12
Anonymous UUID:        E1A9B3E8-17C0-3635-633F-31D076EC9805

Sleep/Wake UUID:       332CAEE2-EDA3-4A75-AAEC-14551B2FDA4D

Time Awake Since Boot: 970000 seconds
Time Since Wake:       3027 seconds

System Integrity Protection: enabled

Notes:
PC register does not match crashing frame (0x0 vs 0x12E0DE8F4)

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000048
Exception Codes:       0x0000000000000001, 0x0000000000000048

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [26793]

VM Region Info: 0x48 is not in any region.  Bytes before following region: 4344762296
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      102f7d000-102f9d000    [  128K] r-x/r-x SM=COW  /Applications/Other Apps/FreeCAD-38768.app/Contents/Resources/bin/freecad

Error Formulating Crash Report:
PC register does not match crashing frame (0x0 vs 0x12E0DE8F4)

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
[... 487 lines cut ...]
487 libQt5Core.5.15.13.dylib      	       0x112acbd37 0x11297f000 + 1363255
488 libQt5Core.5.15.13.dylib      	       0x112acbd37 0x11297f000 + 1363255
489 libQt5Core.5.15.13.dylib      	       0x112acbd37 0x11297f000 + 1363255
490 libQt5Core.5.15.13.dylib      	       0x112acbd37 0x11297f000 + 1363255
491 libQt5Core.5.15.13.dylib      	       0x112acbd37 0x11297f000 + 1363255
492 libQt5Core.5.15.13.dylib      	       0x112acbd37 0x11297f000 + 1363255
493 libQt5Core.5.15.13.dylib      	       0x112acbd37 0x11297f000 + 1363255
494 libQt5Core.5.15.13.dylib      	       0x112acbd37 0x11297f000 + 1363255
495 libQt5Core.5.15.13.dylib      	       0x112acbd37 0x11297f000 + 1363255
496 libQt5Core.5.15.13.dylib      	       0x112acbd37 0x11297f000 + 1363255
497 libQt5Core.5.15.13.dylib      	       0x112acbd37 0x11297f000 + 1363255
498 libQt5Core.5.15.13.dylib      	       0x112acbd37 0x11297f000 + 1363255
499 libQt5Core.5.15.13.dylib      	       0x112acbd37 0x11297f000 + 1363255
500 libQt5Core.5.15.13.dylib      	       0x112acbd37 0x11297f000 + 1363255
501 libQt5Core.5.15.13.dylib      	       0x112acbd37 0x11297f000 + 1363255
502 libQt5Core.5.15.13.dylib      	       0x112acbd37 0x11297f000 + 1363255
503 libQt5Core.5.15.13.dylib      	       0x112acbd37 0x11297f000 + 1363255
504 libQt5Core.5.15.13.dylib      	       0x112acbd37 0x11297f000 + 1363255
505 libQt5Core.5.15.13.dylib      	       0x112acbd37 0x11297f000 + 1363255
506 libQt5Core.5.15.13.dylib      	       0x112acbd37 0x11297f000 + 1363255
507 libQt5Core.5.15.13.dylib      	       0x112acbd37 0x11297f000 + 1363255
508 libQt5Core.5.15.13.dylib      	       0x112acbd37 0x11297f000 + 1363255
509 libQt5Core.5.15.13.dylib      	       0x112acbd37 0x11297f000 + 1363255
510 libQt5Core.5.15.13.dylib      	       0x112acbd37 0x11297f000 + 1363255
```

maxwxyz, 2024-10-07:
No crash on Windows but "Recompute failed".
Issue also occurs in the stable version:
```
[code]
OS: Windows 11 build 26100
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.27.10
  * Curves 0.6.48
  * dodo 1.0.1
  * fasteners 0.5.29
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.24
  * Rocket 3.3.0
  * sheetmetal 0.4.26
  * Silk 0.1.5
  * Silk.backup1727166062.7227929 0.1.5 (Disabled)
[/code]

```

pieterhijma, 2026-09-04:
This is not an issue anymore with OCCT 8.1:

```
OS: Arch Linux (tty)
Architecture: x86_64
Version: 26.3.0dev.48280 (Git)
Build date: 2026/08/28 12:56:10
Build type: Debug
Branch: main
Hash: 9a95906333eca0f8a86dba9e320ece2aff050e06
Python 3.14.7, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.6.2, boost 1_92, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, SMESH 7.7.1.0, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.5-alpha2601161257, OCC 8.1.0.dev1-3d097a0328
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: xcb/via Qt/via Qt
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1.33333
OpenGL version/vendor/renderer: 4.6 (Compatibility Profile) Mesa 26.2.1-arch1.1/AMD/AMD Ryzen 9 7900 12-Core Processor (radeonsi, raphael_mendocino, ACO, DRM 3.64, 7.1.9-arch1-2)
```

Instead of crashing, the command reports that the operation cannot be done:

[image]

closed 2026-09-04 by pieterhijma