FreeCAD/FreeCAD #16650: Model and Task Panel UI

NSsweens, 2024-09-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

On my mac, the model and task panel are changing colour and transparency as I mouse over them.  It is practically unusable because the items I'm trying to interact with are often unable to be seen or read.

### Full version info

```shell
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * dodo 1.0.0
  * 3DfindIT 1.2.0
  * freecad.gears 1.0.0
  * BIM021 2021.12.0
  * FeedsAndSpeeds 0.5.0
  * ArchTextures
  * frame 0.1.1
  * A2plus 0.4.62
  * fasteners 0.5.10
  * lattice2 1.0.0
  * sheetmetal 0.4.1
  * MOOC 2022.4.21
  * 3D_Printing_Tools
  * ExplodedAssembly
  * LCInterlocking
  * Curves 0.6.23
  * Assembly4 0.50.7
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

GAZ082, 2024-09-19:
I think a screen recording would be handy. Also, JUST IN CASE, rename the config files (with FC closed) so they get re-generated with default values and start FC and see if this keeps happening.

NSsweens, 2024-10-06:
I've attached a screen recording of what I'm seeing.  This is with RC2.

https://github.com/user-attachments/assets/5c57106b-5e47-4f5f-b9ed-e343901f6dc3

GAZ082, 2024-10-07:
@NSsweens thanks! Did you try with resetting the config files?

NSsweens, 2024-10-08:
I'm having trouble finding the config files.  I looked around in the application package, and in the library directories.  Not sure which files to delete so they can be rewritten.

Also, in RC2, I closed the task/model window and don't seem to be able to get it back.  I can get the error reporter and the Python console, but not the task and model pane.

Thanks

GAZ082, 2024-10-08:
@NSsweens try this:
On Mac: Navigate to /Users/USERNAME/Library/Preferences/FreeCAD and delete the files user.cfg and system.cfg

GAZ082, 2024-10-08:
or rename if you want to keep them for later

GAZ082, 2024-10-09:
@luzpaz another issue solved by resetting the config files. Crazy. I know we are feature freezed, but would be nice if we can push the Safe Mode to 1.0 @chennes .

luzpaz, 2024-10-09:
It's also relevant to #16370  

Closing this ticket as resolved for now.

closed 2024-10-09 by luzpaz