FreeCAD/FreeCAD #16652: No preference to adjust text size

mousethief, 2024-09-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I can find no way to adjust the UI font size. For accessibility, and to match other software on my system, I would need to make all text about 175-200% of what it comes up as. 'Preferences->General' does give an option to adjust the toolbar icons, and the build I was using previously did give me an option to adjust (most) UI text with 'Preferences->General->Default Font Size'

### Full version info

```shell
I previously used RealThunder's link branch:
[code]
OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 2024.409.0.14555 (Git shallow) AppImage
Build type: Release
Branch: (HEAD detached at 20240407stable)
Hash: 973e4821bcd19a5a1af9ae3e1d2ed961aad0a8ab
Python 3.11.6, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * FreeCAD-themes 2024.7.24
  * Manipulator 1.5.7
[/code]
and I am trying out mainline 1.0RC1:
[code]
OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * FreeCAD-themes 2024.7.24
  * Manipulator 1.5.7
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2025-01-12:
That works well for the whole UI. Note that using that variable things like bitmaps (e.g. recently opened file thumbnails in the Start page) apear noticeably pixelated.

UienZYC, 2026-01-15:
You can modify the shortcut's properties - Target, replacing it with 
`cmd /c "set QT_SCALE_FACTOR=1.25 && start "" "path\to\freecad.exe""  `