FreeCAD/FreeCAD #16653: Crashes when changing parameter

ueli72, 2024-09-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If I try to change a parameter early in the history of the construction, FreeCad crashes.

Example File: https://docs.bws-uster.ch/share/s/7nE51NRhQTG1NHBE6daEVw
Try to change length from 50 to 40 mm in Body "Box", Pad001.

This is the output of the terminal:
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x79b1867a5520]
#1  0x79b188e0b460 in Data::ElementNameComparator::operator()(Data::MappedName const&, Data::MappedName const&) const from /snap/freecad/1142/usr/lib/libFreeCADApp.so+0x20
#2  0x79b14871105d in std::_Rb_tree<Data::MappedName, std::pair<Data::MappedName const, Part::NameInfo>, std::_Select1st<std::pair<Data::MappedName const, Part::NameInfo> >, Data::ElementNameComparator, std::allocator<std::pair<Data::MappedName const, Part::NameInfo> > >::_M_get_insert_hint_unique_pos(std::_Rb_tree_const_iterator<std::pair<Data::MappedName const, Part::NameInfo> >, Data::MappedName const&) from /snap/freecad/1142/usr/lib/Part.so+0x5d
#3  0x79b1486ebd3b in Part::TopoShape::makeShapeWithElementMap(TopoDS_Shape const&, Part::TopoShape::Mapper const&, std::vector<Part::TopoShape, std::allocator<Part::TopoShape> > const&, char const*) from /snap/freecad/1142/usr/lib/Part.so+0x531b
#4  0x79b1486ed899 in Part::TopoShape::makeElementShape(BRepBuilderAPI_MakeShape&, std::vector<Part::TopoShape, std::allocator<Part::TopoShape> > const&, char const*) from /snap/freecad/1142/usr/lib/Part.so+0x129
#5  0x79b1486ed994 in Part::TopoShape::makeElementShape(BRepBuilderAPI_MakeShape&, Part::TopoShape const&, char const*) from /snap/freecad/1142/usr/lib/Part.so+0x84
#6  0x79b1486eef13 in Part::TopoShape::makeElementFillet(Part::TopoShape const&, std::vector<Part::TopoShape, std::allocator<Part::TopoShape> > const&, double, double, char const*) from /snap/freecad/1142/usr/lib/Part.so+0x113
#7  0x79b12db8b7dc in PartDesign::Fillet::execute() from /snap/freecad/1142/usr/lib/_PartDesign.so+0x20c
#8  0x79b188b861ca in App::DocumentObject::recompute() from /snap/freecad/1142/usr/lib/libFreeCADApp.so+0xba
#9  0x79b148489b1d in Part::Feature::recompute() from /snap/freecad/1142/usr/lib/Part.so+0xd
#10  0x79b188b204dd in App::Document::_recomputeFeature(App::DocumentObject*) from /snap/freecad/1142/usr/lib/libFreeCADApp.so+0x38d
#11  0x79b188b27e0b in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /snap/freecad/1142/usr/lib/libFreeCADApp.so+0x65b
#12  0x79b1899cd0bf in Gui::PropertyEditor::PropertyEditor::recomputeDocument(App::Document*) from /snap/freecad/1142/usr/lib/libFreeCADGui.so+0x7f
#13  0x79b1899cd193 in Gui::PropertyEditor::PropertyEditor::closeTransaction() from /snap/freecad/1142/usr/lib/libFreeCADGui.so+0x93
#14  0x79b1899ceb20 in Gui::PropertyEditor::PropertyEditor::closeEditor(QWidget*, QAbstractItemDelegate::EndEditHint) from /snap/freecad/1142/usr/lib/libFreeCADGui.so+0x50
#15  /snap/freecad/1142/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2f4928) [0x79b186eec928]
#16  0x79b187cebec4 in QAbstractItemDelegate::closeEditor(QWidget*, QAbstractItemDelegate::EndEditHint) from /snap/freecad/1142/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x54
#17  0x79b186ee239e in QObject::event(QEvent*) from /snap/freecad/1142/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x2ae
#18  0x79b187a4e763 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /snap/freecad/1142/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x83
#19  0x79b18962aec8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /snap/freecad/1142/usr/lib/libFreeCADGui.so+0x98
#20  0x79b186eb4e4a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /snap/freecad/1142/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x13a
#21  0x79b186eb7f47 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) from /snap/freecad/1142/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x187
#22  /snap/freecad/1142/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x316ce7) [0x79b186f0ece7]
#23  /snap/freecad/1142/kf5/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x26b) [0x79b184964d3b]
#24  /snap/freecad/1142/kf5/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0xab258) [0x79b1849ba258]
#25  /snap/freecad/1142/kf5/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x79b1849623e3]
#26  0x79b186f0e338 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /snap/freecad/1142/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x68
#27  0x79b186eb376b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /snap/freecad/1142/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#28  0x79b186ebbd14 in QCoreApplication::exec() from /snap/freecad/1142/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x94
#29  0x79b189566608 in Gui::Application::runApplication() from /snap/freecad/1142/usr/lib/libFreeCADGui.so+0x6c8
#30  /snap/freecad/1142/usr/bin/FreeCAD(+0xa781) [0x61b59f6f5781]
#31  /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x79b18678cd90]
#32  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x79b18678ce40]
#33  /snap/freecad/1142/usr/bin/FreeCAD(+0xab45) [0x61b59f6f5b45]

### Full version info

```shell
OS: Ubuntu Core 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38774 (Git) Snap 1142
Build type: Release
Branch: main
Hash: 27ca64a2019f2018cb0b720a252329fdcfe0892d
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: German/Switzerland (de_CH)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ueli72, 2024-09-19:
I did the whole process again with the same underlying sketch (but probably with a slightly different order) and now it works flawlessly...

Working file: https://docs.bws-uster.ch/share/s/GN_d3wELT_W94tRylQJ6fA

luzpaz, 2024-09-20:
@ueli72 The link 403s. Can you zip up the file and drag/drop it into github textfield (which auto-attaches it to this ticket)

ueli72, 2024-09-23:
[RFID_crashing_and_working.FCStd.zip](https://github.com/user-attachments/files/17093932/RFID_crashing_and_working.FCStd.zip)
Sorry. Geoblocking....

schmidtw, 2024-10-20:
I'm seeing this on several models as well.  I can't really build & make a patch due to being on Fedora 40 & unable to build the code locally due to python incompatibility.  However, based on examining the stack trace information I can find it appears that `leftName` or `rightName` are able to be null despite being references.

https://github.com/FreeCAD/FreeCAD/blob/ef2156d0f6d116b1642a5531a2d44fc6b7d17042/src/App/MappedElement.cpp#L39

This is based on:
```
0000000000495f90 <_ZNK4Data21ElementNameComparatorclERKNS_10MappedNameES3_>:
  495f90:	41 57                	push   %r15
  495f92:	41 56                	push   %r14
  495f94:	41 55                	push   %r13
  495f96:	41 54                	push   %r12
  495f98:	55                   	push   %rbp
  495f99:	53                   	push   %rbx
  495f9a:	48 83 ec 48          	sub    $0x48,%rsp
  495f9e:	48 8b 3e             	mov    (%rsi),%rdi
  495fa1:	4c 8b 6e 08          	mov    0x8(%rsi),%r13
  495fa5:	4c 8b 22             	mov    (%rdx),%r12
  495fa8:	4c 8b 42 08          	mov    0x8(%rdx),%r8
  495fac:	8b 6f 04             	mov    0x4(%rdi),%ebp
```

For me the crash happens at offset `0x1c` after the function start.  This is address `495fac`.

(This is from Copilot, but it makes sense to me since we're seeing a `SIGSEGV`)

Based on x86-64 calling convention for Linux, the first argument to a function is passed in the %rdi register, and the second argument is passed in the %rsi register.

Therefore:

%rdi holds the address of leftName.
%rsi holds the address of rightName.

If the lines:
```c++
    int size = static_cast<int>(std::min(leftName.size(), rightName.size()));
    if (size == 0U) {
        return leftName.size() < rightName.size();
    }
 ```
were changed to:
```c++
    int sizeLeft = 0;
    int sizeRight = 0;
    if (leftName != NULL) {
        sizeLeft = leftName.Size();
    }
    if (rightName != NULL) {
        sizeRight = rightName.Size();
    }
    int size = static_cast<int>(std::min(sizeLeft, sizeRight));
    if (size == 0U) {
        return sizeLeft < sizeRight;
    }
```

this general class of crash would be eliminated.

luzpaz, 2024-10-20:
@schmidtw can you post the step-by-step to trigger the crash bug ?

schmidtw, 2024-10-20:
[crash.zip](https://github.com/user-attachments/files/17450676/crash.zip)

1. Open the file.
2. Select the `centrotec-bit-holder-large-drills` --> `Part` --> `Body` --> `PlacementSketch010` sketch.
3. Delete any of the green lines.
4. Press the `Close` button in the `Tasks` pane.
5. Crash

I'm using:
```
OS: Fedora Linux 40 (Workstation Edition) (GNOME/gnome-xorg/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38991 (Git) AppImage
Build type: Release
Branch: main
Hash: ef2156d0f6d116b1642a5531a2d44fc6b7d17042
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * sheetmetal 0.4.26
  * lattice2 1.0.0
  * Curves 0.6.50
```

My stack trace is a bit different, but the general theme is the same - I need to edit a model & either a reference was removed or changes names & the program crashes before I can do anything.

This is my specific stack trace when I followed the steps above:
```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x40d00) [0x7f529144fd00]
#1  0x7f5293695fac in Data::ElementNameComparator::operator()(Data::MappedName const&, Data::MappedName const&) const from /tmp/.mount_FreeCAh943X9/usr/bin/../lib/libFreeCADApp.so+0x1c
#2  0x7f523e4e15ec in std::_Rb_tree<Data::MappedName, std::pair<Data::MappedName const, Data::IndexedName>, std::_Select1st<std::pair<Data::MappedName const, Data::IndexedName> >, Data::ElementNameComparator, std::allocator<std::pair<Data::MappedName const, Data::IndexedName> > >::_M_get_insert_hint_unique_pos(std::_Rb_tree_const_iterator<std::pair<Data::MappedName const, Data::IndexedName> >, Data::MappedName const&) from /tmp/.mount_FreeCAh943X9/usr/lib/./Part.so+0x5c
#3  0x7f523e4e1746 in std::_Rb_tree_iterator<std::pair<Data::MappedName const, Data::IndexedName> > std::_Rb_tree<Data::MappedName, std::pair<Data::MappedName const, Data::IndexedName>, std::_Select1st<std::pair<Data::MappedName const, Data::IndexedName> >, Data::ElementNameComparator, std::allocator<std::pair<Data::MappedName const, Data::IndexedName> > >::_M_emplace_hint_unique<Data::MappedName&, Data::IndexedName&>(std::_Rb_tree_const_iterator<std::pair<Data::MappedName const, Data::IndexedName> >, Data::MappedName&, Data::IndexedName&) from /tmp/.mount_FreeCAh943X9/usr/lib/./Part.so+0x76
#4  0x7f523e4f0f5f in Part::TopoShape::makeShapeWithElementMap(TopoDS_Shape const&, Part::TopoShape::Mapper const&, std::vector<Part::TopoShape, std::allocator<Part::TopoShape> > const&, char const*) from /tmp/.mount_FreeCAh943X9/usr/lib/./Part.so+0x5b0f
#5  0x7f523e4f1b9e in Part::TopoShape::makeElementShape(BRepBuilderAPI_MakeShape&, std::vector<Part::TopoShape, std::allocator<Part::TopoShape> > const&, char const*) from /tmp/.mount_FreeCAh943X9/usr/lib/./Part.so+0x11e
#6  0x7f523e5099c8 in Part::TopoShape::makeElementBoolean(char const*, std::vector<Part::TopoShape, std::allocator<Part::TopoShape> > const&, char const*, double) from /tmp/.mount_FreeCAh943X9/usr/lib/./Part.so+0xbb8
#7  /tmp/.mount_FreeCAh943X9/usr/lib/./Part.so(+0x36fd0f) [0x7f523e36fd0f]
#8  0x7f523e37003e in Part::TopoShapePy::staticCallback_cut(_object*, _object*) from /tmp/.mount_FreeCAh943X9/usr/lib/./Part.so+0x1e
#9  /tmp/.mount_FreeCAh943X9/usr/bin/../lib/libpython3.11.so.1.0(+0x206068) [0x7f5293c06068]
#10  /tmp/.mount_FreeCAh943X9/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x26c) [0x7f5293be5b5c]
#11  /tmp/.mount_FreeCAh943X9/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x7f5293bf1532]
#12  /tmp/.mount_FreeCAh943X9/usr/bin/../lib/libpython3.11.so.1.0(+0x235664) [0x7f5293c35664]
#13  0x7f52934fbcf3 in App::FeaturePythonImp::execute() from /tmp/.mount_FreeCAh943X9/usr/bin/../lib/libFreeCADApp.so+0x193
#14  0x7f52456f3c26 in App::FeaturePythonT<PartDesign::Feature>::execute() from /tmp/.mount_FreeCAh943X9/usr/lib/_PartDesign.so+0x16
#15  0x7f52934629a2 in App::DocumentObject::recompute() from /tmp/.mount_FreeCAh943X9/usr/bin/../lib/libFreeCADApp.so+0xc2
#16  0x7f523e2a552c in Part::Feature::recompute() from /tmp/.mount_FreeCAh943X9/usr/lib/./Part.so+0xc
#17  0x7f529343c97c in App::Document::_recomputeFeature(App::DocumentObject*) from /tmp/.mount_FreeCAh943X9/usr/bin/../lib/libFreeCADApp.so+0x32c
#18  0x7f5293448f65 in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /tmp/.mount_FreeCAh943X9/usr/bin/../lib/libFreeCADApp.so+0x625
#19  0x7f52934c80ce in App::DocumentPy::recompute(_object*) from /tmp/.mount_FreeCAh943X9/usr/bin/../lib/libFreeCADApp.so+0x2ee
#20  0x7f52934c81e4 in App::DocumentPy::staticCallback_recompute(_object*, _object*) from /tmp/.mount_FreeCAh943X9/usr/bin/../lib/libFreeCADApp.so+0x24
#21  /tmp/.mount_FreeCAh943X9/usr/bin/../lib/libpython3.11.so.1.0(+0x206068) [0x7f5293c06068]
#22  /tmp/.mount_FreeCAh943X9/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x26c) [0x7f5293be5b5c]
#23  /tmp/.mount_FreeCAh943X9/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x7f5293bf1532]
#24  /tmp/.mount_FreeCAh943X9/usr/bin/../lib/libpython3.11.so.1.0(+0x2abd1d) [0x7f5293cabd1d]
#25  /tmp/.mount_FreeCAh943X9/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x7f5293cab794]
#26  /tmp/.mount_FreeCAh943X9/usr/bin/../lib/libpython3.11.so.1.0(+0x2cabf3) [0x7f5293ccabf3]
[... cut ...]
#78  0x7f52918bd212 in QCoreApplication::exec() from /tmp/.mount_FreeCAh943X9/usr/bin/../lib/libQt5Core.so.5+0x82
#79  0x7f52944a3898 in Gui::Application::runApplication() from /tmp/.mount_FreeCAh943X9/usr/bin/../lib/libFreeCADGui.so+0xa28
#80  /tmp/.mount_FreeCAh943X9/usr/bin/freecad(+0x7658) [0x557b188aa658]
#81  /lib64/libc.so.6(+0x2a088) [0x7f5291439088]
#82  /lib64/libc.so.6(__libc_start_main+0x8b) [0x7f529143914b]
#83  /tmp/.mount_FreeCAh943X9/usr/bin/freecad(+0x79c1) [0x557b188aa9c1]
```

ueli72, 2024-10-20:
Just change the height of the second pad from 50 to 40.

ueli72, 2024-10-20:
[image]

FlachyJoe, 2024-10-21:
I confirm the bug and the diagnostic. 
Fix has to be done in a lower level, I prepare a PR.

chennes, 2024-10-22:
@CalligaroV any insight?

CalligaroV, 2024-10-23:
@chennes, ATM I can't give a detailed feedback as I'm mostly following another (Blocker) issue.

Looking at the backtrace in @schmidtw's https://github.com/FreeCAD/FreeCAD/issues/16653#issuecomment-2425022850,  the first thing I think may be a good (and probably obvious) starting point is to investigate what happens after `SketcherGui::TaskDlgEditSketch::reject()` is called.

Adding this to my ToDo list

maxwxyz, 2024-11-03:
Cannot reproduce on Windows, is this a Linux specific issue?
The file in the original issue throws an error for the subshapebinder but no crash:
[image]
The second file https://github.com/FreeCAD/FreeCAD/issues/16653#issuecomment-2425022850 has no error or crash.

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39100 (Git)
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.27.13
  * Curves 0.6.51
  * dodo 1.0.1
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.25
  * Rocket 3.3.0
  * sheetmetal 0.5.3
  * Silk 0.1.5
```

MiqCG, 2024-11-03:
It's not specific of Linux; cannot reproduce on Linux Devuan 5
Using "RFID_crahsing.FCStd" file, doing "change length from 50 to 40 mm in Body "Box", Pad001", same error on Report view and no crashing as maxwxyz:
`
ShapeBinder: Invalid shape name ?Face2`

```
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39092 (Git) AppImage
Build type: Release
Branch: main
Hash: a65d05c6e495038168eefcba6a1e43218fbb3a20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

```

maxwxyz, 2024-11-04:
@chennes can we drop the blocker tag?

schmidtw, 2024-11-04:
I've been able to build the tip of tree using QT6 on Fedora 40 and reproduced the similar crash I reported using the file & steps [here](https://github.com/FreeCAD/FreeCAD/issues/16653#issuecomment-2425022850).

```
OS: Fedora Linux 40 (Workstation Edition) (GNOME/gnome-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.39101 (Git)
Build type: Release
Branch: main
Hash: b7bdf93654c3940645d8c224bb7b24cf590cb79d
Python 3.12.7, Qt 6.7.2, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * sheetmetal 0.4.26
  * lattice2 1.0.0
  * Curves 0.6.50
```

CalligaroV, 2024-11-04:
@schmidtw can you please post also the backtrace of the crash reproduced with your build of https://github.com/FreeCAD/FreeCAD/issues/16653#issuecomment-2453977535?

It may have some more clues to work on.

TIA!

schmidtw, 2024-11-04:
Yes, here it is:
```
<PropertyLinks> PropertyLinks.cpp(447): centrotec_bit_holder_large_drills#Sketch012.ExternalGeometry missing element reference centrotec_bit_holder_large_drills#Sketch010 ;g14;SKT.Edge5
Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x40d00) [0x7f71d024fd00]
#1  0x7f71d4d3ae6e in QByteArray::operator[](long long) const from /home/schmidtw/projects/FreeCAD/build/lib/libFreeCADApp.so+0x3c
#2  0x7f71d4d3bb58 in Data::MappedName::operator[](int) const from /home/schmidtw/projects/FreeCAD/build/lib/libFreeCADApp.so+0xc8
#3  0x7f71d4d7f4eb in Data::ElementNameComparator::operator()(Data::MappedName const&, Data::MappedName const&) const from /home/schmidtw/projects/FreeCAD/build/lib/libFreeCADApp.so+0x223
#4  0x7f7173c0e7b4 in std::_Rb_tree<Data::MappedName, std::pair<Data::MappedName const, Data::IndexedName>, std::_Select1st<std::pair<Data::MappedName const, Data::IndexedName> >, Data::ElementNameComparator, std::allocator<std::pair<Data::MappedName const, Data::IndexedName> > >::_M_get_insert_hint_unique_pos(std::_Rb_tree_const_iterator<std::pair<Data::MappedName const, Data::IndexedName> >, Data::MappedName const&) from /home/schmidtw/projects/FreeCAD/build/Mod/Part/Part.so+0x20c
#5  0x7f7173c09f28 in std::_Rb_tree_iterator<std::pair<Data::MappedName const, Data::IndexedName> > std::_Rb_tree<Data::MappedName, std::pair<Data::MappedName const, Data::IndexedName>, std::_Select1st<std::pair<Data::MappedName const, Data::IndexedName> >, Data::ElementNameComparator, std::allocator<std::pair<Data::MappedName const, Data::IndexedName> > >::_M_emplace_hint_unique<Data::MappedName&, Data::IndexedName&>(std::_Rb_tree_const_iterator<std::pair<Data::MappedName const, Data::IndexedName> >, Data::MappedName&, Data::IndexedName&) from /home/schmidtw/projects/FreeCAD/build/Mod/Part/Part.so+0x6c
#6  0x7f7173c029a3 in std::_Rb_tree_iterator<std::pair<Data::MappedName const, Data::IndexedName> > std::map<Data::MappedName, Data::IndexedName, Data::ElementNameComparator, std::allocator<std::pair<Data::MappedName const, Data::IndexedName> > >::emplace_hint<Data::MappedName&, Data::IndexedName&>(std::_Rb_tree_const_iterator<std::pair<Data::MappedName const, Data::IndexedName> >, Data::MappedName&, Data::IndexedName&) from /home/schmidtw/projects/FreeCAD/build/Mod/Part/Part.so+0x4f
#7  0x7f7173bfddd9 in std::pair<std::_Rb_tree_iterator<std::pair<Data::MappedName const, Data::IndexedName> >, bool> std::map<Data::MappedName, Data::IndexedName, Data::ElementNameComparator, std::allocator<std::pair<Data::MappedName const, Data::IndexedName> > >::emplace<Data::MappedName&, Data::IndexedName&>(Data::MappedName&, Data::IndexedName&) from /home/schmidtw/projects/FreeCAD/build/Mod/Part/Part.so+0x165
#8  0x7f7173bd1a9a in Part::TopoShape::makeShapeWithElementMap(TopoDS_Shape const&, Part::TopoShape::Mapper const&, std::vector<Part::TopoShape, std::allocator<Part::TopoShape> > const&, char const*) from /home/schmidtw/projects/FreeCAD/build/Mod/Part/Part.so+0x31fa
#9  0x7f7173be8a4a in Part::TopoShape::makeElementShape(BRepBuilderAPI_MakeShape&, std::vector<Part::TopoShape, std::allocator<Part::TopoShape> > const&, char const*) from /home/schmidtw/projects/FreeCAD/build/Mod/Part/Part.so+0xc4
#10  0x7f7173bf6815 in Part::TopoShape::makeElementBoolean(char const*, std::vector<Part::TopoShape, std::allocator<Part::TopoShape> > const&, char const*, double) from /home/schmidtw/projects/FreeCAD/build/Mod/Part/Part.so+0x1e67
#11  /home/schmidtw/projects/FreeCAD/build/Mod/Part/Part.so(+0x7aeffa) [0x7f71739aeffa]
#12  0x7f71739aff9e in Part::TopoShapePy::cut(_object*) from /home/schmidtw/projects/FreeCAD/build/Mod/Part/Part.so+0x38
#13  0x7f717399f940 in Part::TopoShapePy::staticCallback_cut(_object*, _object*) from /home/schmidtw/projects/FreeCAD/build/Mod/Part/Part.so+0xa8
#14  /lib64/libpython3.12.so.1.0(+0x1923b8) [0x7f71d2f923b8]
#15  /lib64/libpython3.12.so.1.0(_PyObject_MakeTpCall+0x76) [0x7f71d2f6a516]
#16  /lib64/libpython3.12.so.1.0(_PyEval_EvalFrameDefault+0xb35) [0x7f71d2f73225]
#17  /lib64/libpython3.12.so.1.0(+0x1b1c45) [0x7f71d2fb1c45]
#18  0x7f71d66f8a7e in Base::pyCall(_object*, _object*) from /home/schmidtw/projects/FreeCAD/build/lib/libFreeCADGui.so+0x2a
#19  0x7f71d4b62ed6 in App::FeaturePythonImp::execute() from /home/schmidtw/projects/FreeCAD/build/lib/libFreeCADApp.so+0x1f2
#20  0x7f714c6310c8 in App::FeaturePythonT<PartDesign::Feature>::execute() from /home/schmidtw/projects/FreeCAD/build/Mod/PartDesign/_PartDesign.so+0x24
#21  0x7f71d4a5e42f in App::DocumentObject::recompute() from /home/schmidtw/projects/FreeCAD/build/lib/libFreeCADApp.so+0x75
#22  0x7f717388eaab in Part::Feature::recompute() from /home/schmidtw/projects/FreeCAD/build/Mod/Part/Part.so+0x1d
#23  0x7f714c629664 in PartDesign::Feature::recompute() from /home/schmidtw/projects/FreeCAD/build/Mod/PartDesign/_PartDesign.so+0xde
#24  0x7f71d49439ad in App::Document::_recomputeFeature(App::DocumentObject*) from /home/schmidtw/projects/FreeCAD/build/lib/libFreeCADApp.so+0x1eb
#25  0x7f71d49412d8 in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /home/schmidtw/projects/FreeCAD/build/lib/libFreeCADApp.so+0x986
#26  0x7f71d4b18c69 in App::DocumentPy::recompute(_object*) from /home/schmidtw/projects/FreeCAD/build/lib/libFreeCADApp.so+0x2c7
[... cut ...]
#87  0x7f71d099fa7c in QCoreApplication::exec() from /lib64/libQt6Core.so.6+0x9c
#88  /home/schmidtw/projects/FreeCAD/build/lib/libFreeCADGui.so(+0x13a807e) [0x7f71d65a807e]
#89  /home/schmidtw/projects/FreeCAD/build/lib/libFreeCADGui.so(+0x13a834d) [0x7f71d65a834d]
#90  0x7f71d65a863c in Gui::Application::runApplication() from /home/schmidtw/projects/FreeCAD/build/lib/libFreeCADGui.so+0x1f4
#91  ./FreeCAD() [0x42646c]
#92  /lib64/libc.so.6(+0x2a088) [0x7f71d0239088]
#93  /lib64/libc.so.6(__libc_start_main+0x8b) [0x7f71d023914b]
#94  ./FreeCAD() [0x425725]
```

maxwxyz, 2024-11-07:
@chennes did you already have time to look at this and reproduce? Otherwise I'd recommend to drop the `Blocker` tag.

FlachyJoe, 2024-11-08:
Maybe we're facing this problem:
https://stackoverflow.com/questions/59351752/c-unordered-map-emplace-function-throwing-seg-fault-and-i-have-no-idea-why

`emplace` copy the MappedName and destroy the copy but the `data` member is not copied and so points to a address marked free and reused.

chennes, 2024-11-10:
It doesn't crash for me, but I do get a similar error to @MiqCG:
```
Pad007: Sub shape not found: RFID_crahsing#Sketch002.?Edge75
```

Nevertheless, the problem is too isolated to be a blocker. We should continue working on it, hopefully to resolve for the 1.0.1 release.

wsteffe, 2025-01-07:
Is anybody working on this problem ? Not blocking but, IMO, anyway a serious bug affecting iterative parameter changes occurring, in example, in optimization tasks.

chennes, 2025-07-24:
Still no crash for me (even running with ASAN on), but here's my Report View output:
```
21:39:34  63.8205 <App> Document.cpp(423): auto transaction (800) 'RFID_crahsing#Pad001.Length
21:39:34  <Gui> Document.cpp(1086): RFID_crahsing#Pad001.Length modified
21:39:34  63.9285 Tree.cpp(3038): begin update status
21:39:34  63.9289 Tree.cpp(3105): update item status
21:39:34  63.9344 Tree.cpp(3201): done update status
21:39:34  63.964 MainWindow.cpp(1679): update actions
21:39:34  64.1212 Tree.cpp(3038): begin update status
21:39:34  64.1214 Tree.cpp(3105): update item status
21:39:34  64.1222 Tree.cpp(3201): done update status
21:39:34  64.3011 MainWindow.cpp(1679): update actions
21:39:36  65.4919 <App> Document.cpp(2688): Recompute pass 0
21:39:36  65.4921 <App> Document.cpp(2983): Recomputing RFID_crahsing#Pad001
21:39:36  <Gui> Document.cpp(2834): RFID_crahsing#Pad001.AlongSketchNormal editor changed
21:39:36  <Gui> Document.cpp(2834): RFID_crahsing#Pad001.AlongSketchNormal editor changed
21:39:36  65.5608 <App> Document.cpp(2983): Recomputing RFID_crahsing#Thickness
21:39:36  65.5898 <App> Document.cpp(2983): Recomputing RFID_crahsing#Pad002
21:39:36  65.6896 <App> Document.cpp(2983): Recomputing RFID_crahsing#Pad003
21:39:36  65.7902 <App> Document.cpp(2983): Recomputing RFID_crahsing#Pad004
21:39:36  65.8485 <App> Document.cpp(2983): Recomputing RFID_crahsing#Pad005
21:39:36  65.9438 <App> Document.cpp(2983): Recomputing RFID_crahsing#Pad006
21:39:36  <ElementMap> ElementMap.cpp(779): duplicate element mapping '#27;SKT;:H:4,V;:H8df,V -> #27;SKT;:H:4,V;:H8df,V;D1;:H8df:3,V Vertex3/Vertex2
21:39:36  <ElementMap> ElementMap.cpp(779): duplicate element mapping '#28;SKT;:H:4,V;:H8df,V -> #28;SKT;:H:4,V;:H8df,V;D1;:H8df:3,V Vertex5/Vertex4
21:39:36  <ElementMap> ElementMap.cpp(779): duplicate element mapping '#29;SKT;:H:4,V;:H8df,V -> #29;SKT;:H:4,V;:H8df,V;D1;:H8df:3,V Vertex7/Vertex6
21:39:36  <ElementMap> ElementMap.cpp(779): duplicate element mapping '#2a;SKT;:H:4,V;:H8df,V -> #2a;SKT;:H:4,V;:H8df,V;D1;:H8df:3,V Vertex8/Vertex1
21:39:36  <ElementMap> ElementMap.cpp(779): duplicate element mapping '#26;SKT;:H:4,V;:H8df,V -> #26;SKT;:H:4,V;:H8df,V;D1;:H8df:3,V Vertex11/Vertex10
21:39:36  <ElementMap> ElementMap.cpp(779): duplicate element mapping '#23;SKT;:H:4,V;:H8df,V -> #23;SKT;:H:4,V;:H8df,V;D1;:H8df:3,V Vertex13/Vertex12
21:39:36  <ElementMap> ElementMap.cpp(779): duplicate element mapping '#24;SKT;:H:4,V;:H8df,V -> #24;SKT;:H:4,V;:H8df,V;D1;:H8df:3,V Vertex15/Vertex14
21:39:36  <ElementMap> ElementMap.cpp(779): duplicate element mapping '#25;SKT;:H:4,V;:H8df,V -> #25;SKT;:H:4,V;:H8df,V;D1;:H8df:3,V Vertex16/Vertex9
21:39:36  <ElementMap> ElementMap.cpp(779): duplicate element mapping '#20;SKT;:H:4,V;:H8df,V -> #20;SKT;:H:4,V;:H8df,V;D1;:H8df:3,V Vertex20/Vertex17
21:39:36  <ElementMap> ElementMap.cpp(779): duplicate element mapping '#1f;SKT;:H:4,V;:H8df,V -> #1f;SKT;:H:4,V;:H8df,V;D1;:H8df:3,V Vertex22/Vertex19
21:39:36  <ElementMap> ElementMap.cpp(779): duplicate element mapping '#21;SKT;:H:4,V;:H8df,V -> #21;SKT;:H:4,V;:H8df,V;D1;:H8df:3,V Vertex23/Vertex18
21:39:36  <ElementMap> ElementMap.cpp(779): duplicate element mapping '#22;SKT;:H:4,V;:H8df,V -> #22;SKT;:H:4,V;:H8df,V;D1;:H8df:3,V Vertex24/Vertex21
21:39:36  <ElementMap> ElementMap.cpp(779): duplicate element mapping '#2e;SKT;:H:4,V;:H8df,V -> #2e;SKT;:H:4,V;:H8df,V;D1;:H8df:3,V Vertex27/Vertex26
21:39:36  <ElementMap> ElementMap.cpp(779): duplicate element mapping '#2d;SKT;:H:4,V;:H8df,V -> #2d;SKT;:H:4,V;:H8df,V;D1;:H8df:3,V Vertex30/Vertex25
21:39:36  <ElementMap> ElementMap.cpp(779): duplicate element mapping '#2b;SKT;:H:4,V;:H8df,V -> #2b;SKT;:H:4,V;:H8df,V;D1;:H8df:3,V Vertex31/Vertex28
21:39:36  <ElementMap> ElementMap.cpp(779): duplicate element mapping '#2c;SKT;:H:4,V;:H8df,V -> #2c;SKT;:H:4,V;:H8df,V;D1;:H8df:3,V Vertex32/Vertex29
21:39:36  66.1314 <App> Document.cpp(2983): Recomputing RFID_crahsing#Sketch001
21:39:36  Sketcher::setUpSketch()-T:0.00184421
21:39:36  Sketcher::setUpSketch()-T:0.00132667
21:39:36  Sketcher::Solve()-DogLeg-T:0.00159221
21:39:36  <Sketch> SketchObject.cpp(752): update geometry history (4, 4) time: 9.6667e-05s
21:39:36  66.1433 <App> Document.cpp(2983): Recomputing RFID_crahsing#Pocket
21:39:36  <PropertyLinks> PropertyLinks.cpp(474): RFID_crahsing#Fillet.Base auto change element reference RFID_crahsing#Pocket ;#50b:30;:H8e7,E.Edge42 -> ;#1935:30;:H8e7,E.Edge42
21:39:36  <PropertyLinks> PropertyLinks.cpp(474): RFID_crahsing#Fillet.Base auto change element reference RFID_crahsing#Pocket ;#50b:31;:H8e7,E.Edge41 -> ;#1935:31;:H8e7,E.Edge41
21:39:36  <PropertyLinks> PropertyLinks.cpp(474): RFID_crahsing#Fillet.Base auto change element reference RFID_crahsing#Pocket ;#50a:2c;:H8e7,E.Edge39 -> ;#1934:2c;:H8e7,E.Edge39
21:39:36  <PropertyLinks> PropertyLinks.cpp(474): RFID_crahsing#Fillet.Base auto change element reference RFID_crahsing#Pocket ;#50a:2d;:H8e7,E.Edge37 -> ;#1934:2d;:H8e7,E.Edge37
21:39:36  <PropertyLinks> PropertyLinks.cpp(474): RFID_crahsing#Fillet.Base auto change element reference RFID_crahsing#Pocket ;#50b:2e;:H8e7,E.Edge38 -> ;#1935:2e;:H8e7,E.Edge38
21:39:36  <PropertyLinks> PropertyLinks.cpp(474): RFID_crahsing#Fillet.Base auto change element reference RFID_crahsing#Pocket ;#50b:2f;:H8e7,E.Edge40 -> ;#1935:2f;:H8e7,E.Edge40
21:39:37  66.318 <App> Document.cpp(2983): Recomputing RFID_crahsing#Fillet
[... cut ...]
21:39:38  67.6094 <App> Document.cpp(2751): Recompute time: 2.11876s
21:39:38  67.6095 <App> Document.cpp(2763): Recompute total time: 2.12463s
21:39:38  Fillet003: Invalid edge link: ;#94e:5;:H8ed,E.Edge79
21:39:38  <App> AutoTransaction.cpp(209): close transaction 'Edit Pad001.Length' 0
21:39:38  67.6243 MainWindow.cpp(1679): update actions
21:39:38  67.637 Tree.cpp(3038): begin update status
21:39:38  67.6373 Tree.cpp(3105): update item status
21:39:38  67.6397 Tree.cpp(3201): done update status
```

chennes, 2025-07-24:
I'll note that @bgbsww left us this tiny clue in `Fillet::execute()`:
```
            // Toponaming project March 2024:  Replaced this code because it wouldn't work:
//            TopoDS_Shape edge;
//            try {
//                edge = baseTopoShape.getSubShape(ref.c_str());
//            }catch(...){}
```

drwho495, 2026-05-06:
The issue is still there on the V2, what seems to be happening is FreeCAD is losing track of a face for whatever reason (could be an OpenCASCADE issue, but I'm holding out hope that it isn't)

chennes, 2026-09-11:
@drwho495 did anything ever come of this?

drwho495, 2026-09-11:
> [@drwho495](https://github.com/drwho495) did anything ever come of this?

The changes to fix naming for the V2 can be found at: https://github.com/drwho495/FreeCAD/commit/36f1d2bafe2cb9d8eabbc1bb0bfab01f9075ef97

Not sure how this affects the crash