FreeCAD/FreeCAD #16654: Fillet fails on refined shape, where it succeeds on same edges before refinement.

SoulWager, 2024-09-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The fillet fails if the same edges are picked on the refined shape. I believe this is a different bug than #5799  because the geometry check passes. 

Hopefully this will be useful for troubleshooting: [Fillet example.zip](https://github.com/user-attachments/files/17054454/Fillet.example.zip) 

The model is based on a drawing from one of the solidworks competitions. 

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.5.28
  * Curves 0.6.47
  * POV-Ray-Rendering
  * lattice2 1.0.0
  * Assembly4 1.27.10
  * kicadStepUpMod
  * kicadStepUpMod.backup1699473187.113371 10.17.1
  * fcgear 1.0.0
  * Silk 0.1.5
  * Alternate_OpenSCAD 1.0.0
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.25
  * freecad.gears 1.0.0
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

SoulWager, 2025-07-26:
Still exists in build 42721.

luzpaz, 2025-07-26:
@maxwxyz does this look OCC related ?

maxwxyz, 2025-07-26:
not sure @FlachyJoe @pieterhijma

FEA-eng, 2026-09-03:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 26.3.0dev.20260902 (Git shallow)
Build date: 2026/09/02 02:30:22
Build type: Release
Branch: (HEAD detached at c68c45227)
Hash: c68c452275ec188f35c02f6b30aba880b3783cf1
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```