FreeCAD/FreeCAD #16656: Feature request: DELETE key functionalty under macOS

chriskuku, 2024-09-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Under macOS (in my case 11.7.10 BigSur) the DELETE key (backspace key, upper right under the power button, above the <CR>-key) does not DELETE anything that's selected for deletion. Instead one has to prefix the keypress by the "fn" key. 

This is awkward. Other CAD-Systems, e.g. QCAD, estabish the backspace key such that it directly deletes the respective object.

I thing this feature would be due and would make life easier for the user.

### Full version info

```shell
FreeCAD-0.21.2-33771
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hyarion, 2024-09-19:
I can verify this behaviour. It's not really a deal breaker to me, but it sure is an annoyance.

Regarding mapping delete to backspace: It's not just other cad software, most programs allows backspace to remove things on mac.

Are there any cases in freecad where both delete and backspace are used, but with different functionality (except when editing text)?

luzpaz, 2024-09-19:
> Version: 0.21.2.33771 (Git)

Hi @chriskuku, JFYI, we aren't backporting changes to 0.21.x anymore. Please use the RC1 release https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/1.0rc1

----

@hyarion can you add your full about info? TIA

----

A shortcut backspace/delete issues was tackled #9639 but perhaps there is more to do. See @johnocfii's comment: https://github.com/FreeCAD/FreeCAD/issues/9639#issuecomment-2335750864

hyarion, 2024-09-19:
Full version information (The main brach was pulled yesterday):
```
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38774 (Git)
Build type: Release
Branch: main
Hash: 27ca64a2019f2018cb0b720a252329fdcfe0892d
Python 3.11.8, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Manipulator 1.5.7
  * sheetmetal 0.2.61
  * Curves 0.6.6
```

I didn't know that it was possible to assign a shortcut to delete. I tested it and it works, thanks!

However, I think it would be good to have this backspace-to-delete shortcut enabled by default for new mac users.

luzpaz, 2024-09-19:
> However, I think it would be good to have this backspace-to-delete shortcut enabled by default for new mac users.

Agreed as this is something they assume by definition that these keys will do.

closed 2024-09-21 by chennes