FreeCAD/FreeCAD #16658: There is something wrong when saving a file.

leoheck, 2024-09-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Since the latest 1.0-RC it is taking to much time to save a file.
The file went from 50MB to 300MB I did not add much only a bunch of fasteners.

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38768 (Git) AppImage
Build type: Release
Branch: main
Hash: 758674d40be45e365f4148fb2a5ea6bd6f93bc74
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * Assembly4 0.50.16-dev
  * sheetmetal 0.4.26
  * InventorLoader 1.4.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.0.22
  * lattice2 1.0.0
  * kicadStepUpMod 11.3.2
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

aries2004-bit, 2024-09-19:
download
https://bit.ly/4gwC82r
I put the necessary dlls in the archive

leoheck, 2024-09-19:
Can someone you block the scammer (and hide this msg too)

maxwxyz, 2024-09-21:
@leoheck what should we do with such a specific and detailed issue? You know what is expected at least...

closed 2024-09-21 by maxwxyz

leoheck, 2024-09-21:
I was investigating this. But I don't know the known tools to diagnose this. Do you know something that can help investigate this problem?

Anyways. I think I found something.
I modeled a couple of wires/cables.

I used the Part Section to generate a "3d" path using 2 sketches, each one in a different plane. Then Part Extruding to make those planes intersect. These sketches were drawn using BSplines. Then another Sketch (or more) was used as a profile to produce a Part Sweep making the path look like a ribbon cable.

The problem with that is that the file size grows really fast. I isolated one of those cables in an unique file, and it produced a file with 20 MB.

Then I tried to reduce the number of points improving my non-efficient and quickly created Sketches in another file. The file size was reduced to 164 KB.

Then on another file I replaced BSplines with normal lines/curves. The produced file was further reduced to 64 KB.

Something in this process is taking a lot of space quickly. Mostly due to the BSplines, I guess. I am wondering if there is still space for optimization on some of these features. 

My goal is to be able to model 3d wires/cables that helps planning wire routing that doesn't explode the file size quickly.

luzpaz, 2024-09-22:
@leoheck can you drill down and see how to create a minimal example for reproducibility ?  
@AjinkyaDahale have you seen size bloat in conjunction with bspline behavior ?

AjinkyaDahale, 2024-09-22:
@luzpaz I haven't done any benchmarks with b-splines, and I test with relatively trivial examples so can't tell.

leoheck, 2024-09-24:
I was trying to remake my cables.
This one for instance, 
[image]

Has 700MB already.
```
/tmp …
➜ du -sh cable.FCStd                                     
693M	cable.FCStd
```

Interceptions make the 3d path
[image]

You can find this file here,
https://www.dropbox.com/scl/fi/gt0fcydlzcbns59478fqr/cable_big_file_issue.FCStd?rlkey=qp6ghzn5xf3qdjycksuph35xe&dl=0

leoheck, 2024-09-27:
@AjinkyaDahale did you have a chance to check the file?