FreeCAD/FreeCAD #16659: Dependency graph of selected objects only

leoheck, 2024-09-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

For complex designs, this would be a nice feature.

It would be nice to generate this graph for selected objects only.
It could add its dependencies as well, not the whole dependency, but something indicating a reference to another Body, for instance, when working with Part Design.

For instance, I have this project that I am trying to understand why it is so slow to be saved.
[image]

Extra info per Module/Object would be a nice addition too, for instance, the disk space of each object.

### Full version info

```shell
1.1
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Buqra55, 2024-09-19:
download

https://bit.ly/4eugCty
I put the necessary dlls in the archive

hasecilu, 2024-09-19:
> This feature could also help with triage and troubleshooting bugs. @hasecilu any idea what a proposal for this would look like? How complex would this be?

Give me some days to check this