FreeCAD/FreeCAD #16660: V-bit tools not mapped to corresponding OCL tool

Unl0gic, 2024-09-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When trying to use v-bit shaped tools with the 3D Surface operations in the CAM/Path workbench you get the following error and the operation is aborted:

Failed to map selected tool to an OCL tool type.

This problem is described in the thread linked below.

https://forum.freecad.org/viewtopic.php?t=76665

It seems to be caused by a miss match when looking up the corresponding OCL tool shape in the SurfaceSupport.py file using the toolMap.

```
            toolMap = {
                "endmill": "CylCutter",
                "ballend": "BallCutter",
                "bullnose": "BullCutter",
                "drill": "ConeCutter",
                "engraver": "ConeCutter",
                "v-bit": "ConeCutter",
                "chamfer": "None",
            }
```

Changing from "v-bit" to "v_bit" in the code above seems to solve the problem.

### Full version info

```shell
At the time of writing this bug is exist in the latest stable release (0.21.2) as well as in the latest weekly build (38728).
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-20:
@Unl0gic do you want to submit a PR ?

Unl0gic, 2024-09-23:
@luzpaz I've created a pull request here now:

https://github.com/FreeCAD/FreeCAD/pull/16742

closed 2024-10-07 by yorikvanhavre