FreeCAD/FreeCAD #16663: 1.0.0-rc1-2 : Caught signal 11 (Segmentation fault: address not mapped to object at address 0x18)

BogueUser, 2024-09-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Segmentation fault occurs at random points. Most occur shortly after opening a save file.

```
[customer:56796:0:56796] Caught signal 11 (Segmentation fault: address not mapped to object at address 0x18)
==== backtrace (tid:  56796) ====
 0 0x000000000004d212 ucs_event_set_fd_get()  ???:0
 1 0x000000000004d3dd ucs_event_set_fd_get()  ???:0
 2 0x000000000003d1d0 __sigaction()  ???:0
 3 0x0000000000c4dee4 Gui::LinkView::linkGetElementPicked()  ???:0
 4 0x0000000000c4f15b Gui::ViewProviderLink::getElementPicked()  ???:0
 5 0x0000000000bf717d Gui::ViewProviderDocumentObject::getElementPicked()  ???:0
 6 0x0000000000bf717d Gui::ViewProviderDocumentObject::getElementPicked()  ???:0
 7 0x0000000000a6234f Gui::SoFCUnifiedSelection::getPickedList()  ???:0
 8 0x0000000000a66740 Gui::SoFCUnifiedSelection::handleEvent()  ???:0
 9 0x00000000002b537c SoAction::traverse()  ???:0
10 0x000000000048843b SoChildList::traverse()  ???:0
11 0x00000000004d7486 SoGroup::doAction()  ???:0
12 0x00000000004ee26f SoSeparator::doAction()  ???:0
13 0x00000000002b537c SoAction::traverse()  ???:0
14 0x00000000002bc564 SoHandleEventAction::beginTraversal()  ???:0
15 0x00000000002ac551 SoAction::apply()  ???:0
16 0x00000000004a1359 SoEventManager::actuallyProcessEvent()  ???:0
17 0x0000000000b43352 SIM::Coin3D::Quarter::QuarterWidget::processSoEvent()  ???:0
18 0x0000000000b52d4e SIM::Coin3D::Quarter::SoQTQuarterAdaptor::processSoEvent()  ???:0
19 0x0000000000b6121d Gui::NavigationStyle::processSoEvent()  ???:0
20 0x0000000000b64476 Gui::BlenderNavigationStyle::processSoEvent()  ???:0
21 0x0000000000b6143c Gui::NavigationStyle::processEvent()  ???:0
22 0x0000000000b9627f Gui::View3DInventorViewer::processSoEvent()  ???:0
23 0x0000000000b3c985 SIM::Coin3D::Quarter::EventFilter::eventFilter()  ???:0
24 0x0000000000144ef8 QCoreApplicationPrivate::sendThroughObjectEventFilters()  ???:0
25 0x00000000000fc8bc QApplicationPrivate::notify_helper()  ???:0
26 0x0000000000101c92 QApplication::notify()  ???:0
27 0x00000000006a1c89 Gui::GUIApplication::notify()  ???:0
28 0x0000000000145aa8 QCoreApplication::notifyInternal2()  ???:0
29 0x00000000000f3f9d QApplicationPrivate::sendMouseEvent()  ???:0
30 0x00000000001635c2 QCommonStyle::~QCommonStyle()  ???:0
31 0x0000000000164c20 QCommonStyle::~QCommonStyle()  ???:0
32 0x00000000000fc8cc QApplicationPrivate::notify_helper()  ???:0
33 0x00000000006a1c89 Gui::GUIApplication::notify()  ???:0
34 0x0000000000145aa8 QCoreApplication::notifyInternal2()  ???:0
35 0x0000000000178140 QGuiApplicationPrivate::processMouseEvent()  ???:0
36 0x00000000001e7724 QWindowSystemInterface::sendWindowSystemEvents()  ???:0
37 0x0000000000051b87 QXcbConnection::xi2HandleEvent()  ???:0
38 0x000000000005d299 g_source_query_unix_fd()  ???:0
39 0x00000000000bfec7 g_io_channel_new_file()  ???:0
40 0x000000000005c795 g_main_context_iteration()  ???:0
41 0x00000000003a82bd QEventDispatcherGlib::processEvents()  ???:0
42 0x000000000014ff66 QEventLoop::exec()  ???:0
43 0x000000000014a11d QCoreApplication::exec()  ???:0
44 0x00000000005aa2a6 Gui::Application::runApplication()  ???:0
45 0x0000000000006d0c std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_replace_cold()  ???:0
46 0x0000000000025e08 __libc_init_first()  ???:0
47 0x0000000000025ecc __libc_start_main()  ???:0
48 0x0000000000007105 std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_replace_cold()  ???:0
=================================
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3d1d0) [0x7a1f3084c1d0]
#1  0x7a1f33c4dee4 in Gui::LinkView::linkGetElementPicked(SoPickedPoint const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) const from /usr/lib/freecad/lib/libFreeCADGui.so+0x8d4
#2  0x7a1f33c4f15b in Gui::ViewProviderLink::getElementPicked(SoPickedPoint const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) const from /usr/lib/freecad/lib/libFreeCADGui.so+0x12b
#3  0x7a1f33bf717d in Gui::ViewProviderDocumentObject::getElementPicked(SoPickedPoint const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) const from /usr/lib/freecad/lib/libFreeCADGui.so+0x2ed
#4  0x7a1f33bf717d in Gui::ViewProviderDocumentObject::getElementPicked(SoPickedPoint const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) const from /usr/lib/freecad/lib/libFreeCADGui.so+0x2ed
#5  0x7a1f33a6234f in Gui::SoFCUnifiedSelection::getPickedList(SoHandleEventAction*, bool) const from /usr/lib/freecad/lib/libFreeCADGui.so+0x59f
#6  0x7a1f33a66740 in Gui::SoFCUnifiedSelection::handleEvent(SoHandleEventAction*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x1f0
#7  0x7a1f2f6b537c in SoAction::traverse(SoNode*) from /usr/lib/libCoin.so.80+0x6c
[... 41 lines cut ...]
### Full version info

```shell
OS: Arch Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: makepkg
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.12.6, Qt 6.7.2, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * Dracula 0.0.5
  * OpenTheme 2024.9.1
  * freecad.gears 1.3.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-20:
@BogueUser could you try running the dev build to see if you can reproduce? 
(https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

BogueUser, 2024-09-21:
> @BogueUser could you try running the dev build to see if you can reproduce? (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

Will do. 

As a note, it also does this with freecad-1.0rc1-3. The 1.0 pre-releases got pushed to Arch's main repository for some reason.

BogueUser, 2024-09-21:
> @BogueUser could you try running the dev build to see if you can reproduce? (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

@luzpaz I have compiled and tested the weekly release. Aside from it not being able to find pyside2, it seems to be working well. No crashes in the past hour which is better than a crash every 10 minutes or so. 
I'll mark this as solved unless the problem re-appears.

As a side note, [the wiki build instructions for arch](https://wiki.freecad.org/Compile_on_Linux#Arch_Linux) seem to be slightly out of date. ```pyside2-tools pyside2 shiboken2``` are now on the AUR but I cannot install them since yay could not find the relevant version of clang.
```qt5-tools``` Also needed to be installed for me to build.
I also didn't encounter the error here ```Note: Since version 5.0.1 of the openmpi package the binary libmpi_cxx.so stopped being shipped, this breaks the compilation with the current CMake process. To fix this you need to downgrade the package, see [this forum post](https://forum.freecad.org/viewtopic.php?t=85207)```

It seems I am unable to edit the wiki so adding this on here in case anyone with perms wants to fix it.

closed 2024-09-21 by BogueUser