FreeCAD/FreeCAD #16668: Different preferences window on different languages

pOmelchenko, 2024-09-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

i use macos and I installed 1.0rc version and it's fine. I open preferences window
[image]
but than I change language to russian and saw this
[image]

all tries I open thees windows with menu 

en
[image]

ru
[image]

ua
[image]

### Full version info

```shell
OS: macOS 12.7.6
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-20:
Sometimes restarting freecad between language changes fixes weird behavior. Can you test that ?

pOmelchenko, 2024-09-21:
> Can you test that ?
It doesn't help. Restart freecad a few times and restart OS, but problem wasn't go

pOmelchenko, 2024-09-21:
I'll check this bug with weekly-build version

pOmelchenko, 2024-09-22:
```
OS: macOS 12.7.6
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git)
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
```

The problem remains

luzpaz, 2024-09-22:
@johnocfii sorry for the odd request but could you reproduce this on macOS ? TIA!

luzpaz, 2024-09-22:
@kaktusus do you see this ?

kaktusus, 2024-09-22:
Now the situation looks correct. It must be mistakes from the past.
https://crowdin.com/editor/freecad/27908/en-ru?view=comfortable#6622867

In the Polish translation we had an analogous problem that surfaced out of nowhere ...
[image]

The next translation export from Crowdin should fix this.
It's worth watching. :stuck_out_tongue_winking_eye:

JohnOCFII, 2024-09-22:
> @JohnOCFII sorry for the odd request but could you reproduce this on macOS ? TIA!

I'd agree that something isn't working correctly. (Now to figure out how to get back to English!)

[image]

```
OS: macOS 15.0
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git)
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * Curves 0.6.47
```

JohnOCFII, 2024-09-22:
> I'd agree that something isn't working correctly. (Now to figure out how to get back to English!)

Found it: `          <FCText Name="Language">English</FCText>`

maxwxyz, 2024-11-10:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/12006

closed 2024-11-10 by maxwxyz