FreeCAD/FreeCAD #16670: Set Color per Face not working

leoheck, 2024-09-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

My other related ticket were closed with no comments.

Sample file, please, remove the .zip from the filename.
[file.FCStd.zip](https://github.com/user-attachments/files/17068043/file.FCStd.zip)

[Screencast from 2024-09-19 21-00-43.webm](https://github.com/user-attachments/assets/d2284a35-bc3f-468d-8faf-79cb4040ead0)

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38768 (Git) AppImage
Build type: Release
Branch: main
Hash: 758674d40be45e365f4148fb2a5ea6bd6f93bc74
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-09-22:
Try changing the Diffuse color instead of the Ambient one.

leoheck, 2024-09-23:
I will, thanks.

Is there any wiki explaining how this new colors properties work?

FEA-eng, 2024-09-23:
> Is there any wiki explaining how this new colors properties work?

This tool is described [here](https://wiki.freecad.org/Part_ColorPerFace). Individual color settings aren't explained there yet but those are quite general terms. Some explanations: https://bassemtodary.wordpress.com/2013/04/13/ambient-diffuse-specular-and-emissive-lighting/ and https://help.autodesk.com/view/MOBPRO/2025/ENU/?guid=GUID-16D28F9B-DF09-4529-AA44-82F6570EEB0B

leoheck, 2024-09-23:
This indeed helped a lot, thanks.

closed 2024-09-23 by leoheck