FreeCAD/FreeCAD #16672: TechDraw: Insert view > Segfault

leoheck, 2024-09-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps.. I repeated this 4 times. It breaks every time I click on that button. My project takes so much time to open.

1. Model something... my project is a really complex design that is taking 555MB
1. Add Page
2. Insert View
3. Segfault

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45320) [0x7cddc5e45320]
#1  0x7cddc73a9760 in QWidget::setFocus(Qt::FocusReason) from /tmp/.mount_FreeCA51Qc3E/usr/bin/../lib/libQt5Widgets.so.5+0
#2  0x7cdd5c37227b in CmdTechDrawView::activated(int) from /tmp/.mount_FreeCA51Qc3E/usr/lib/TechDrawGui.so+0x9b
#3  0x7cddc9197d18 in Gui::Command::_invoke(int, bool) from /tmp/.mount_FreeCA51Qc3E/usr/bin/../lib/libFreeCADGui.so+0x2a8
#4  0x7cddc91980d4 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /tmp/.mount_FreeCA51Qc3E/usr/bin/../lib/libFreeCADGui.so+0x134
#5  /tmp/.mount_FreeCA51Qc3E/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x7cddc64eb6f2]
#6  0x7cddc736a462 in QAction::triggered(bool) from /tmp/.mount_FreeCA51Qc3E/usr/bin/../lib/libQt5Widgets.so.5+0x32
#7  0x7cddc736ce2b in QAction::activate(QAction::ActionEvent) from /tmp/.mount_FreeCA51Qc3E/usr/bin/../lib/libQt5Widgets.so.5+0xab
#8  /tmp/.mount_FreeCA51Qc3E/usr/bin/../lib/libQt5Widgets.so.5(+0x25d29a) [0x7cddc745d29a]
#9  0x7cddc745d3d7 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /tmp/.mount_FreeCA51Qc3E/usr/bin/../lib/libQt5Widgets.so.5+0xe7
#10  0x7cddc754610a in QToolButton::mouseReleaseEvent(QMouseEvent*) from /tmp/.mount_FreeCA51Qc3E/usr/bin/../lib/libQt5Widgets.so.5+0xa
#11  0x7cddc73aec08 in QWidget::event(QEvent*) from /tmp/.mount_FreeCA51Qc3E/usr/bin/../lib/libQt5Widgets.so.5+0x178
#12  0x7cddc73703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCA51Qc3E/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#13  0x7cddc7378262 in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA51Qc3E/usr/bin/../lib/libQt5Widgets.so.5+0x13a2
#14  0x7cddc9142519 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA51Qc3E/usr/bin/../lib/libFreeCADGui.so+0x99
#15  0x7cddc64b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCA51Qc3E/usr/bin/../lib/libQt5Core.so.5+0x108
#16  0x7cddc73763eb in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCA51Qc3E/usr/bin/../lib/libQt5Widgets.so.5+0x1ab
#17  /tmp/.mount_FreeCA51Qc3E/usr/bin/../lib/libQt5Widgets.so.5(+0x1c6e6f) [0x7cddc73c6e6f]
#18  /tmp/.mount_FreeCA51Qc3E/usr/bin/../lib/libQt5Widgets.so.5(+0x1ca0c9) [0x7cddc73ca0c9]
#19  0x7cddc73703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCA51Qc3E/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#20  0x7cddc9142519 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA51Qc3E/usr/bin/../lib/libFreeCADGui.so+0x99
#21  0x7cddc64b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCA51Qc3E/usr/bin/../lib/libQt5Core.so.5+0x108
#22  0x7cddc6b353ec in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_FreeCA51Qc3E/usr/bin/../lib/libQt5Gui.so.5+0x84c
#23  0x7cddc6b0b0bc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCA51Qc3E/usr/bin/../lib/libQt5Gui.so.5+0xac
#24  /tmp/.mount_FreeCA51Qc3E/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x64bea) [0x7cddc0d56bea]
#25  /tmp/.mount_FreeCA51Qc3E/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x7cddc4b10ad7]
#26  /tmp/.mount_FreeCA51Qc3E/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x7cddc4b13ac7]
#27  /tmp/.mount_FreeCA51Qc3E/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7cddc4b140ce]
#28  0x7cddc650aa36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCA51Qc3E/usr/bin/../lib/libQt5Core.so.5+0x66
#29  0x7cddc64b570b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCA51Qc3E/usr/bin/../lib/libQt5Core.so.5+0x11b
#30  0x7cddc64bd212 in QCoreApplication::exec() from /tmp/.mount_FreeCA51Qc3E/usr/bin/../lib/libQt5Core.so.5+0x82
#31  0x7cddc90a3a23 in Gui::Application::runApplication() from /tmp/.mount_FreeCA51Qc3E/usr/bin/../lib/libFreeCADGui.so+0x983
#32  /tmp/.mount_FreeCA51Qc3E/usr/bin/freecad(+0x7640) [0x64dd867b5640]
#33  /lib/x86_64-linux-gnu/libc.so.6(+0x2a1ca) [0x7cddc5e2a1ca]
#34  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x7cddc5e2a28b]
#35  /tmp/.mount_FreeCA51Qc3E/usr/bin/freecad(+0x79a9) [0x64dd867b59a9]
```

### Full version info

```shell
FreeCAD_weekly-builds-38768-conda-Linux-x86_64-py311.AppImage
FreeCAD 1.1.0, Libs: 1.1.0devR38768 (Git
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pinkavaj, 2024-09-20:
@leoheck I cannot reproduce the issue, pls can you:

* Go to FreeCad menu, select Help -> About FreeCad -> click Copy to Clipboard
* if possible share project file which needs to be open to reproduce the crash

xtemp09, 2024-09-20:
@leoheck, you haven't even reported the OS you're using, X or Wayland, DE and so on.

```cpp
#1  0x7cddc73a9760 in QWidget::setFocus(Qt::FocusReason) from /tmp/.mount_FreeCA51Qc3E/usr/bin/../lib/libQt5Widgets.so.5+0
```

Looks like the culprits are DE or Wayland or some tweaks.

leoheck, 2024-09-20:
> * Go to FreeCad menu, select Help -> About FreeCad -> click Copy to Clipboard, and paste the info here.

Sure @pinkavaj , the main message was updated with the complete info.

Sure , @xtemp09 the system was burning yesterday. Here is the info you need.

[image]

benj5378, 2024-09-30:
@leoheck, did it crash actively? Or did it freeze and then give a backtrace, when you forced closed it after it froze?

This is specific to your model, so we will need the model in order to investigate this.

leoheck, 2024-10-01:
It closed the program immediately. For 4 consecutive times. I don't have the same file anymore, since I could reduce it's complexity, but I am going to try reproduce this later today.

xtemp09, 2024-10-02:
@leoheck, do you have Gnome tweaks installed?

leoheck, 2024-10-02:
Yes I do, why?

These are the extensions I have installed too.
[image]

WandererFan, 2024-11-06:
additional info: https://forum.freecad.org/viewtopic.php?p=790323#p790323

WandererFan, 2024-11-06:
Last FC code:

Forum case:
#3  0x7ff9550dcf9f in TechDrawGui::ViewProviderPage::switchToMdiViewPage() from /usr/local/lib/TechDrawGui.so+0x45          
void ViewProviderPage::switchToMdiViewPage()   
{
    Gui::getMainWindow()->setActiveWindow(m_mdiView);
    m_graphicsView->setFocus();
}

Issue case
#2  0x7cdd5c37227b in CmdTechDrawView::activated(int) from /tmp/.mount_FreeCA51Qc3E/usr/lib/TechDrawGui.so+0x9b
void CmdTechDrawView::activated(int iMsg)
{
    Q_UNUSED(iMsg);
    bool viewCreated = false;
    TechDraw::DrawPage* page = DrawGuiUtil::findPage(this);
    if (!page) {
        return;
    }
    std::string PageName = page->getNameInDocument();

    // switch to the page if it's not current active window
    auto* vpp = dynamic_cast<ViewProviderPage*>
        (Gui::Application::Instance->getViewProvider(page));
    if (vpp) {
        vpp->switchToMdiViewPage();  // <<<<<<<<<<<<<
    }
....

WandererFan, 2024-11-06:
I don't like this feature.  If I want to look at a different tab, I'm quite capable of clicking on the one I want.  I don't see a way to turn it off either.

To reproduce:
- create page
- hide page (close tab) <<<<<
- add view
- boom

commit 992c234e (2024-12-15): [TD] Backport PR #18083: Fix crash on no mdi (fix #16672)

[TD]add preference for auto workbench switch

[TD]prevent fail on auto switch to hidden tab

[TD]prevent crash on attempt to print hidden page

[TD]prevent use of non-exist/hidden tab in exports
closed 2024-11-29 by WandererFan