FreeCAD/FreeCAD #16677: UI: 1.0RC1 Title Menu Indication not same on Windows and Linux

pike2k, 2024-09-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have compared how the Title menu functions on Windows and Linux, and it's not the same on both platforms. On Linux you only get a 'blue indicator' on a Menu line if the menu item is active. On Windows this blue indication is always active.

Example on Windows:
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/Sweden (en_SE)
Installed mods: 
  * ExtremeProDark 2.6.4
  * fasteners 0.5.24
  * Render 2024.6.5
  * ThreadProfile 1.90.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-20:
Do you happen to know the Qt Theme you're running on Windows ? The 'Fusion' theme is recommended for Windows. You can check by installing the weekly and pasting the full [About](https://wiki.freecad.org) info.

Also can you add a screenshot of the Linux example for comparison ?

pike2k, 2024-09-21:
Thanks for the reply. I do not know what the 'Fusion theme' is, not even google could help me. I was testing the RC1 on the same machine where I was previously running 0.21.2 and also weeklies. Does RC1 import old settings and use them? This could be the issue. 

I did test another theme that I found 'OpenTheme'. After installation and applying OpenThemes recommended settings the behavior in topic changed for 1.0RC1 on Windows to act the same way as (vanilla?) RC1 on linux.

So conclusion seems to be that this behavior is related to which Theme you are running.

luzpaz, 2024-09-22:
> I was testing the RC1 on the same machine where I was previously running 0.21.2 and also weeklies. Does RC1 import old settings and use them? This could be the issue.

RC1 is basically a tag a recent weekly (from several weeks ago) that was designated as a release candidate. So settings should work. That being said we're having a bunch of https://github.com/FreeCAD/FreeCAD/labels/Topic%3A%20Stylesheets related bugs crop up and are having trouble nailing them down. Thanks for your feedback and follow-up.

pike2k, 2024-10-11:
I am closing this for now since I don't think we'll get much further?

closed 2024-10-11 by pike2k