FreeCAD/FreeCAD #16680: Failed to compile keymap for custom XKB keymap

kungtiger, 2024-09-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a custom XKB keymap added to the XFCE keyboard manager (xf`ce4-keyboard-manager`) that adds support for German 
umlauts to the UK layout.
When I start FreeCAD it will simply crash. When I start it from the terminal it'll crash with the following print-out:

```
> freecad
FreeCAD 0.21.2, Libs: 0.21.2R33771 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

xkbcommon: ERROR: [XKB-661] Couldn't process include statement for 'gb(de-)'
xkbcommon: ERROR: [XKB-769] Abandoning symbols file "(unnamed)"
xkbcommon: ERROR: Failed to compile xkb_symbols
xkbcommon: ERROR: Failed to compile keymap
[453928:453971:0920/142058.257336:FATAL:xkb_keyboard_layout_engine.cc(654)] Keymap file failed to load: gb,gb-de-,
Trace/breakpoint trap (core dumped)
> _
```

When I remove the custom layout from XFCE keyboard manager and start FreeCAD again it will load without crashing.
I am not sure if this is a FreeCAD issue as `xkbcommon` is a library that FreeCAD includes/loads but it seems strange to me that FreeCAD crashes when other programmes can load and handle custom layouts just fine.

### Full version info

```shell
[code]
OS: Manjaro Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: makepkg
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.1, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
[/code]
```

### Subproject(s) affected?

None

### Anything else?

[freecad.log](https://github.com/user-attachments/files/17076029/freecad.log)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-20:
@kungtiger can you please try with 1.1.0dev weekly ?  (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

maxwxyz, 2024-10-16:
@kungtiger please check if there is still an issue with the latest version, otherwise I'll close this issue.

maxwxyz, 2024-11-02:
Let us know if there is still an issue.

closed 2024-11-02 by maxwxyz