FreeCAD/FreeCAD #16681: Arc by 3 points + 360 revolution create a model that crashes FreeCad when loaded

fouronnes, 2024-09-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Using the lastest x84-64 linux appimage, I can create a model which when exported to .step, then imported back in makes freecad hang (yields the dialog "freecad is not responding").

Here's how to do it:

1. Create a sketch with an arc using the "arc with 3 points" tool:

(note that the bug doesn't happen if putting the arc's center on the Y axis at this step, using the "arc by center" tool for example).

[image]

3. Close the sketch with a horizontal and vertical line:

[image]

4. Revolve the sketch 360° around the sketch vertical axis to create this body part:

[image]

5. Export to a .step file, and import back into freecad

[image]

[image]

[image]

### Bonus bug maybe related

I also found out that if doing step 1, then immediately trying to draw a horizontal line from the arc's lowest point to the vertical sketch axis using the polyline tool, a visual glitch happens:

[Screencast from 2024-09-20 16-34-39.webm](https://github.com/user-attachments/assets/2b8afc1d-9697-429c-84dd-8ac3a343e271)

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38768 (Git) AppImage
Build type: Release
Branch: main
Hash: 758674d40be45e365f4148fb2a5ea6bd6f93bc74
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-09-20:
I can confirm the main issue.

The visual glitch is just how the polyline tool works. It's first mode when connecting to an existing edge is tangential. Press the M key 2x to just connect the polyline.

fouronnes, 2024-09-20:
> The visual glitch is just how the polyline tool works. It's first mode when connecting to an existing edge is tangential. Press the M key 2x to just connect the polyline.

Good to know, thanks for the tip!

chennes, 2024-10-07:
@adrianinsaval is going to check to see if this is a regression.

adrianinsaval, 2024-10-14:
not a regression, 0.21 shows same behavior for me

FEA-eng, 2025-10-17:
Still relevant

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/10/14 15:14:51
Build type: Release
Branch: main
Hash: 41ce37233d64c81c2f105de085635d4dcd669943
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
Installed mods: 
  * A2plus 0.4.68
  * freecad.gears 1.3.0
```

FEA-eng, 2026-11-05:
Still relevant (freeze)

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261020 (Git shallow)
Build date: 2026/10/20 22:15:52
Build type: Release
Branch: (HEAD detached at 9946038aa)
Hash: 9946038aace97af288eb28f2727a169f9d1158c7
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 621.14/NVIDIA Corporation/NVIDIA GeForce RTX 5080/PCIe/SSE2
```