FreeCAD/FreeCAD #16684: 1.0 SKETCHER pan issue

cxrc, 2024-09-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In **Sketcher** Workbench pan with _left button + right button_ in **Blender** mode doesn't work. Pan with _Shift + Wheel Click_ works fine.
[image]
I've tested it with two computers, a laptop with Win11 x64 and a desktop with Win10 x64, with the same result. The laptop was free of extra workbenches.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Spanish/Spain (es_ES)
Installed mods: 
  * A2plus 0.4.65
  * Assembly4 0.50.15
  * CurvedShapes 1.0.9
  * Curves 0.6.45
  * ExplodedAssembly
  * fasteners 0.5.27
  * QuickMeasure 2022.10.28
  * woodworking 0.21.2.33771
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-09-20:
AFAICT LMB+RMB pan works, but you have to make sure you do not press the LMB first. If you do that you start a box-selection in sketch edit mode.

cxrc, 2024-09-20:
Thanks. It works fine pressing RMB first.

cxrc, 2024-09-20:
Solved for me.

luzpaz, 2024-09-20:
We should keep this open because people will keep opening tickets for this. Is there a solution besides clicking RMB first ?

cxrc, 2024-09-20:
You are right. Let's keep this open.

Roy-043, 2024-09-23:
IMO this can, and should, be closed.

There is no way to change the current situation without seriously changing how sketch edit mode works. The fact that you can box-select in sketch edit mode without calling a command is a feature that will hopefully be implemented globally, I would certainly not want to lose it.

Blender navigation already has an alternative way to pan.
@luzpaz What more do you expect?

luzpaz, 2024-09-23:
@Roy-043 it's a nuanced for request for sure but this can't be the only time someone has encountered this UX navigational  issue? Do we force them to learn to click RMB first or can we accomodate some 'range of motion' for LMB to be pressed first with a slight delay on what it triggers if RMB follows immediately after?

cxrc, 2024-09-23:
Ok. I get it.
It's simple, just press RMB first and so won't lose the box-selection feature.
Thanks. I'll close the issue.

closed 2024-09-23 by cxrc