FreeCAD/FreeCAD #16687: Core: 1.0rc: qt-qpa.wayland.textinput constant warning on Wayland

hasecilu, 2024-09-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After closing the preferences page the report view is bombarded with the same warning but with different values at about $2\frac{warning}{s}$ and until you press a key it stops, could be dozens or hundreds of warnings.

It affects Wayland machines.

```python
16:33:06  (qt.qpa.wayland.textinput) virtual void QtWaylandClient::QWaylandTextInputv3::zwp_text_input_v3_leave(wl_surface*) Got leave event for surface 0x0 focused surface 0x61b28f750b10
16:33:06  (qt.qpa.wayland.textinput) virtual void QtWaylandClient::QWaylandTextInputv3::zwp_text_input_v3_leave(wl_surface*) Got leave event for surface 0x0 focused surface 0x61b290b7d580
16:33:07  (qt.qpa.wayland.textinput) virtual void QtWaylandClient::QWaylandTextInputv3::zwp_text_input_v3_leave(wl_surface*) Got leave event for surface 0x0 focused surface 0x61b29530df50
```

Adrián mentioned:
> this should probably be added to freecad's ignore list for warnings,

### Full version info

```shell
OS: Arch Linux (Hyprland/hyprland)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.36460 +2244 (Git)
Build type: Release
Branch: enhancement/add_px_suffix
Hash: 50ddf436362f597186c6e8dd9f94306cce86a582
Python 3.12.6, Qt 6.7.2, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: Spanish/Mexico (es_MX)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Installed mods: 
  * FreeCAD-Movie 2024.2.15
  * freecad.gears 1.2.0
  * Ship 2022.4.11
  * OpenTheme 2024.9.1
  * FreeCAD_WeldFeature 0.1.0
  * FreeGrid 2.0.0
  * FreeCAD-Reinforcement
  * FreeCAD_SheetMetal 0.4.25
  * FreeCAD-Pyramids-and-Polyhedrons
  * Assembly4 0.50.10
  * FastenersWB 0.5.28
  * lattice2 1.0.0
  * Rocket 3.3.0
  * dodo 1.0.1
  * DidYouKnowThat
  * FreeCAD-motion
  * Profiles 0.0.1
  * Alternate_OpenSCAD 1.0.0
  * osh-autodoc-workbench 0.2.3
  * CfdOF 1.27.10
  * SteelColumn
  * A2plus 0.4.67
  * kicadStepUpMod 11.2.4
  * frame 0.1.1
  * Curves 0.6.47
  * FreeCAD_assembly3 0.12.2
  * Render 2024.8.4
```

### Subproject(s) affected?

Core

### Anything else?

Another user reported it on Arch Linux bug tracker using GNOME: https://gitlab.archlinux.org/archlinux/packaging/packages/freecad/-/issues/11

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-23:
Blocker?

hasecilu, 2024-09-23:
I would say it's annoying, I don't know how bad is on other Wayland environments.

InRiPa, 2024-10-12:
Just to respond to your question about how annoying it is for other environments. I'd say very, since it's not only a terminal message but pops up on pretty much every action that I do. To make it stop, I always need to click on another window to let Freecad get out of focus. 

Using Arch Linux with Gnome.

moritzmar, 2024-10-19:
Hey, experiencing the same issue. Just to give you an idea how annoying this is, I created a short screencast...
[output.webm](https://github.com/user-attachments/assets/de3bf089-1ae8-4d91-afd6-53d4fd814f76)
For me, this makes it almost unusable.
Thanks for your great work devs!

luzpaz, 2024-10-19:
@moritzmar can you update to the latest weekly and recreate? (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

luzpaz, 2024-10-19:
labeling this blocker because it is a jarring Ui/UX experience.

moritzmar, 2024-10-19:
About info of the version shipped via Arch repos::
```
OS: Arch Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: makepkg
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
```

Issue seems to be not occurring with AppImage Weekly build amd64
https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/FreeCAD_weekly-builds-38991-conda-Linux-x86_64-py311.AppImage
About info of the AppImage weekly build
```
OS: Arch Linux (GNOME/gnome/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38991 (Git) AppImage
Build type: Release
Branch: main
Hash: ef2156d0f6d116b1642a5531a2d44fc6b7d17042
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion

```

My guess here is that bug is not triggered as the AppImage is using XWayland though...
To verify this, I built the current main branch and can report that the bug is still there (see log in the screenshot at the bottom...)
[image]

About info of the dev(main-[ef2156d](https://github.com/FreeCAD/FreeCAD/commit/ef2156d0f6d116b1642a5531a2d44fc6b7d17042)) version built via PKGBUILD
```
OS: Arch Linux (GNOME/gnome/wayland)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38991 (Git)
Build type: Release
Branch: makepkg
Hash: ef2156d0f6d116b1642a5531a2d44fc6b7d17042
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
```

moritzmar, 2024-10-20:
FYI, if you need a quick workaround, you can launch it with forcing the QT rendering backend to X11 like so:
`QT_QPA_PLATFORM=xcb freecad`

adrianinsaval, 2024-10-22:
please test if the latest weekly version doesn't have a problem anymore

frysztak, 2024-10-23:
weekly AppImage doesn't support Wayland (`(qt.qpa.plugin) Could not find the Qt platform plugin "wayland"`), so it's not possible to test this. however, sometime yesterday I built FreeCAD from `releases/FreeCAD-1-0` branch and the issue still persists. I'm running Arch with GNOME

luzpaz, 2024-10-23:
@frysztak can you paste the [About](https://wiki.freecad.org/About) version of the Arch/Gnome FC you tested on please? TIA

frysztak, 2024-10-23:
of course:
```
OS: Arch Linux (GNOME/gnome/wayland)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38967 (Git)
Build type: Release
Branch: makepkg
Hash: 3ea129fa0bd8002c47c5fa3a75601c41a683d46d
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * Curves 0.6.47
  * ThreadProfile 1.92.0
  * fasteners 0.5.29
  * A2plus 0.4.67
  * Pyramids-and-Polyhedrons
  * OpenDark 2024.3.13
```

adrianinsaval, 2024-10-23:
@wwmayer any clue why these warnings aren't filtered out?

adrianinsaval, 2024-10-23:
I can't reproduce this on kde so I wonder if this is a bug in gnome's compositor. Consider reporting it to them.

Still, freecad should probably be ignoring thees warning since it's not our problem and there's nothing the user can do about them

moritzmar, 2024-10-23:
Just built current main branch. Issue is still there.

[image]

```
OS: Arch Linux (GNOME/gnome/wayland)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39035 (Git)
Build type: Release
Branch: makepkg
Hash: b935c6a740bfe6bdd2e6bf18fd0bdb9f69e5d696
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
```

hasecilu, 2024-10-23:
Still an issue as moritzmar reported, same on self-compiled version on Hyprland. I just installed KDE plasma and effectively no messages there.

```
OS: Arch Linux (Hyprland/hyprland/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39024 (Git) AppImage
Build type: Release
Branch: main
Hash: cf8ad6637351819dfa07d49cfd65d511fe14a084
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Spanish/Mexico (es_MX)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
```

chennes, 2024-10-25:
@adrianinsaval is it possible that those aren't warnings, they are either "info" or "critical"? We're quite specifically only blocking warnings, not all messages.

adrianinsaval, 2024-10-25:
here's the code sending the message, it's using `qCWarning` function so I expected it to be a warning https://github.com/qt/qtwayland/blob/0f3f56b84be4d0fe8cff61ef1c2f71145368b1dd/src/client/qwaylandtextinputv3.cpp#L63

on the other hand we're filtering `qt.qpa.wayland` and this is `qt.qpa.wayland.textinput`, I'm not well versed on this but maybe we should be specific and filter `qt.qpa.wayland.textinput`, can somebody try this?

chennes, 2024-10-25:
The Qt documentation suggests that we could use a wildcard, or be more specific (if we really only care about that one): https://doc.qt.io/qt-6/qloggingcategory.html#installFilter

chennes, 2024-10-25:
Also, those of you testing: you can set the environment variable `QT_LOGGING_RULES` to evaluate different rules. So, e.g. try "QT_LOGGING_RULES=qt.qpa.wayland.*.warning=false".

wwmayer, 2024-10-25:
> on the other hand we're filtering qt.qpa.wayland and this is qt.qpa.wayland.textinput

No, it's not. I have added this test code
```
    QLoggingCategory category2("qt.qpa.wayland");
    qCWarning(category2, "a qt.qpa.wayland warning");
    qCCritical(category2, "a qt.qpa.wayland error");

    QLoggingCategory category3("qt.qpa.wayland.textinput");
    qCWarning(category3, "a qt.qpa.wayland.textinput warning");
```
and it correctly filters the "a qt.qpa.wayland warning" warning but lets pass "a qt.qpa.wayland.textinput warning".

To filter that one this rule must be added: `"qt.qpa.wayland.textinput.warning=false\n"`

hasecilu, 2024-10-25:
> Also, those of you testing: you can set the environment variable `QT_LOGGING_RULES` to evaluate different rules. So, e.g. try `QT_LOGGING_RULES=qt.qpa.wayland.*.warning=false`.

I have some minutes moving things and seems to be effective.

adrianinsaval, 2024-10-25:
> To filter that one this rule must be added: `"qt.qpa.wayland.textinput.warning=false\n"`

do you think we should use a wildcard `"qt.qpa.wayland.*.warning=false\n"` or be specific?

wwmayer, 2024-10-25:
We can try it.

frysztak, 2024-10-25:
> Also, those of you testing: you can set the environment variable `QT_LOGGING_RULES` to evaluate different rules. So, e.g. try `QT_LOGGING_RULES=qt.qpa.wayland.*.warning=false`.

it seems to work, now the only message that shows is this one:

```
The cached device pixel ratio value was stale on window update.  Please file a QTBUG which explains how to reproduce.
```

bhartshorn, 2024-10-27:
> Also, those of you testing: you can set the environment variable `QT_LOGGING_RULES` to evaluate different rules. So, e.g. try `QT_LOGGING_RULES=qt.qpa.wayland.*.warning=false`.

Another confirmation that this works for me, same versions of Arch, Gnome, and FreeCAD as other users on this issue.

closed 2024-10-28 by yorikvanhavre

adrianinsaval, 2024-10-28:
please retest latest main

hasecilu, 2024-10-29:
Just recompiled, looks good!

frysztak, 2024-10-29:
works for me as well, many thanks!

PR #17381 (2024-10-21): Gui: extend warning filters
filter out:
-qt.qpa.wayland
-kf.config.core
-kf.kio.widgets
-kf.service.sycoca
-kf.solid.backends.udisks2

all the kf messages come from kde components usually when using the file open dialog so I don't think they are relevant to us. Should fix #16687 but I couldn't reproduce to actually test it
commit 763fab69 (2024-10-28): Gui: use wildcard for some warnings filters. Fix #16687