FreeCAD/FreeCAD #16690: (1.1.0dev BUG) [Part Design WB] Pocket up-to-face works differently/incorrectly (than in 0.21.2 stable)

nikkoderic, 2024-09-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This particular Pocket up-to-face operation works as intended in 0.21.2 but NOT in 1.1.0dev (latest weekly as of yesterday).

How to reproduce:

1. In Part Design workbench, create a Part, a Body, then a Sketch.
2. Draw two cocentric circles at the center.
2. Pad to create a pipe.
3. Create another Sketch perpendicular to the first Sketch.
4. Draw a circle at the center.
5. Pocket up-to-face of the pipe

OR you can open the attached FCStd file on both version. Make sure to press the Refresh button.
[holeonpipe.zip](https://github.com/user-attachments/files/17082515/holeonpipe.zip)

In 0.21.2, a hole is created on the pipe (as intended):
[image]

However in 1.1.0dev, no hole (just an outline of the hole):
[image]

### Full version info

```shell
1.1.0dev
--------
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38728 (Git)
Build type: Release
Branch: main
Hash: 9de78e27f4148e8032594788a9804486e769388b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/Philippines (en_PH)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

0.21.2
------
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Philippines (en_PH)
```

### Subproject(s) affected?

PartDesign

### Anything else?

Forum post:
https://forum.freecad.org/viewtopic.php?t=90766

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-21:
@FlachyJoe FYI

FlachyJoe, 2024-09-21:
Confirmed, please assign me.

FlachyJoe, 2024-09-21:
Phew! It's not due to #16030  :relieved: 
So it needs more search :thinking:

maxwxyz, 2024-09-25:
If you need some tests let me know.

AIRCAP, 2024-09-25:
```
9bc5675ab3e731ad160ab5adf80dbc9dda6f88d7 is the first bad commit
commit 9bc5675ab3e731ad160ab5adf80dbc9dda6f88d7
Author: Chris Hennes <chennes@pioneerlibrarysystem.org>
Date:   Tue Apr 30 16:19:21 2024 -0500

    Correct flag

 cMake/FreeCAD_Helpers/SetGlobalCompilerAndLinkerSettings.cmake | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

```

AIRCAP, 2024-09-25:
As I understand this commit enables the refactored code with FC_USE_TNP_FIX - and disables the old legacy code, which later got removed.

Now in case of Pocket, the old legacy code had a dedicated implementation in
https://github.com/FreeCAD/FreeCAD/blob/9bc5675ab3e731ad160ab5adf80dbc9dda6f88d7/src/Mod/PartDesign/App/FeaturePocket.cpp#L71C6-L71C30

while the new code treats Pocket as an Extrusion implemented in
https://github.com/FreeCAD/FreeCAD/blob/9bc5675ab3e731ad160ab5adf80dbc9dda6f88d7/src/Mod/PartDesign/App/FeatureExtrude.cpp#L430

so the fun job will be to make buildExtrusion backwards compatible to pocket - without introducing incompatible changes in the extrusion feature itself...

this is beyond me - have fun ;)

AIRCAP, 2024-09-25:
fun detail of "Offset to face is set to anything but exact zero, the pocket works. but positive values go all the way through and further, negative values go almost through - in the right place. but if its zero there is no pocket at all

AIRCAP, 2024-09-25:
the old code would in such cases throw an exception "Pocket: SketchBased: Up to Face: Offset not supported yet for non-planar faces"

FlachyJoe, 2024-09-25:
Offset works has expected: positive go further (out of cylinder so it breaks through), negative go closer.

We're facing an OCC bug we'll have to workaround. I'm still testing.

AIRCAP, 2024-09-25:
Fun fact: If one does the following change:

```
--- a/src/Mod/PartDesign/App/FeatureSketchBased.cpp
+++ b/src/Mod/PartDesign/App/FeatureSketchBased.cpp
@@ -932,12 +932,12 @@ void ProfileBased::addOffsetToFace(TopoShape& upToFace, const gp_Dir& dir, doubl
 {
     // Move the face in the extrusion direction
     // TODO: For non-planar faces, we could consider offsetting the surface
-    if (fabs(offset) > Precision::Confusion()) {
+    //if (fabs(offset) > Precision::Confusion()) {^M
         gp_Trsf mov;
         mov.SetTranslation(offset * gp_Vec(dir));
         TopLoc_Location loc(mov);
         upToFace.move(loc);
-    }
+    //}^M
 }
```

The hole gets created correctly - even with an offset of zero

**BUT**

here is a corollary for the pocket, the equivalent pad  using the same sketch and the same (up to) face

[padonpipe.zip](https://github.com/user-attachments/files/17139223/padonpipe.zip)

if the above patch is applied (removing the fabs(offset)>Precision::Confusion() check) which means the offset is always applied, then the pad disappears (it gets a size of 0 because for some reason the offset is calculated from the base instead of the end face, unless the pad direction is reversed)

a similar issue might be with the pocket - since internally I think it actually applies ExtrudeOption::InverseDirection.

all very confusing

so the trick is to fix the hole without breaking the above pad ;)

AIRCAP, 2024-09-25:
note: I did some debug code to visualize the prism generated in

https://github.com/FreeCAD/FreeCAD/blob/69676a0f9d2d5b288bda3d1c6dbad5a6ec26bfaa/src/Mod/PartDesign/App/FeatureExtrude.cpp#L687

this prism has already the wrong shape. If the "offset" is set to zero, it goes to exactly the **inner** surface of the pipe.   if the offset is nonzero, it goes to the outer surface of the pipe (Face 1, as it should)  plus the offset

AIRCAP, 2024-09-25:
another interesting thing happens if the pad direction is reversed - with reversed pad direction, the offset in the pad is again applied to the uptoface face - not on the sketch (origin) side... i don't know if that is intended

bgbsww, 2024-09-26:
So the new code should exhibit the same behaviors as LinkStage3 where the replacement code came from.  If it doesn't, this is just a flat out bug, and an LS3 code comparison might help to narrow it down.  If it does, and that's different than 0.21.2, then we're going to have to put on our thinking caps.

A quick skim of that comparison shows that in buildExtrusion, we are doing
```
       this->positionByPrevious();
        auto invObjLoc = getLocation().Inverted();

        auto invTrsf = invObjLoc.Transformation();

        base.move(invObjLoc);

```and LS3:
```
        auto invObjLoc = this->positionByPrevious();
        auto invTrsf = invObjLoc.Transformation();

        base.move(invObjLoc);
```
and we do

```
            else{
                if (fabs(Offset.getValue()) > Precision::Confusion()){
                    return new App::DocumentObjectExecReturn(QT_TRANSLATE_NOOP("Exception", "Extrude: Can only offset one face"));
                }
                // open the shell by removing the furthest face
                upToShape = makeShellFromUpToShape(upToShape, sketchshape, dir);
            }
```
where LS3 does nothing, and that makeShellFromUpToShape is brand new, but LS3 doesn't support that operation, and it doesn't appear germane.

Maybe that Inverted() is worth playing with.

AIRCAP, 2024-09-27:
which means the following patch curiously works around the issue (I wouldn't call it a fix, just again sidestepping OCCT bugs)
```
diff --git a/src/Mod/PartDesign/App/FeatureExtrude.cpp b/src/Mod/PartDesign/App/FeatureExtrude.cpp
index d0ea1a748c..0f47828d44 100644
--- a/src/Mod/PartDesign/App/FeatureExtrude.cpp
+++ b/src/Mod/PartDesign/App/FeatureExtrude.cpp
@@ -684,7 +684,11 @@ App::DocumentObjectExecReturn* FeatureExtrude::buildExtrusion(ExtrudeOptions opt
                 return App::DocumentObject::StdReturn;
             }
             try {
-                prism.makeElementPrismUntil(base,
+                TopoShape _base(base);
+                if (inverseDirection) {
+                    _base.makeElementPrism(upToShape,dir);
+                }
+                prism.makeElementPrismUntil(_base,
                                             sketchshape,
                                             supportface,
                                             upToShape,

```

pull request?

AIRCAP, 2024-09-27:
note the inverse direction check is needed to avoid breaking above "padonpipe.FCStd" - the pad would otherwise disappear (the workaround breaks the pad - which also means if an offset is provided - which triggers the workaround thanks to the exception - the pad also fails)

AIRCAP, 2024-09-27:
i simplified the fix - instead of calling that other function we can simply provide an empty base - the prism then goes from the sketch to the selected face disregarding the base shape completely - which is fine thanks to the later subtractive operation

FlachyJoe, 2024-09-27:
Hi,
if you provide a null base, you trigger 
https://github.com/FreeCAD/FreeCAD/blob/d519f24df9931e4a98d19ee39cdcd6124ae6a7bd/src/Mod/Part/App/TopoShapeExpansion.cpp#L4255-L4258

so the profile is used as found.

I found the fuse or cut is made twice, one with
https://github.com/FreeCAD/FreeCAD/blob/d519f24df9931e4a98d19ee39cdcd6124ae6a7bd/src/Mod/Part/App/TopoShapeExpansion.cpp#L4401-L4408

and one with
https://github.com/FreeCAD/FreeCAD/blob/d519f24df9931e4a98d19ee39cdcd6124ae6a7bd/src/Mod/PartDesign/App/FeatureExtrude.cpp#L765-L772

@realthunder can you explain?

FEA-eng, 2024-10-01:
Seems to be fixed by #16870.

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git)
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

maxwxyz, 2024-10-01:
I'll close this then.
@FlachyJoe @nikkoderic please check and otherwise I'll reopen the issue.

closed 2024-10-01 by maxwxyz