FreeCAD/FreeCAD #16693: TechDraw: BrokenView warning message is misleading - should not mention objects

furgo16, 2024-09-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

As described on https://forum.freecad.org/viewtopic.php?t=90657 as well.

Steps to reproduce:

1. Switch to TechDraw workbench
2. Create an empty page
3. Press the [Broken view ](https://wiki.freecad.org/TechDraw_BrokenView) button with an empty selection
4. The warning message instructs the user to select objects to break. However, the tool seems to work only with existing views, not objects

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38782 (Git) Snap 1143
Build type: Release
Branch: main
Hash: 8b9f5bdc4f4bd3cf4a440365fcc7cbcd2afe8580
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * ezydxf
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2024-09-21:
If it's supposed to work with e.g. bodies, I could not get it to work:

https://github.com/user-attachments/assets/b7e55303-a948-45ad-bb01-8b9239794409

[BrokenView.zip](https://github.com/user-attachments/files/17083043/BrokenView.zip)

WandererFan, 2024-09-21:
https://github.com/FreeCAD/FreeCAD/pull/16641

furgo16, 2024-09-21:
Great, thank you!

jangalasriramd7, 2024-09-22:
Hi there, can i work on this issue..

luzpaz, 2024-09-23:
@jangalasriramd7 looks like it's already been addressed

closed 2024-09-30 by WandererFan