FreeCAD/FreeCAD #16695: Crushes on apply settings on "first start setup"

pOmelchenko, 2024-09-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. click button "open first start setup"
2. click "done" button
3. masOS show message with some information
    - gist with report https://gist.github.com/pOmelchenko/0595aa9ca6f85e6e925c272bd7836612

### Full version info

```shell
OS: macOS 12.7.6
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-09-21:
I cannot reproduce.

```ruby
OS: Ubuntu 24.04.1 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git)
Build type: Debug
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk , OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

```ruby
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git)
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

pOmelchenko, 2024-09-21:
> Version: 1.1.0dev.38794 (Git)

🤔 I'll check with weekly-bundle version

pOmelchenko, 2024-09-22:
```
OS: macOS 12.7.6
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git)
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
```

do not reproduces now

closed 2024-09-22 by pOmelchenko