FreeCAD/FreeCAD #16696: Segmentation fault in 20sec just rotating/panning/zooming the sample BIMExample.FCStd file every time.

klugd, 2024-09-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Opening the BIMEcxamle.FCStd file from the start page, just some rotating, zooming, panning movement and it crashes in max 20 sec every time.

### Full version info

```shell
Opening the BIMEcxamle.FCStd file from the start page, just some rotating, zooming, panning movement and it crashes in max 20 sec every time.

FreeCAD 1.0.0RC1 
Rev: 1.0.0RC1R38642 (Git)
Release date: 24/09/05
OS: Linux Mint 22 64-bit
```

### Subproject(s) affected?

None

### Anything else?

Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45320) [0x75b5e6e45320]
#1  0x75b5ea5bfed3 in Gui::LinkView::linkGetElementPicked(SoPickedPoint const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) const from /tmp/.mount_freecagT1bVv/usr/bin/../lib/libFreeCADGui.so+0x543
#2  0x75b5ea5c0423 in Gui::ViewProviderLink::getElementPicked(SoPickedPoint const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) const from /tmp/.mount_freecagT1bVv/usr/bin/../lib/libFreeCADGui.so+0x113
#3  0x75b5ea43a9d2 in Gui::SoFCUnifiedSelection::getPickedList(SoHandleEventAction*, bool) const from /tmp/.mount_freecagT1bVv/usr/bin/../lib/libFreeCADGui.so+0x572
#4  0x75b5ea43f797 in Gui::SoFCUnifiedSelection::handleEvent(SoHandleEventAction*) from /tmp/.mount_freecagT1bVv/usr/bin/../lib/libFreeCADGui.so+0x1e7
#5  0x75b5e66c5e38 in SoAction::traverse(SoNode*) from /tmp/.mount_freecagT1bVv/usr/bin/../lib/./libCoin.so.80+0x68
#6  0x75b5e687fcca in SoChildList::traverse(SoAction*, int, int) from /tmp/.mount_freecagT1bVv/usr/bin/../lib/./libCoin.so.80+0x15a
#7  0x75b5e68cfce8 in SoGroup::doAction(SoAction*) from /tmp/.mount_freecagT1bVv/usr/bin/../lib/./libCoin.so.80+0x58
#8  0x75b5e68eaf8a in SoSeparator::doAction(SoAction*) from /tmp/.mount_freecagT1bVv/usr/bin/../lib/./libCoin.so.80+0x2a
#9  0x75b5e66c5e38 in SoAction::traverse(SoNode*) from /tmp/.mount_freecagT1bVv/usr/bin/../lib/./libCoin.so.80+0x68
#10  0x75b5e66cfa2c in SoHandleEventAction::beginTraversal(SoNode*) from /tmp/.mount_freecagT1bVv/usr/bin/../lib/./libCoin.so.80+0xac
#11  0x75b5e66c6b1a in SoAction::apply(SoNode*) from /tmp/.mount_freecagT1bVv/usr/bin/../lib/./libCoin.so.80+0x2ea
#12  0x75b5e689bdd2 in SoEventManager::actuallyProcessEvent(SoEvent const*) from /tmp/.mount_freecagT1bVv/usr/bin/../lib/./libCoin.so.80+0x72
#13  0x75b5ea4edfee in SIM::Coin3D::Quarter::QuarterWidget::processSoEvent(SoEvent const*) from /tmp/.mount_freecagT1bVv/usr/bin/../lib/libFreeCADGui.so+0x1e
#14  0x75b5ea4f9588 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::processSoEvent(SoEvent const*) from /tmp/.mount_freecagT1bVv/usr/bin/../lib/libFreeCADGui.so+0x58
#15  0x75b5ea506230 in Gui::NavigationStyle::processSoEvent(SoEvent const*) from /tmp/.mount_freecagT1bVv/usr/bin/../lib/libFreeCADGui.so+0x40
#16  0x75b5ea508f2b in Gui::BlenderNavigationStyle::processSoEvent(SoEvent const*) from /tmp/.mount_freecagT1bVv/usr/bin/../lib/libFreeCADGui.so+0x2db
#17  0x75b5ea5064c5 in Gui::NavigationStyle::processEvent(SoEvent const*) from /tmp/.mount_freecagT1bVv/usr/bin/../lib/libFreeCADGui.so+0x65
#18  0x75b5ea52e6bb in Gui::View3DInventorViewer::processSoEvent(SoEvent const*) from /tmp/.mount_freecagT1bVv/usr/bin/../lib/libFreeCADGui.so+0x6b
#19  0x75b5ea4e491d in SIM::Coin3D::Quarter::EventFilter::eventFilter(QObject*, QEvent*) from /tmp/.mount_freecagT1bVv/usr/bin/../lib/libFreeCADGui.so+0x1fd
#20  0x75b5e74b68bb in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /tmp/.mount_freecagT1bVv/usr/bin/../lib/libQt5Core.so.5+0x7b
#21  0x75b5e837039e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_freecagT1bVv/usr/bin/../lib/libQt5Widgets.so.5+0x6e
#22  0x75b5e83778f9 in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_freecagT1bVv/usr/bin/../lib/libQt5Widgets.so.5+0xa39
#23  0x75b5ea13e499 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_freecagT1bVv/usr/bin/../lib/libFreeCADGui.so+0x99
#24  0x75b5e74b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_freecagT1bVv/usr/bin/../lib/libQt5Core.so.5+0x108
#25  0x75b5e83763eb in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_freecagT1bVv/usr/bin/../lib/libQt5Widgets.so.5+0x1ab
#26  /tmp/.mount_freecagT1bVv/usr/bin/../lib/libQt5Widgets.so.5(+0x1c6e6f) [0x75b5e83c6e6f]
#27  /tmp/.mount_freecagT1bVv/usr/bin/../lib/libQt5Widgets.so.5(+0x1ca0c9) [0x75b5e83ca0c9]
#28  0x75b5e83703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_freecagT1bVv/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#29  0x75b5ea13e499 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_freecagT1bVv/usr/bin/../lib/libFreeCADGui.so+0x99
#30  0x75b5e74b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_freecagT1bVv/usr/bin/../lib/libQt5Core.so.5+0x108
#31  0x75b5e7b353ec in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_freecagT1bVv/usr/bin/../lib/libQt5Gui.so.5+0x84c
#32  0x75b5e7b0b0bc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_freecagT1bVv/usr/bin/../lib/libQt5Gui.so.5+0xac
#33  /tmp/.mount_freecagT1bVv/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x64bea) [0x75b5e1b15bea]
#34  /tmp/.mount_freecagT1bVv/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x75b5e5b10ad7]
#35  /tmp/.mount_freecagT1bVv/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x75b5e5b13ac7]
#36  /tmp/.mount_freecagT1bVv/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x75b5e5b140ce]
#37  0x75b5e750aa36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_freecagT1bVv/usr/bin/../lib/libQt5Core.so.5+0x66
#38  0x75b5e74b570b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_freecagT1bVv/usr/bin/../lib/libQt5Core.so.5+0x11b
#39  0x75b5e74bd212 in QCoreApplication::exec() from /tmp/.mount_freecagT1bVv/usr/bin/../lib/libQt5Core.so.5+0x82
#40  0x75b5ea09f943 in Gui::Application::runApplication() from /tmp/.mount_freecagT1bVv/usr/bin/../lib/libFreeCADGui.so+0x983
#41  /tmp/.mount_freecagT1bVv/usr/bin/freecad(+0x7640) [0x5593c810c640]
#42  /lib/x86_64-linux-gnu/libc.so.6(+0x2a1ca) [0x75b5e6e2a1ca]
#43  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x75b5e6e2a28b]
#44  /tmp/.mount_freecagT1bVv/usr/bin/freecad(+0x79a9) [0x5593c810c9a9]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

klugd, 2024-09-21:
I have tried with Blender and CAD, and among the movements, rotating is enough.

klugd, 2024-09-21:
Tested with an other pc. It takes more time (app 1-1.5 min), but the same crash with just rotating the same file in Blender mode
I could reproduce it  multiple times.
Same OS, same Freecad appimage.
thx

FlachyJoe, 2024-09-21:
Please copy/paste the full information given by Help, About FreeCAD, Copy to clipboard.

klugd, 2024-09-22:
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)

yorikvanhavre, 2024-09-23:
1 minute and going, nothing happens. I can't reproduce that crash...

Your backtrace mentions a LinkView object. There are indeed some in that file. Could you try this?:

1. See if another model file with no App::Link object crashes? More models on https://github.com/yorikvanhavre/FreeCAD-BIM-examples
2. Try to create a simple object, and use the Draft Array tool to make an array. The Draft array tool produces App::Links. Save that file, see if it crashes too?

klugd, 2024-09-23:
1. Not all, but I have tested most of them, they where stable
2. Indeed, creating a simple object and using the Draft Array tool, it crashes almost immediately at some movements (even with zoom).

benj5378, 2024-09-30:
Looks duplicate to #16732 (same backtrace)

@klugd, can you test on RC2?

closed 2024-10-04 by klugd