FreeCAD/FreeCAD #16697: No trackball settings and taskbar moving

edwin1959, 2024-09-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I am using a trackball from 3Dconnection. In version 0.21.2 there where settings for the speed of axis. 
These settings are missing in the new version RC1.
Every time i create something my taskbar move futher to the bottom of the screen. I use a second screen for task en model plane.

Bye the way you guys did a great job. Compliments.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL)
Installed mods: 
  * 3D_Printing_Tools
  * A2plus 0.4.60n
  * Assembly4 0.50.4
  * btl 0.9.9
  * Curves 0.6.13
  * ExplodedAssembly
  * fasteners 0.4.66
  * freecad.gears 1.0.0
  * OpticsWorkbench 1.0.12
  * python36_x64-ocl
  * sheetmetal 0.4.6
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-09-21:
3Dconnexion settings in FreeCAD were removed in favor of the global settings in the 3dconnexion driver app.

edwin1959, 2024-10-06:
Thanks for your reply.

closed 2024-10-06 by edwin1959